springfall2008/batpred #1816: Monitor Mode/Read only mode

dochtie, 2024-12-30:
**Describe the bug**
In monitor mode Predbat overrides charge speed back to default speeds. 

In read only mode this doesn’t happen. 

I also noticed that in read only mode with low power mode on, Predbat doesn’t take into account the set charge speed and assumes the low power mode will be active. 

**Expected behaviour**
This may be intentional but I would expect Predbat to not to adjust charge speed setting when in monitor mode.

In read only mode disregard low power mode and assumes current values set will remain.

**Predbat version**

8.8.18

**Environment details**

- Inverter and battery setup
- GivEnergy AIO
- Standard HAOS installer or Docker
- HAOS

<head></head>
2024-12-30 00:04:26.814768: Inverter 0 count register writes 1
--
2895 | 2024-12-30 00:04:26.814732: Inverter 0 Current reserve is 4.0 already at target
2894 | 2024-12-30 00:04:26.814646: Inverter 0 set charge rate 6000 via REST successful on retry 0
2893 | 2024-12-30 00:04:25.583094: Inverter 0 current charge rate is 4120W and new target is 6000W
2892 | 2024-12-30 00:04:25.583050: Inverter 0 Set charge window is disabled

springfall2008, 2025-01-01:
For the charge rate/reserve changes I agree.

The idea of read-only mode is that it will show you what Predbat would do, so in this case if set_low_power is turned on then I think it should show the plan shouldn't it?

dochtie, 2025-01-01:
> For the charge rate/reserve changes I agree.
> 
> The idea of read-only mode is that it will show you what Predbat would do, so in this case if set_low_power is turned on then I think it should show the plan shouldn't it?

Agreed!

dochtie, 2025-01-02:
Actually I think this is a (very minor) bug after all. In monitor mode Predbat is assuming low power mode is going to be active. 

But my interpretation of monitor mode is it’s monitoring the inverter with its current settings. The monitor mode plan is incorrect if the low power switch is on.

springfall2008, 2025-01-04:
I agree it shouldn't assume in it Monitor mode, will look

springfall2008, 2025-01-04:
Fix on main for testing

closed 2025-01-06 by dochtie