springfall2008/batpred #1821: Sub-optimal import/export plan since 8.8.17

sammort, 2024-12-30:
### Describe the bug
Since v8.8.17 of Predbat I'm getting a seriously sub-optimal charge/discharge plan for my AIO. As I'm on IOG it is usual for predbat to plan a charge > discharge > charge during the off-peak period overnight. However, v8.8.17 and above show the same sort of pattern but with an extremely low limit (charging/discharging from 4% limit > 5% limit > 4% limit).

Versions before v8.8.17 behave as expected, charging and discharging as much as possible before ending with a full battery by the time peak period begins at 5:30am

### Expected behaviour
Predbat continues to optimise the plan to minimize costs

### Predbat version
v8.8.17 and above

### Environment details

- AIO
- Docker install
- IOG tariff

### Screenshots
#### 8.8.17 plan
[image]

#### 8.8.16 plan
[image]

### Predbat debug yaml file
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/18277859/predbat_debug.yaml.txt)

springfall2008, 2024-12-31:
That's bonkers I agree, will look

springfall2008, 2024-12-31:
So I can see your plan but when I re-run it I get the more sensible plan instead. Could you please re-create the debug YAML with 'debug_enable' turned on so I get the intermediate load data information? It maybe worth updating to the latest release also in case I've already fixed it.

springfall2008, 2024-12-31:
Okay I managed to reproduce this now, it seems to relate to the car charging - I'll debug it further

springfall2008, 2024-12-31:
A new release v8.8.20 has been made to fix this issue

sammort, 2024-12-31:
Thanks for taking a look at this and getting a fix in place so quickly @springfall2008.

Hope you have a Happy New Year.

closed 2024-12-31 by sammort