springfall2008/batpred #1823: Charge script called before charge rate set

chris-y, 2024-12-31:
**Describe the bug**
The charge_start_service is being called before timed_charge_current has been set.

This was fixed as part of #1690 but appears to have returned.

**Expected behaviour**
timed_charge_current should be set before the service is called so the value can be used within the service.

**Predbat version**

8.8.19

**Environment details**

- Solis inverter using script to control via cloud
- HAOS

**Screenshots**
Service ran at 06:00:18:

[image]

timed_charge_current was set to correct value at 06:00:27:

[image]

**Log file**
```

2024-12-31 06:00:03.446106: Inverter 0: Type MINE Solis Inverter
2024-12-31 06:00:03.446261: Note: Inverter 0: Using inverter type MINE - not all features are available on all Inverter types
2024-12-31 06:00:03.446369: Inverter 0 does not have timed pause support enabled
2024-12-31 06:00:03.446599: Reserve min: 10.0% Battery_min:10.0%
2024-12-31 06:00:03.446707: Inverter 0 with soc_max 9.6 kWh nominal_capacity 9.6 kWh battery rate raw 3600.0 w charge rate 3.6 kW discharge rate 3.6 kW battery_rate_min 0.0 w ac limit 3.6 kW export limit 3.6 kW reserve 10.0 % current_reserve 10.0 %
2024-12-31 06:00:03.447116: Inverter 0 scheduled_charge_enable off scheduled_discharge_enable on
2024-12-31 06:00:03.447365: Inverter 0 SOC: 2.59kW 27% Current charge rate 3240W Current discharge rate 3600W Current power -317.0W Current voltage 48.7V
2024-12-31 06:00:03.447706: Inverter 0 No write needed for scheduled_charge_enable as False == False
2024-12-31 06:00:03.447751: Inverter 0 scheduled_charge_enable set to False
2024-12-31 06:00:03.447868: Inverter 0 scheduled charge enable is False
2024-12-31 06:00:03.447906: Inverter 0 charge windows currently []
2024-12-31 06:00:03.447951: Inverter 0 Charge settings: timed charged is disabled, power 3.24 kW
2024-12-31 06:00:03.448160: Inverter 0 No write needed for scheduled_discharge_enable as True == True
2024-12-31 06:00:03.448199: Inverter 0 sensor.predbat_MINE_0_scheduled_discharge_enable set to True
2024-12-31 06:00:03.448284: Inverter 0 scheduled discharge enable is True
2024-12-31 06:00:03.448534: Inverter 0 idle time is 12-31 06:00:00-12-31 23:59:00
2024-12-31 06:00:03.448591: Inverter 0 discharge windows currently [{'start': 0, 'end': 360, 'average': 0}, {'start': 1440, 'end': 1800, 'average': 0}]
2024-12-31 06:00:03.448633: Note: Inverter does not support discharge freeze - disabled
2024-12-31 06:00:03.448665: Note: Inverter does not support charge freeze - disabled
2024-12-31 06:00:03.448702: Note: Inverter does not support reserve - disabling reserve functions
2024-12-31 06:00:03.448875: Found 1 inverters totals: min reserve 0.96 current reserve 0.96 soc_max 9.6 soc 2.592 charge rate 3.2399999999999998 kW discharge rate 3.5999999999999996 kW battery_rate_min 0.0 w ac limit 3.6 export limit 3.6 kW loss charge 4 % loss discharge 4 % inverter loss 4 %
2024-12-31 06:00:03.606486: Base charge window [  ]
2024-12-31 06:00:03.606879: Base export window [ 31-12 00:00:00 - 31-12 06:00:00 @ 0p 0.0%, 01-01 00:00:00 - 01-01 06:00:00 @ 0p 0.0% ]
2024-12-31 06:00:03.606999: Plan was last updated on 2024-12-31 05:50:00+00:00 and is now 10.0 minutes old
2024-12-31 06:00:03.607063: Current export window has expired, removing it
2024-12-31 06:00:03.607336: Best charge window [ 31-12 06:00:00 - 31-12 08:30:00 @ 0.0p 100%, 31-12 23:30:00 - 01-01 05:30:00 @ 7.0p 100%, 01-01 23:30:00 - 02-01 05:30:00 @ 7.0p 100% ]
2024-12-31 06:00:03.607385: Best export window [  ]
2024-12-31 06:00:03.611267: PV Forecast 2.6 kWh and 10% Forecast 1.2 kWh pv cloud factor 0.3
2024-12-31 06:00:03.634150: Load divergence over 8.0 hours mean 647.31 W, min 174.17 W, max 3392.57 W, std dev 475.65 W, divergence 36.74%

2024-12-31 06:00:06.699029:  STATE:  [    Chrg,     Chrg,     Chrg,     Chrg,     Chrg,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,         ,     Chrg,     Chrg,     Chrg,     Chrg,     Chrg,     Chrg,     Chrg,     Chrg,     Chrg,     Chrg,     Chrg,     Chrg,         ,         ]
2024-12-31 06:00:07.595312: Evaluate trigger {'name': 'large', 'minutes': 60, 'energy': 1.0} results off total_energy 0
2024-12-31 06:00:07.632781: Evaluate trigger {'name': 'small', 'minutes': 15, 'energy': 0.25} results off total_energy 0
2024-12-31 06:00:07.672485: Best charging limit socs [9.6, 9.6, 9.6] export [] gives import battery 48.37 house 17.79 export 0.24 metric 375.3 metric10 450.99
2024-12-31 06:00:08.056428: Web interface history update
2024-12-31 06:00:08.635888: Export threshold used for optimisation was 13.8p
2024-12-31 06:00:08.954304: High export rate windows [0: 12-31 06:00:00 - 12-31 06:30:00 @ 15.0, 1: 12-31 06:30:00 - 12-31 07:00:00 @ 15.0, 2: 12-31 07:00:00 - 12-31 07:30:00 @ 15.0, 3: 12-31 07:30:00 - 12-31 08:00:00 @ 15.0, 4: 12-31 08:00:00 - 12-31 08:30:00 @ 15.0, 5: 12-31 08:30:00 - 12-31 09:00:00 @ 15.0, 6: 12-31 09:00:00 - 12-31 09:30:00 @ 15.0, 7: 12-31 09:30:00 - 12-31 10:00:00 @ 15.0, 8: 12-31 10:00:00 - 12-31 10:30:00 @ 15.0, 9: 12-31 10:30:00 - 12-31 11:00:00 @ 15.0, 10: 12-31 11:00:00 - 12-31 11:30:00 @ 15.0, 11: 12-31 11:30:00 - 12-31 12:00:00 @ 15.0, 12: 12-31 12:00:00 - 12-31 12:30:00 @ 15.0, 13: 12-31 12:30:00 - 12-31 13:00:00 @ 15.0, 14: 12-31 13:00:00 - 12-31 13:30:00 @ 15.0, 15: 12-31 13:30:00 - 12-31 14:00:00 @ 15.0, 16: 12-31 14:00:00 - 12-31 14:30:00 @ 15.0, 17: 12-31 14:30:00 - 12-31 15:00:00 @ 15.0, 18: 12-31 15:00:00 - 12-31 15:30:00 @ 15.0, 19: 12-31 15:30:00 - 12-31 16:00:00 @ 15.0, 20: 12-31 16:00:00 - 12-31 16:30:00 @ 15.0, 21: 12-31 16:30:00 - 12-31 17:00:00 @ 15.0, 22: 12-31 17:00:00 - 12-31 17:30:00 @ 15.0, 23: 12-31 17:30:00 - 12-31 18:00:00 @ 15.0, 24: 12-31 18:00:00 - 12-31 18:30:00 @ 15.0, 25: 12-31 18:30:00 - 12-31 19:00:00 @ 15.0, 26: 12-31 19:00:00 - 12-31 19:30:00 @ 15.0, 27: 12-31 19:30:00 - 12-31 20:00:00 @ 15.0, 28: 12-31 20:00:00 - 12-31 20:30:00 @ 15.0, 29: 12-31 20:30:00 - 12-31 21:00:00 @ 15.0, 30: 12-31 21:00:00 - 12-31 21:30:00 @ 15.0, 31: 12-31 21:30:00 - 12-31 22:00:00 @ 15.0, 32: 12-31 22:00:00 - 12-31 22:30:00 @ 15.0, 33: 12-31 22:30:00 - 12-31 23:00:00 @ 15.0, 34: 12-31 23:00:00 - 12-31 23:30:00 @ 15.0, 35: 12-31 23:30:00 - 01-01 00:00:00 @ 15.0, 36: 01-01 00:00:00 - 01-01 00:30:00 @ 15.0, 37: 01-01 00:30:00 - 01-01 01:00:00 @ 15.0, 38: 01-01 01:00:00 - 01-01 01:30:00 @ 15.0, 39: 01-01 01:30:00 - 01-01 02:00:00 @ 15.0, 40: 01-01 02:00:00 - 01-01 02:30:00 @ 15.0, 41: 01-01 02:30:00 - 01-01 03:00:00 @ 15.0, 42: 01-01 03:00:00 - 01-01 03:30:00 @ 15.0, 43: 01-01 03:30:00 - 01-01 04:00:00 @ 15.0, 44: 01-01 04:00:00 - 01-01 04:30:00 @ 15.0, 45: 01-01 04:30:00 - 01-01 05:00:00 @ 15.0, 46: 01-01 05:00:00 - 01-01 05:30:00 @ 15.0, 47: 01-01 05:30:00 - 01-01 06:00:00 @ 15.0, 48: 01-01 06:00:00 - 01-01 06:30:00 @ 15.0, 49: 01-01 06:30:00 - 01-01 07:00:00 @ 15.0, 50: 01-01 07:00:00 - 01-01 07:30:00 @ 15.0, 51: 01-01 07:30:00 - 01-01 08:00:00 @ 15.0, 52: 01-01 08:00:00 - 01-01 08:30:00 @ 15.0, 53: 01-01 08:30:00 - 01-01 09:00:00 @ 15.0, 54: 01-01 09:00:00 - 01-01 09:30:00 @ 15.0, 55: 01-01 09:30:00 - 01-01 10:00:00 @ 15.0, 56: 01-01 10:00:00 - 01-01 10:30:00 @ 15.0, 57: 01-01 10:30:00 - 01-01 11:00:00 @ 15.0, 58: 01-01 11:00:00 - 01-01 11:30:00 @ 15.0, 59: 01-01 11:30:00 - 01-01 12:00:00 @ 15.0, 60: 01-01 12:00:00 - 01-01 12:30:00 @ 15.0, 61: 01-01 12:30:00 - 01-01 13:00:00 @ 15.0, 62: 01-01 13:00:00 - 01-01 13:30:00 @ 15.0, 63: 01-01 13:30:00 - 01-01 14:00:00 @ 15.0, 64: 01-01 14:00:00 - 01-01 14:30:00 @ 15.0, 65: 01-01 14:30:00 - 01-01 15:00:00 @ 15.0, 66: 01-01 15:00:00 - 01-01 15:30:00 @ 15.0, 67: 01-01 15:30:00 - 01-01 16:00:00 @ 15.0, 68: 01-01 16:00:00 - 01-01 16:30:00 @ 15.0, 69: 01-01 16:30:00 - 01-01 17:00:00 @ 15.0, 70: 01-01 17:00:00 - 01-01 17:30:00 @ 15.0, 71: 01-01 17:30:00 - 01-01 18:00:00 @ 15.0, 72: 01-01 18:00:00 - 01-01 18:30:00 @ 15.0, 73: 01-01 18:30:00 - 01-01 19:00:00 @ 15.0, 74: 01-01 19:00:00 - 01-01 19:30:00 @ 15.0, 75: 01-01 19:30:00 - 01-01 20:00:00 @ 15.0, 76: 01-01 20:00:00 - 01-01 20:30:00 @ 15.0, 77: 01-01 20:30:00 - 01-01 21:00:00 @ 15.0, 78: 01-01 21:00:00 - 01-01 21:30:00 @ 15.0, 79: 01-01 21:30:00 - 01-01 22:00:00 @ 15.0, 80: 01-01 22:00:00 - 01-01 22:30:00 @ 15.0, 81: 01-01 22:30:00 - 01-01 23:00:00 @ 15.0, 82: 01-01 23:00:00 - 01-01 23:30:00 @ 15.0, 83: 01-01 23:30:00 - 01-0 [... cut ...]
2024-12-31 06:00:47.175365: Current SOC 27% is less than Target SOC 100. Grid charging enabled with charge current set to 74.00
2024-12-31 06:00:47.175402: Inverter 0 count register writes 0
2024-12-31 06:00:47.318506: Total inverter register writes now 2261
2024-12-31 06:00:47.503450: Completed run status Charging
2024-12-31 06:00:47.961451: Info: record_status Charging target 27%-100%
2024-12-31 06:00:48.002572: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json

``

springfall2008, 2024-12-31:
I don't think this was every fully fixed, a new version on 'main' is available for testing that hopefully fixes this

chris-y, 2025-01-01:
23:30:12 - rate set
23:30:20 - script ran 

Looks good! 

Happy new year!

closed 2025-01-01 by springfall2008