springfall2008/batpred #1824: predbat reporting incorrect inverter time, plus predbat.status entity is missing

stevedundee2, 2024-12-31:
**Describe the bug**
Predbat is saying there is an error, with the information "Invertor time is 2024-12-30 23:59:59+00:00, Predbat computer time 2024-12-31 11:25:05.280617+00:00, this is -685.12 minutes skewed, Predbat may not function correctly, please fix this by updating your inverter time, checking HA is synchronising with your inverter, or fixing Predbat computer time zone"

Also the predbat.status entity is missing, which I assume is related.

**Expected behaviour**
Predbat not in error state and predbat.status reporting what pedbat is doing e.g. "Demand"

**Predbat version**

1.2.2 (but first noticed the problem this morning in version 1.2.1)
Update 8.8.19

**Environment details**
Core 2024.12.4
Supervisor 2024.12.0
Operating System 14.1
Frontend 20241127.8

- Inverter and battery setup

Givenergy 9.5kW, single hybrid inverter and single battery 

- Standard HAOS installer or Docker
HAOS

- Anything else?

The inverter time and behaviour is fine when I look using the givenergy app. 

The inverter looks fine from HA.

I can control the inverter from predbat (I set manual force charge freeze and the battery stopped supplying power)

So maybe not urgent but I don't know what will happen when the battery should charge tonight. I might turn off predbat until tomorrow.

**Screenshots**
[image]

**Log file**
[predbat.log](https://github.com/user-attachments/files/18281234/predbat.log)

**Predbat debug yaml file**
I get Not Authorised if I try to download the debug yaml. I am admin and on my local network so I don't know why.

gcoan, 2024-12-31:
I'm fairly confident that this is not a Predbat bug, it is that GivTCP stopped working last night and Predbat is complaining that its not received an updated inverter time since midnight

If you look at predbat.status you'll probably see a whole load of REST errors overnight and in the givtcp logs a load of date/time format errors.

The issue and fix is described in this GivTCP bug I raised last night https://github.com/britkat1980/giv_tcp/issues/318

stevedundee2, 2024-12-31:
Hi

Thanks for that pointer. You are right. I have followed the proposed fix (Givenergy.cloud portal -> Remote Control -> click SEND under Inverter Time and Date) and it's all back to normal.

Happy New Year!

Steve

closed 2024-12-31 by stevedundee2