springfall2008/batpred #1835: Errors in the log

Emmoore103, 2025-01-02:
**Describe the bug**
All seems to be working okay. However there are some errors and warnings in the log file. Not sure what these are and not sure if they are having an adverse effect on performance. I want to try and understand these and clear them for the future. I have uploaded the log file, a screenshot of the warnings from the web console and the debug yaml file.

Thank you for your support. 

Eric 

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Predbat version**

8.9.3

**Environment details**

- Givenergy Hybrid with 2 batteries
- Octopus GO intelligent 

**Screenshots**
[image]

**Log file**
[predbat.log](https://github.com/user-attachments/files/18290463/predbat.log)

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to [predbat_debug_12_10_00.yaml.txt](https://github.com/user-attachments/files/18290514/predbat_debug_12_10_00.yaml.txt) and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

Emmoore103, 2025-01-02:
Also an error on the dashboard.

[image]

springfall2008, 2025-01-02:
You probably need to comment out pause start/end time register settings in apps.yaml as you don't have these registers

gcoan, 2025-01-02:
I think he probably needs to comment the battery pause mode line out in apps.yaml as well as the error relates to setting pause_mode, not pause_start/end_time

You say you have a hybrid inverter, what generation and firmware is it on?  If it's 450/451 then this doesn't support pause mode at all.

The warning about gaps in load data can be safely ignored

Emmoore103, 2025-01-03:
Thank you for looking into this for me.
The Hybrid inverter is a GIV-HY5.0 and the software version is
D0.915-A0.915 . I have 2x hi energy 9.5kwh batteries with firmware version
3017.
I can't see the references 450/451. Can you let me know where I would find
this?
KR
Eric

Emmoore103, 2025-01-03:
Should read " 2x Givenergy 9.5kwh batteries " not Hi Energy. Predictive text modified it!

gcoan, 2025-01-03:
> The Hybrid inverter is a GIV-HY5.0 and the software version is D0.915-A0.915

You have a 5 kW hybrid inverter, its whether its a generation 1, 2 or 3 hybrid inverter that's the question

Does it look like this https://www.powerland.co.uk/products/givenergy-hybrid-3-6kw-inverter-gen-1 ?

(the 3.6 and 5 are identical externally)

But the firmware 915-915 makes me think it's a generation 3 hybrid which does have pause mode support which Predbat is trying to use, and it's failing.

Can you have a look in the givtcp log, are there messages about the same time as the predbat errors you see

Emmoore103, 2025-01-03:
Hi

I am sure it is a Gen 2 inverter looking at the pictures of the 3 versions.

This morning I changed to GIVTCP V3 - maybe not wise lol. I also uncommented the lines to do with pause mode. I previously commented them out when advised. However I want to see what happens with them back in.

Now, after giving things time to settle down, Predbat seems to have no major errors at the moment except for the errors show below this message from the Predbat log. Also GiveTCP log seems to be okay.  

I will continue to monitor and I will give it 24hrs before coming to a judgement on whether things are working correctly. 

Thank you for your help and any further suggestions and observations are greatly appreciated.

Kind regards 
Eric

2025-01-03 09:15:02.708656: Warn: octopus_charge_limit is set to a bad value unavailable in apps.yaml, must be a number
2025-01-03 09:15:04.459060: Warn: Historical day 7 has 35 minutes of gap in the data, filled from 15.3 kWh to make new average 15.68 kWh (percent 98%)
2025-01-03 09:15:04.732384: Warn: Inverter 0 unable to read REST data from http://homeassistant.local:6345/readData - REST will be disabled
2025-01-03 09:15:04.778704: Warn: record_status Inverter 0 unable to read REST data from http://homeassistant.local:6345/readData - REST will be disabled
2025-01-03 09:15:04.778978: Warn: Inverter control auto restart already active, waiting...
2025-01-03 09:15:04.804646: Warn: Inverter 0 unable to read REST data from http://homeassistant.local:6345/readData - REST will be disabled
2025-01-03 09:15:04.844584: Warn: record_status Inverter 0 unable to read REST data from http://homeassistant.local:6345/readData - REST will be disabled
2025-01-03 09:20:02.809720: Warn: octopus_charge_limit is set to a bad value unavailable in apps.yaml, must be a number
2025-01-03 09:20:04.535504: Warn: Historical day 7 has 35 minutes of gap in the data, filled from 15.13 kWh to make new average 15.51 kWh (percent 98%)
2025-01-03 09:21:52.844942: Info: record_status Demand
2025-01-03 09:25:02.732430: Warn: octopus_charge_limit is set to a bad value unavailable in apps.yaml, must be a number
2025-01-03 09:25:04.480187: Warn: Historical day 7 has 35 minutes of gap in the data, filled from 14.96 kWh to make new average 15.34 kWh (percent 98%)
2025-01-03 09:25:32.444044: Info: record_status Demand
2025-01-03 09:30:02.550150: Warn: octopus_charge_limit is set to a bad value unavailable in apps.yaml, must be a number
2025-01-03 09:30:04.290422: Warn: Historical day 7 has 35 minutes of gap in the data, filled from 14.68 kWh to make new average 15.05 kWh (percent 98%)

Emmoore103, 2025-01-03:
One thing to add since changing to GivTCP v3 is there is a scaling issue with the solar prediction graph. I am sure I am not generating that much!! It is probably already documented so I will have a search for that problem:

[image]

Emmoore103, 2025-01-03:
Graph sorted with a re-copy of the template. All looking good.

gcoan, 2025-01-03:
> Graph sorted with a re-copy of the template. All looking good.

Your graph issue is probably due to the sensor unit of measure being different when you upgraded to GivTCP v3.  kW vs W.  If you're happy now that's fine, just need to make sure you are using the same units consistently across all your sensors - I prefer kW, others prefer W.

Its better to be on GivTCP 3, that's where all the enhancements and bug fixes will be

gcoan, 2025-01-03:
Looking at each of your logfile messages:

> 2025-01-03 09:15:02.708656: Warn: octopus_charge_limit is set to a bad value unavailable in apps.yaml, must be a number

This indicates an mis-configuration of octopus_charge_limit in your apps.yaml.  Do you have an EV and are you on IOG?  Can you share what this is configured to in apps.yaml

> 2025-01-03 09:15:04.459060: Warn: Historical day 7 has 35 minutes of gap in the data, filled from 15.3 kWh to make new average 15.68 kWh (percent 98%)

This can safely be ignored, its saying that there was a small gap in your historical load and predbat has 'filled in the gap' by interpolation.  I get it quite often as well.

> 2025-01-03 09:15:04.732384: Warn: Inverter 0 unable to read REST data from http://homeassistant.local:6345/readData - REST will be disabled
> 2025-01-03 09:15:04.778704: Warn: record_status Inverter 0 unable to read REST data from http://homeassistant.local:6345/readData - REST will be disabled

This is strange, Predbat should be able to talk to GivTCP by REST.  Might be worth restarting GivTCP and Predbat to see if it clears this.

> 2025-01-03 09:15:04.778978: Warn: Inverter control auto restart already active, waiting...

Again this is strange. When you upgraded to GivTCP v3 did you update the predbat apps.yaml auto-restart code to use the different slug id for GivTCP v3?  I'm wondering if you didn't and Predbat has restarted GivTCP 2 so 2 and 3 are running alongside one another which is causing the above REST error?

Make sure GivTCP v2 is stopped and isn't set to auto start on boot.

Any errors in the GivTCP log?

Emmoore103, 2025-01-03:
Hi

I have IOG and EV. 

Just noticed there was missing syntax around the sensor names ('). Maybe something I did a long time ago when setting all this up qd I never noticed. I have added them in. So now it is as below (I have changed the octupus_ready_time sensor reference after an update to the Octupus integration as a warning has come up to say the existing one is being depreciated - if that makes sense - just in case I have hashed out the old name). 

  octopus_intelligent_slot: 're:(binary_sensor.octopus_intelligent_slot|re:binary_sensor.octopus_energy_intelligent_dispatching)'
  #   octopus_ready_time: 're:(time.octopus_energy_([0-9a-z_]+|)_intelligent_target_time)'
  octopus_ready_time: 're:(select.octopus_energy_([0-9a-z_]+|)_intelligent_target_time)'
  octopus_charge_limit: 're:(number.octopus_energy([0-9a-z_]+|)_intelligent_charge_target)'

I think that solves that issue.

I will check your other points shortly.

Emmoore103, 2025-01-03:
I think restart is now correct. I have compared and corrected entries compared to the official template.

Hopefully all will work now. I will check the log files through the day.

This is the apps.yaml file:

pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
   - '£'
   - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  geserial: 're:sensor.givtcp_(.+)_soc_kwh'
  geserial2: 're:sensor.givtcp2_(.+)_soc_kwh'

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.givtcp_{geserial}_load_energy_today_kwh
  import_today:
    - sensor.givtcp_{geserial}_import_energy_today_kwh
  export_today:
    - sensor.givtcp_{geserial}_export_energy_today_kwh
  pv_today:
    - sensor.givtcp_{geserial}_pv_energy_today_kwh

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
  #
  # If you enable ge_cloud_data then the load/import and export data will be fetches from the GE cloud instead of from GivTCP sensors
  # this is usually less efficient and of course prone to internet downtime, but could be useful if you lost your GivTCP data
  # Set the serial to the inverter serial to pull the data from and the key to your API key
  # When this is set load_today, import_today and export_today are not used
  #
  ge_cloud_data: False
  ge_cloud_serial: '{geserial}'
  ge_cloud_key: 'xxxx'
  #
  # Controls/status - must by 1 per inverter
  #
  inverter_type:
    - "GE"
[... 344 lines cut ...]
  #iboost_energy_scaling: 1.0
  # Gas rates for comparison
  #metric_octopus_gas: 're:(sensor.(octopus_energy_|)gas_[0-9a-z]+_[0-9a-z]+_current_rate)'

  # Nordpool market energy rates
  #futurerate_url: 'https://www.nordpoolgroup.com/api/marketdata/page/325?currency=GBP'
  #futurerate_adjust_import: True
  #futurerate_adjust_export: False
  #futurerate_peak_start: "16:00:00"
  #futurerate_peak_end: "19:00:00"
  #futurerate_peak_premium_import: 14
  #futurerate_peak_premium_export: 6.5

  # Watch list, a list of sensors to watch for changes and then update the plan if they change
  # This is useful for things like the Octopus Intelligent Slot sensor so that the plan update as soon as you plugin in
  # Only uncomment the items you actually have set up above in apps.yaml, of course you can add your own as well
  # Note those using +[] are lists that are appended to this list, whereas {} items are single items only
  #watch_list:
  #  - '{octopus_intelligent_slot}'
  #  - '{octopus_ready_time}'
  #  - '{octopus_charge_limit}'
  #  - '{octopus_saving_session}'
  #  - '+[car_charging_planned]'
  #  - '+[car_charging_soc]'
  #  - '{car_charging_now}'

Emmoore103, 2025-01-03:
I think restart is now correct. I have compared and corrected entries compared to the official template on GITHUB page.

Hopefully all will work now. I will check the log files through the day.

I attach the apps.yaml for reference.

[apps.yaml.txt](https://github.com/user-attachments/files/18300121/apps.yaml.txt)

Emmoore103, 2025-01-03:
Okay

I am happy that I have got rid of the errors in Predbat. In GivTCP log it does error as follows (see last line):

2025-01-03 14:15:14,264 - startup     -  [INFO    ] - ==================== STARTING GivTCP==========================
2025-01-03 14:15:14,305 - startup     -  [INFO    ] - Network Found: 192.168.178.1/24
2025-01-03 14:15:16,307 - startup     -  [INFO    ] - Searching for Inverters
2025-01-03 14:15:16,307 - startup     -  [INFO    ] - Scanning network for GivEnergy Devices...
2025-01-03 14:15:18,883 - startup     -  [INFO    ] - INV- Scanning network (1):192.168.178.1/24
2025-01-03 14:15:26,119 - client      -  [INFO    ] - Connection established to 192.168.178.22:8899
2025-01-03 14:15:26,119 - client      -  [INFO    ] - Detecting plant
2025-01-03 14:15:27,361 - client      -  [INFO    ] - Plant Detected
2025-01-03 14:15:30,123 - startup     -  [INFO    ] - Inverter ED2301G466 which is a Gen2 - Hybrid with 2 batteries has been found at: 192.168.178.22
2025-01-03 14:15:30,137 - startup     -  [INFO    ] - Setting up invertor: 1
2025-01-03 14:15:30,270 - startup     -  [INFO    ] - ==============================================================
2025-01-03 14:15:30,271 - startup     -  [INFO    ] - ====             Web Gui Config is at                     ====
2025-01-03 14:15:30,272 - startup     -  [INFO    ] - ====     http://192.168.178.70:8099/config.html             ====
2025-01-03 14:15:30,273 - startup     -  [INFO    ] - ==============================================================
2025-01-03 14:15:30,275 - startup     -  [INFO    ] - Running Invertor 1 (ED2301G466) read loop every 15/120s
2025-01-03 14:15:30,291 - startup     -  [INFO    ] - Serving Web Dashboard from port 3000
[2025-01-03 14:15:30 +0000] [70] [INFO] Starting gunicorn 23.0.0
[2025-01-03 14:15:30 +0000] [70] [INFO] Listening at: http://0.0.0.0:6350 (70)
[2025-01-03 14:15:30 +0000] [70] [INFO] Using worker: sync
[2025-01-03 14:15:30 +0000] [87] [INFO] Booting worker with pid: 87
[2025-01-03 14:15:31 +0000] [81] [INFO] Starting gunicorn 23.0.0
[2025-01-03 14:15:31 +0000] [81] [INFO] Listening at: http://0.0.0.0:6345 (81)
[2025-01-03 14:15:31 +0000] [81] [INFO] Using worker: sync
[2025-01-03 14:15:31 +0000] [88] [INFO] Booting worker with pid: 88
[2025-01-03 14:15:31 +0000] [90] [INFO] Booting worker with pid: 90
[2025-01-03 14:15:31 +0000] [89] [INFO] Booting worker with pid: 89
2025-01-03 14:15:33,559 - GivTCP - read        -  [INFO    ] - Starting watch_plant loop...
2025-01-03 14:15:33,560 - GivTCP - GivLUT      -  [CRITICAL] - Opening Modbus Connection to 192.168.178.22
2025-01-03 14:15:33,564 - GivTCP - read        -  [CRITICAL] - Detecting inverter characteristics...
2025-01-03 14:15:41,625 - GivTCP - read        -  [INFO    ] - Publishing Home Assistant Discovery messages
2025-01-03 14:15:48,677 - GivTCP - HA_Discovery -  [ERROR   ] - Error connecting to MQTT Broker: ('RuntimeError', 'HA_Discovery.py', 399)

Emmoore103, 2025-01-03:
I believe GivTCP is functioning correctly now, as I can see historical data recorded against the entities on the MQTT broker.

It seems to be working as expected, but I’ll let it run for a while to confirm everything meets my expectations.

Thank you so much, Geoffrey Coan and Trefor Southwell, for your assistance. I truly appreciate your time and expertise. Without the support of knowledgeable individuals like you, I would have been completely stuck trying to implement these complex systems. There’s still so much for me to learn, but your help has made a significant difference.

If there’s nothing further to address, I’ll go ahead and mark this as completed.

gcoan, 2025-01-03:
I've looked at your apps.yaml and the auto restart is pointing to the GivTCP v2 slug id (the template has the v3 slug id in it)

See the documentation on auto-restart https://springfall2008.github.io/batpred/apps-yaml/#automatic-restarts 

So when predbat detects a givtcp issue it will restart givtcp v2, i.e. start it alongside v3.  Not good ...

your givtcp_rest should work, but there is a tip to replace the homeassistant.local with the IP address as this means you don't have lots of DNS lookups https://springfall2008.github.io/batpred/apps-yaml/#rest-interface-inverter-control

as you only have one inverter you should comment out all the lines with geserial2 in them and comment them out

you can if you want comment the inverter_clock_skew out as you're not using this

I see you changed the octopus_ready_time sensor, the predbat template still uses the time.octopus... sensor which will be removed I see https://bottlecapdave.github.io/HomeAssistant-OctopusEnergy/entities/intelligent/#target-time so I'll need to update the predbat templates, thanks

Emmoore103, 2025-01-03:
Super. Thank you for these clear instructions. The slug id has been
corrected and I will carry out the rest of the advised changes. Have a good
weekend.

closed 2026-08-28 by chalfontchubby