springfall2008/batpred #1836: My installation preferring grid sourced electricity, rather than last nights cheaper battery storage electricity

adibbins, 2025-01-02:
**Describe the bug**
My system, frequently (every 7 or 8 days or so) decides to use grid sourced electricity, despite having charged the batteries to 100% full during last nights cheap rate period. My system seems to work perfectly on most other days.

**Expected behaviour**
Don't use expsensive grid provided electricity, when there's battery power available

**Predbat version**

v8.9.3

**Environment details**

- Inverter and battery setup
- Solis hybrid inverter RHI-5K-48ES-5G, 6x Pylontech US3000c = 21Kw
- Standard HAOS installer or Docker
- Standard HAOS latest
- HAOS, Solax Inverter Modbus and anything else, all versions are always kept up to date
- 
[older predbat.zip](https://github.com/user-attachments/files/18291520/older.predbat.zip)
[predbat.zip](https://github.com/user-attachments/files/18291521/predbat.zip)

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.
[image]

[predbat (1).zip](https://github.com/user-attachments/files/18291594/predbat.1.zip)
[predbat_plan.zip](https://github.com/user-attachments/files/18291595/predbat_plan.zip)
[predbat_debug.zip](https://github.com/user-attachments/files/18291596/predbat_debug.zip)
[apps.zip](https://github.com/user-attachments/files/18291597/apps.zip)
**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

I hope I've upload all thats required, I'm not sure about the predbat_debug_xx_xx_xx.yaml files as despite enabling debug there's no debug file with the current timestamp.

Thanks in advance for the help.
Regards
Andy

gcoan, 2025-01-02:
Can you explain the bit of the plan that doesn't look right to you?

In the plan (the cost column cut off), but up to 13:00 the solar forecast is > than your load forecast so your costs are going down as excess solar is exported

Then from 13:00 the load forecast is > than solar and the battery is shown as discharging (down arrow in state column) and SoC % is dropping

Cost is flat (sideways arrow) apart from 14:00 when there is a bit of cost PREDICTED, this is presumably because your load history shows that your load will exceed your inverter throughput and there will be a bit of grid import as a result

From what I see, it looks OK to me

adibbins, 2025-01-02:
Hi Geoffrey,

Here's an unclipped photo of the plan, from this evening
[image: image.png]

But here;s the thing, the battery SOC is still at 100 or 99% is hasn't
decreased at all today and surely we'd expect it to?
After midnight tonight, the battery won't charge because it doesn't need
to, because its at 100% and tomorrow things will work normally, the house
load will be consumed from the battery and the batteries SOC will decrease
through the day,

Have I configured something wrongly?

Thx
Andy

gcoan, 2025-01-03:
Hi Andy

Your photo hasn't come through unfortunately - did 
you reply by email (that can strip out attachments)?

Actually I found the plan attached as a zip to the original github issue.

Yes your battery remaining full doesn't sound right.

According to the predbat plan you attached it looks like predbat planned to let the battery discharge and it didn't which points to a predbat execution issue or a potential givtcp or inverter issue.

What's in the predbat status for today, was it on demand for most of the day?

And the predbat logfile, are there any errors?

I would start with the givtcp logfile, is it communicating with the inverter ok, one sign of this is that the invertor_time entity is being updated regularly (look in the logfile and you should see this).

adibbins, 2025-01-03:
Hi Geoffrey,

So, everything is working perfectly today.
There was next to no charge from cheap rate electricity overnight because the batteries were basically full.
And, today I am consuming the stored electricity from the battieries and SOC is decreasing as I consume electricity.

I've attached todays plan and the logs
[predbat.zip](https://github.com/user-attachments/files/18299185/predbat.zip)
[predbat_plan.zip](https://github.com/user-attachments/files/18299187/predbat_plan.zip)

If past performance is another thing to go by, this will work perfectly until about the 10th or 11th Jan and then the problem will repeat for a day and then it'll be perfect again.

I was wondering if this was load history related so you may see me playing with the history days in the apps.yaml file. Am of course happy to be advised as to the best settings.

FYI: No GivEnergy inverter but a Solis hybrid inverter.

Thx
Andy

gcoan, 2025-01-03:
Hi Andy

Sorry about my mis-expectation about you using GivTCP, force of habit !

If this is a repeated pattern then probably worth you taking a screen shot of the predbat plan every day and then looking at how the SoC tracked to the plan.  Load and PV will of course change but should give you an idea of whether there's an underlying issue with execution not tracking the plan

Also have you installed the Predbat Apex charts (see the output data part of the documentation).  There's a chart called the calibration chart that shows how well the plan tracks the prediction (from 1 hour, 8 hour, 12 hours ago).  Worth looking at that.

I quickly looked at your predbat logfile and saw that Historical days is set to [1, 4, 5, 6, 7, 11].  

Have a look at the documentation https://springfall2008.github.io/batpred/apps-yaml/#days_previous and https://springfall2008.github.io/batpred/apps-yaml/#understanding-how-days_previous-works

It's generally not recommended to use days previous of 1 - see explanation.  I have mine set to 2, 3, 4, 5, 6, 7, 8 to give an average of the previous week, and I set switch.predbat_load_filter_modal to on (expert mode switch https://springfall2008.github.io/batpred/customisation/#historical-load-data) to exclude the lowest day's consumption.

adibbins, 2025-01-04:
Hi Geoffrey,

I do have all the charts installed.

The Historical days setting, including day 1, missing 2 and 3 and 7,8,9,10 were changes I made to try and fix the problem, my setting before this issue was 2,3,4,5,6,7,8, to 14, but it made no effect on the problem.

I;m wondering if predbat is performimg correctly and is sending the commands to my inverter but my inverter is ignoring them. I;ve tried intervening and sending a "battery awaken" state command to my inverter, this works for a few minutes and my house battery current rises from zero to say 12A but after a fews minutes it;s back to zero current being taken from the battery and the inverter staus is changed back to "self Use"

I'm also wondering if my solis inverter is due a firmware update, these updates would be delivered by the solis cloud and I can't be connected to solis cloud and predbat at the same time via the datalogger I have (its a limitation of the datalogger I believe), if someone out there has a solis inverter what firmware version are they running (front panel, my software version says 470044)?

thanks for your help.
Andy

springfall2008, 2025-01-05:
I can't see anything wrong in the logfile, not sure if anyone else spotted an issue?

Its possible the bug I fixed today might be impacting this so I'd suggest updating and seeing. Try to capture a log around a failure time if possible.

adibbins, 2025-01-16:
Hi Geoffrey, Trefor,

I think we can close my issue as since upgrading to v8.10.1, v8.10.2 and v8.11 things are performing as expected.
Thanks for your good work.
Andrew

closed 2025-01-16 by adibbins