springfall2008/batpred #1837: Factoring in uncertainty

WillCalderwood, 2025-01-02:
**Is your feature request related to a problem? Please describe.**
It's better to act on what we know now rather than what we believe might happen in the future. In the example below, there's no advantage that I can see in delaying the battery charging to the end of the cheap period, but there is a disadvantage, as there's a good chance that Octopus will change the cheap slots. If the future is guaranteed, then there's no cost, but the future is never guaranteed, so factoring on an uncertainty cost as we predict the future might be a good way to handle this.
[image]

**Describe the solution you'd like**
If we know we can do something for the same predicted cost now as in the future, then do it now.

**Describe alternatives you've considered**
Increasing import costs for time periods that I want to discourage charging, but this feel inelegant.

gcoan, 2025-01-03:
you could try turning switch.calculate_export_high_import on, this could help because the default false value for this switch results in the later export slots being chosen

https://springfall2008.github.io/batpred/customisation/#calculation-options

WillCalderwood, 2025-01-03:
I didn't see this until it was too late. Reading the docs, the calculate_export_high_import setting changes export slots, not charge slots. Does it also affect charge slots?

springfall2008, 2025-01-04:
Try turning on combine_charge

WillCalderwood, 2025-01-05:
@springfall2008 Next time I notice it trying to charge at the end of a cheap period I'll make the change and see if it has an effect.

closed 2026-08-28 by chalfontchubby