springfall2008/batpred #1840: Prebat suddenly stopped updating the plan last night.

jrwm47, 2025-01-04:
**Describe the bug**
After running continuously for many weeks and generating and implementing good-looking plans, _predbat suddenly stopped producing plans_ at around 2100 last night. I only noticed a few minutes ago (18 hours after the event) that the current plan had not updated since last night.

HA, my home network, and Internet are all running well.

I RESTARTED just the predbat Add-On and it now appears to be running normally and producing regular updates to the plan.

_But why did it stop?_

**Expected behaviour**
I have come to expect predbat to run continuously without any intervention.

**Predbat version**
v1.2.2 according to Settings Add-On
v8.9.3 according to Log.
Auto-Update is ON
Watchdog is ON

**Environment details**
GE AIO without Gateway
Standard HAOS install on dedicated RPi4
GivTCP v3

**Screenshots**
Unfortunately, I failed to take screenshots before re-starting. However, I did notice that the most recent timelog at the top of the plan was around 9pm yesterday. Strangely, the battery SOC graph appeared normal with the time cursor at the correct time but I think this entity comes from GivTCP. 

**Log file**
Extract from log file at the time of the issue:
...
2025-01-03 20:50:41.437094: Info: record_status Freeze charging target 100%
2025-01-03 20:55:07.860011: Info: record_status Freeze charging target 100%
2025-01-03 21:00:00.568517: Error: HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /core/api/states (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fb37ed180>: Failed to establish a new connection: [Errno 111] Connection refused'))
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connection.py", line 198, in _new_conn
    sock = connection.create_connection(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/util/connection.py", line 85, in create_connection
    raise err
  File "/usr/local/lib/python3.10/dist-packages/urllib3/util/connection.py", line 73, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py", line 787, in urlopen
    response = self._make_request(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py", line 493, in _make_request
    conn.request(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connection.py", line 445, in request
    self.endheaders()
  File "/usr/lib/python3.10/http/client.py", line 1278, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.10/http/client.py", line 1038, in _send_output
    self.send(msg)
  File "/usr/lib/python3.10/http/client.py", line 976, in send
    self.connect()
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connection.py", line 276, in connect
    self.sock = self._new_conn()
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connection.py", line 213, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0x7fb37ed180>: Failed to establish a new connection: [Errno 111] Connection refused
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/requests/adapters.py", line 667, in send
    resp = conn.urlopen(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/connectionpool.py", line 841, in urlopen
    retries = retries.increment(
  File "/usr/local/lib/python3.10/dist-packages/urllib3/util/retry.py", line 519, in increment
    raise MaxRetryError(_pool, url, reason) from reason  # type: ignore[arg-type]
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /core/api/states (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fb37ed180>: Failed to establish a new connection: [Errno 111] Connection refused'))
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/config/hass.py", line 184, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 980, in run_time_loop
    self.ha_interface.update_states()
  File "/config/ha.py", line 400, in update_states
    res = self.api_call("/api/states")
  File "/config/ha.py", line 686, in api_call
    response = requests.get(url, headers=headers, timeout=TIMEOUT)
  File "/usr/local/lib/python3.10/dist-packages/requests/api.py", line 73, in get
    return request("get", url, params=params, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/requests/api.py", line 59, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/requests/sessions.py", line 589, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.10/dist-packages/requests/sessions.py", line 703, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/requests/adapters.py", line 700, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /core/api/states (Caused by 

** log, yaml files, and screen shots available if that helps. **

springfall2008, 2025-01-04:
This is a communication error with Home Assistant

The way to avoid this problem if it occurs is normally to set the API Key in the apps.yaml as sometimes I see the add-on generated key expiring.

jrwm47, 2025-01-04:
Like this?

# If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
   ha_url: 'http://192.168.86.36:8123'
   ha_key: 'eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiIwNjM5OTZkYzc0ODI0NzBmYmRjYTM3M2Y3NWM2N2Y2NSIsImlhdCI6MTczNjAxOTk4NiwiZXhwIjoyMDUxMzc5OTg2fQ.U8M5TqwX5839VtjqV1EVBlbdw6m35G3W5N8OkVoVLSE'

gcoan, 2025-01-04:
yes

Amigos2222, 2025-01-04:
The exact same thing happened to me at the same time yesterday. I look at home assistant supervisor logs and it seemed to auto update itself. Once it restarted predbat wasnt able to reconnect until i restarted predbat. 

Some googling showed me its possible to use a command to stop the supervisor auto updating but is there a way to get predbat to auto restart?

gcoan, 2025-01-04:
> is there a way to get predbat to auto restart?

there are some automations I wrote in the documentation to detect predbat, givtcp or mqtt not working and to restart them 

https://springfall2008.github.io/batpred/output-data/#automated-monitoring-that-predbat-and-givtcp-are-running-ok

jrwm47, 2025-01-04:
Thanks. I'll try both things in the morning. Julian

nargotik, 2025-01-05:
Mine stoped working yesterday also having these errors on log

975 | raise WSServerHandshake Error(
-- | --
969 | 2025-01-05 01:08:20.445427: Error: Traceback (most recent call last):
968 | 2025-01-05 01:08:20.444956: Error: Web Socket exception in startup: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')
966 | 2025-01-05 01:08:15.432354: Warn: Web Socket closed, will try to reconnect in 5 seconds
964 | aiohttp.client_exceptions.W SServerHandshakeError: 502, message='Invalid response status', url=URL('http://supervisor/core/api/websocket')

<br class="Apple-interchange-newline">

nargotik, 2025-01-05:
it has started at 21:15 yesterday

025-01-03 21:15:14.272751: Base charge window [  ]
2025-01-03 21:15:14.272792: Base export window [  ]
2025-01-03 21:15:14.272838: Inverter 0 Charge window will be: 2025-01-03 22:00:00+00:00 - 2025-01-03 23:59:00+00:00 - current soc 3 target 100
2025-01-03 21:15:14.272877: Inverter 0 disabled charge window while waiting for schedule (now 01-03 21:15:00 target set_window_minutes 30 charge start time 01-03 22:00:00)
2025-01-03 21:15:14.272920: Adjust idle time, charge 00:00:00-00:00:00 discharge 00:00:00-00:00:00
2025-01-03 21:15:14.273027: Adjust demand (idle) time computed is 06:00:00-23:59:00
2025-01-03 21:15:14.273112: Inverter 0 adjust target soc for export to 0.0% based on requested all inverter soc 0%
2025-01-03 21:15:14.273138: Inverter 0 Current Target SOC is 4%, already at target
2025-01-03 21:15:14.273164: Setting charging SOC to 0 as we are not within the charge window and inverter doesn't support target soc
2025-01-03 21:15:14.273201: Inverter 0 Skipping service charge_stop_service domain charge with data {'device_id': 'xxx'} extra_data {} as already called
2025-01-03 21:15:14.273215: Inverter 0 count register writes 0
2025-01-03 21:15:14.313937: Total inverter register writes now 0
2025-01-03 21:15:14.361155: Completed run status Demand
2025-01-03 21:15:14.372390: Info: record_status Demand
2025-01-03 21:15:14.383501: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2025-01-03 21:20:00.822619: Error: HTTPConnectionPool(host='supervisor', port=80): Max retries exceeded with url: /core/api/states (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fd5c9797be0>: Failed to establish a new connection: [Errno 111] Connection refused'))
2025-01-03 21:20:03.779410: Warn: Web Socket closed, will try to reconnect in 5 seconds
2025-01-03 21:20:08.782643: Info: Start socket for url http://supervisor/core/api/websocket

springfall2008, 2025-01-05:
Setting the HA Key should help (don't set HA URL if you are using an add on)

gcoan, 2025-01-06:
I've added a bit more to the documentation advising about setting ha_url

PhilPJL, 2025-01-16:
Is this discussion the same as the error I'm seeing?

In the desktop app I'm getting:
[image]

But the plan shows up fine in the mobile app.

gcoan, 2025-01-16:
> Is this discussion the same as the error I'm seeing?
> 
> In the desktop app I'm getting:

No, this is a browser error, it can't load the html template card for some reason and so can't render the predbat plan on the screen.  Try clearing your browser cache and it should work OK

closed 2025-01-16 by gcoan