springfall2008/batpred #1843: Multiple short exports are planned throughout the day for no clear reason

bagfatnick, 2025-01-05:
**Describe the bug**
Looking at the plan for tomorrow, short exports are planned despite minimal Solar generation, battery is predicted to be at 4%, and import cost > export cost. 

Never noticed this issue before, and have been running Predbat since April '24.

**Expected behaviour**
No export during this period. 

**Predbat version**

8.9.3

**Environment details**

- GivEnergy Hybrid 5kw, 2x 5kwh batteries. 
- HA on python Virtual Env, Predbat in standalone docker container. Control to inverter via GivTCP v3
- IOG tariff. 
- Expert mode enabled. all config at default values, except calculate export on charge (true) and car charging from battery (true).

**Screenshots**
[image]

**Log file**
[predbat.log](https://github.com/user-attachments/files/18308643/predbat.log)

**Predbat debug yaml file**
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/18308644/predbat_debug.yaml.txt)

cyberkryten, 2025-01-05:
I'm seeing identical behaviour after upgrading, plus some other issues I'm tracking.  However, I came from I came from an old version (8.7.2) and have changed from AppDaemon to direct running plus have also upgraded HA and several related integrations (including SolarEdgeModbus, Octopus and SolCast) so I cannot be sure all the odd Predbat behaviour is purely down to this integration (so I'll not add debug here for now whilst I check all those are operating correctly)

I am also seeing that there's something off with the plan calculations as it is constantly planning short exports in the same way your plan shows.  

It is also set to export once the Battery hits the planned minimum - in your example on Sun 16:00 there is no solar and battery is 4% but then it plans lots of exports and a solid export slot from21:00 (battery at 4%) until 23:00 (battery still at 4%) and for some reason thinks it is exporting the whole time.  

I'm seeing identical issues here too as my plan shows min level hit and then several hours of profitable export without the battery level changing.  However, on my system, it is also failing to command the inverter regularly to keep it in the intended Charge/Export state (SolarEdge which resets to Default: Maximise Self Consumption after 3600 seconds), it always used to rewrite the states every plan run (which kept the 3600 refreshed) but now it just expires and then stops exporting/charging, so doesn't hold in the Exporting state for the four hours planned anyway.

springfall2008, 2025-01-05:
That's bonkers, I'm debugging this now

springfall2008, 2025-01-05:
Fix on main for testing, it seems the model didn't reset the discharge rate correctly after it was turned off

cyberkryten, 2025-01-05:
Wow, so quick!  

The odd plan was executed overnight - constant switches between Discharge, Charge and Self Consumption, often for seconds at a time:

```
Solaredgemodbus I1 Storage Command Mode  changed to Charge from Solar Power and Grid
05:00:50 - 6 hours ago

Solaredgemodbus I1 Storage Command Mode  changed to Maximize Self Consumption triggered by action Select: Select
05:00:38 - 6 hours ago - Supervisor

Solaredgemodbus I1 Storage Command Mode  changed to Discharge to Maximize Export
04:51:23 - 6 hours ago

Solaredgemodbus I1 Storage Command Mode  changed to Maximize Self Consumption triggered by action Select: Select
04:51:12 - 6 hours ago - Supervisor

Solaredgemodbus I1 Storage Command Mode  changed to Charge from Solar Power and Grid
04:47:07 - 6 hours ago

Solaredgemodbus I1 Storage Command Mode  changed to Maximize Self Consumption triggered by action Select: Select
04:46:56 - 6 hours ago - Supervisor

Solaredgemodbus I1 Storage Command Mode  changed to Discharge to Maximize Export
04:46:23 - 6 hours ago

Solaredgemodbus I1 Storage Command Mode  changed to Maximize Self Consumption triggered by action Select: Select
04:46:12 - 6 hours ago - Supervisor

Solaredgemodbus I1 Storage Command Mode  changed to Charge from Solar Power and Grid
04:30:50 - 7 hours ago

Solaredgemodbus I1 Storage Command Mode  changed to Maximize Self Consumption triggered by action Select: Select
04:30:39 - 7 hours ago - Supervisor

Solaredgemodbus I1 Storage Command Mode  changed to Discharge to Maximize Export
04:21:23 - 7 hours ago

Solaredgemodbus I1 Storage Command Mode  changed to Maximize Self Consumption triggered by action Select: Select
04:21:12 - 7 hours ago - Supervisor

Solaredgemodbus I1 Storage Command Mode  changed to Charge from Solar Power and Grid triggered by action Select: Select
04:16:57 - 7 hours ago - Supervisor

Solaredgemodbus I1 Storage Command Mode  changed to Discharge to Maximize Export
04:16:20 - 7 hours ago
```

I've just installed 8.10.0 and now the plan looks far better and much more what I'd expect, plus it doesn't get to 5% and think it can still export more.

I'll monitor today and update tonight after the charge/discharge cycles have run.

bagfatnick, 2025-01-05:
I’ve just updated to main and it’s fixed the issue! 
Thanks for being so responsive @springfall2008

springfall2008, 2025-01-05:
Now fixed in v8.10.0

closed 2025-01-05 by springfall2008