springfall2008/batpred #1846: PB Predicted load graph not trending correctly

Eeebygum23, 2025-01-05:
**Describe the bug**
The 'In Day Adjustment' graph is not reading correctly. For example if the current load on the house was 15kwh, the current reading on the graph shows significatly less e.g. 4kwh. Also, when the 'Load energy actual (filtered)' is interrogated, it resets at 00:00hrs to a much higher reading than it was previously at before the midnight reset. I would have expected this reading to have reset to 0kwh at midnight.

- Inverter and battery setup
- Standard HAOS installer using Home Assistant Yellow
- PB operating system up to date.

[image]
[image]
[predbat_debug.txt](https://github.com/user-attachments/files/18310426/predbat_debug.txt)

johnwb87, 2025-01-05:
I asked this question a while back.
https://github.com/springfall2008/batpred/issues/1434

When you graph the load energy actual it is giving you the end of the day value. So actual, actual and the remaining prediction for that day.

The time series is in the attributes of the entity

gcoan, 2025-01-06:
> When you graph the load energy actual it is giving you the end of the day value. So actual, actual and the remaining prediction for that day.

Its not just an end of day value, its end of plan value, so depending on your plan duration this can be a couple of days of load predictions

chalfontchubby, 2026-08-29:
This feels like a resolved issue - marking for closure unless there is a reply saying to keep it open.

chalfontchubby, 2026-09-09:
Closing as flagged - no response to the note above, so we're assuming this is no longer live.

If you're still hitting this, please reopen or open a fresh issue with a current version number and a `predbat_debug_*.yaml` captured while it's happening; we can replay that directly against current code.

_Posted by Claude on Rik's behalf._

closed 2026-09-09 by chalfontchubby