springfall2008/batpred #1850: Predbat crashed with Exception Error

WilliamGillin, 2025-01-05:
**Describe the bug**
Have been running Predbat for a couple of months and it's been great. Today I noticed that it stopped updatuing the plan at 12:35 and was showing an exception error.
I've tried completely removing Predbat, GivTCP V3. I also cleared the MQTT cache of all GivTCP messages and removed the existing GivTCP devices so that I was starting with a clean system. Unfortunately I get exactly the same error.
I can't tell if this is an issue with Predbat, GivTCP or my battery!

**Expected behaviour**
Predbat generating plans as usual!

**Predbat version**
Add-on V1.2.2
V8.10.0

**Environment details**

- GE AIO and Gateway
- Standard HAOS installed on virtual machine on synology NAS
- GivTCP3
- Mosquito MQTT broker

**Screenshots**

**Log file**
Bootstrap Predbat
Startup
Predbat files are installed correctly for version v8.10.0
**** Starting Standalone Predbat ****
2025-01-05 17:48:45.920796: Loading apps.yaml
2025-01-05 17:48:45.996942: Info: Connected to Home Assistant at http://supervisor/core
2025-01-05 17:48:46.007551: Info: Add-on slug is 6adb4f0d_predbat
2025-01-05 17:48:46.007634: Creating task: <coroutine object HAInterface.socketLoop at 0x7f7737a26570>
2025-01-05 17:48:46.008030: Info: Web Socket task started
2025-01-05 17:48:46.008109: Creating task: <coroutine object WebInterface.start at 0x7f7737a265e0>
2025-01-05 17:48:46.008631: Info: Start socket for url http://supervisor/core/api/websocket
Web interface started
2025-01-05 17:48:46.026236: Info: Web Socket active
2025-01-05 17:48:46.057748: Warn: Regular expression argument: geserial2 unable to match re:sensor.givtcp2_(.+)_soc_kwh, now will disable
2025-01-05 17:48:46.062077: Warn: Regular expression argument: car_charging_energy unable to match re:(sensor.myenergi_zappi_[0-9a-z]+_charge_added_session|sensor.wallbox_portal_added_energy), now will disable
2025-01-05 17:48:46.063552: Warn: Regular argument car_charging_planned expression re:(sensor.wallbox_portal_status_description|sensor.myenergi_zappi_[0-9a-z]+_plug_status) failed to match - disabling this item
2025-01-05 17:48:46.064783: Warn: Regular expression argument: carbon_intensity unable to match re:(sensor.carbon_intensity_uk), now will disable
Watching ['./inverter.py', './utils.py', './userinterface.py', './octopus.py', './output.py', './predheat.py', './apps.yaml', './futurerate.py', './hass.py', './config.py', './predbat.py', './ui.py', './unit_test.py', './solcast.py', './ha.py', './web.py', './execute.py', './gecloud.py', './fetch.py', './plan.py', './download.py', './prediction.py'] for changes
2025-01-05 17:48:49.893645: Error: Unable to fetch history for sensor.givtcp_battery_stack_1_stack_load_energy_today_kwh
2025-01-05 17:48:49.910492: Warn: record_status Error: Unable to fetch history from sensor.givtcp_battery_stack_1_stack_load_energy_today_kwh
2025-01-05 17:48:49.910597: Error: Exception raised 
2025-01-05 17:48:49.911548: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 935, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 559, in update_pred
    self.fetch_sensor_data()
  File "/config/fetch.py", line 775, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
  File "/config/fetch.py", line 407, in minute_data_load
    raise ValueError
ValueError
2025-01-05 17:48:49.928273: Info: record_status Error: Exception raised 
2025-01-05 17:48:49.928389: Error: 
Traceback (most recent call last):
  File "/config/hass.py", line 184, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 941, in update_time_loop
    raise e
  File "/config/predbat.py", line 935, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 559, in update_pred
    self.fetch_sensor_data()
  File "/config/fetch.py", line 775, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
  File "/config/fetch.py", line 407, in minute_data_load
    raise ValueError
ValueError
2025-01-05 17:49:04.335163: Error: Unable to fetch history for sensor.givtcp_battery_stack_1_stack_load_energy_today_kwh
[... 64 lines cut ...]
    self.fetch_sensor_data()
  File "/config/fetch.py", line 775, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
  File "/config/fetch.py", line 407, in minute_data_load
    raise ValueError
ValueError
2025-01-05 17:55:00.271101: Info: record_status Error: Exception raised 
2025-01-05 17:55:00.271242: Error: 
Traceback (most recent call last):
  File "/config/hass.py", line 184, in timer_tick
    item["callback"](None)
  File "/config/predbat.py", line 994, in run_time_loop
    raise e
  File "/config/predbat.py", line 989, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 559, in update_pred
    self.fetch_sensor_data()
  File "/config/fetch.py", line 775, in fetch_sensor_data
    self.load_minutes, self.load_minutes_age = self.minute_data_load(self.now_utc, "load_today", self.max_days_previous, required_unit="kWh")
  File "/config/fetch.py", line 407, in minute_data_load
    raise ValueError
ValueError

**Predbat debug yaml file**
Not available, presumably due to exception error.

springfall2008, 2025-01-05:
This relates to issues getting Home Assistant history

>2025-01-05 17:48:49.893645: Error: Unable to fetch history for sensor.givtcp_battery_stack_1_stack_load_energy_today_kwh

Did you try to restart the add-on?

WilliamGillin, 2025-01-05:
Many thanks for your response.
Yes. I've restarted many times as well as doing a complet reinstall.
I've just looked at the MQTT data and there is a 
Stack_Charge_Energy_today_kWh and Stack_Discharge_Energy_today_kWh
but no stack_load_energy_today_kwh, which is what's causing the exception. The only thing on MQTT that mentions load is stack_load_voltage.

gcoan, 2025-01-06:
I think you are picking up the wrong entity name with givtcp 3, its not battery stack 1.

In the docs https://springfall2008.github.io/batpred/inverter-setup/#givenergy-with-givtcp

"If you are using GivTCP v3 and have an AIO or 3 phase inverter then you will need to manually set geserial in apps.yaml to your inverter serial number?

WilliamGillin, 2025-01-06:
Thank you. That has fixed it. 
I changed from GivTCP V2 to V3 last week and everything was working fine until yesterday, so I assumed all was OK. I think I might have rebooted the VM and that's could have been what caused the issue. 
Anyway I've learnt a bit more and am very grateful.

closed 2025-01-06 by WilliamGillin

gcoan, 2025-01-06:
> Thank you. That has fixed it. 
> 
> I changed from GivTCP V2 to V3 last week and everything was working fine until yesterday, so I assumed all was OK. I think I might have rebooted the VM and that's could have been what caused the issue. 
> 
> Anyway I've learnt a bit more and am very grateful.

Thanks for the extra info and glad that its fixed.

Not quite sure how predbat was still working when you upgraded givtcp, possibly still finding the v2 entities and after a week or so they got purged off.

Unfortunately the regular expression to find the inverter serial number doesn't work with AIO's on v3 as there are additional entities for the gateway and those get found when they shouldn't be.

Cheers