springfall2008/batpred #1852: Increase battery charge to 100%

nbullus, 2025-01-05:
**Describe the bug**
Battery not always charging to 100% most nights and as I am having battery SOC issues, I need to get as much form the battery as I can.

**Expected behaviour**
Encourage battery to charge to 100% in off-peak period and then export any unused prior to the next off-peak charge period.

**Predbat version**

v8.10

**Environment details**

- GE AIO
- Standard HAOS installer

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**

N/A

**Predbat debug yaml file**

[predbat_debug_19_30_00.yaml.txt](https://github.com/user-attachments/files/18311474/predbat_debug_19_30_00.yaml.txt)

springfall2008, 2025-01-05:
So your computed charge curve puts the last few percent quite slow "  
  95: 0.64
  96: 0.52
  97: 0.42
  98: 0.31
  99: 0.2
  100: 0.2"
If Predbat went to 100% then you wouldn't get as much export (during the night period at least)

- So if you turn off calculate export during charge then you get 100%
- I found setting best_soc_keep to 4.0 seems to get you to 99% and keeps the export

springfall2008, 2025-01-11:
Resolving this, please re-open if still an issue

closed 2025-01-11 by springfall2008