springfall2008/batpred #1853: Warnings since HA update

Emmoore103, 2025-01-06:
**Describe the bug**
There seems to be some warnings occurring since the latest HA Core Update. All seems to be working correctly at the moment despite this.

Message:
Logger: homeassistant.components.recorder.db_schema
Source: components/recorder/db_schema.py:658
integration: Recorder (documentation, issues)
First occurred: January 5, 2025 at 11:20:12 PM (1008 occurrences)
Last logged: 12:35:09 PM

State attributes for predbat.pv_energy exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for predbat.soc_kw_best10 exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for predbat.plan_html exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for predbat.soc_kw_best exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
State attributes for predbat.battery_cycle_best exceed maximum size of 16384 bytes. This can cause database performance issues; Attributes will not be stored
[predbat_debug_09_40_00.yaml.txt](https://github.com/user-attachments/files/18319461/predbat_debug_09_40_00.yaml.txt)

 

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Predbat version**

8.10.00

**Environment details**

- Givenergy HYBRID 5 - ver2 - 2x9.5KWh batteries
- Standard HAOS installer
- Octopus Go Intelligent
- GIVTCP v3
- EV - Zappi

**Screenshots**
[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

jrwm47, 2025-01-06:
There's a "don't worry about it" message covering this issue in the [predbat FAQs](https://springfall2008.github.io/batpred/faq/).

Emmoore103, 2025-01-06:
Cool, thank you.

closed 2025-01-06 by Emmoore103