springfall2008/batpred #1854: Multiple Duplicate Inverter Writes on AIO when throttling maximum discharge rate

7Van737, 2025-01-06:
For my All in One whenever the inverter_limit_discharge is active in apps.yaml and set at any value there are repeated duplicate writes to the inverter e.g.

2025-01-06 06:15:04,153 - GivTCP - write       -  [INFO    ] - Setting battery discharge limit 5250 was a success
2025-01-06 06:20:04,502 - GivTCP - write       -  [INFO    ] - Setting battery discharge limit 5250 was a success
2025-01-06 06:20:52,858 - GivTCP - write       -  [INFO    ] - Setting battery discharge limit 5250 was a success
2025-01-06 06:25:04,442 - GivTCP - write       -  [INFO    ] - Setting battery discharge limit 5250 was a success
2025-01-06 06:30:03,379 - GivTCP - write       -  [INFO    ] - Setting battery discharge limit 5250 was a success
2025-01-06 06:30:50,695 - GivTCP - write       -  [INFO    ] - Setting battery discharge limit 5250 was a success
2025-01-06 06:35:04,051 - GivTCP - write       -  [INFO    ] - Setting battery discharge limit 5250 was a success
2025-01-06 06:40:04,161 - GivTCP - write       -  [INFO    ] - Setting battery discharge limit 5250 was a success
2025-01-06 06:40:51,786 - GivTCP - write       -  [INFO    ] - Setting battery discharge limit 5250 was a success
2025-01-06 06:45:04,117 - GivTCP - write       -  [INFO    ] - Setting battery discharge limit 5250 was a success
2025-01-06 06:50:03,820 - GivTCP - write       -  [INFO    ] - Setting battery discharge limit 5250 was a success
2025-01-06 06:50:51,198 - GivTCP - write       -  [INFO    ] - Setting battery discharge limit 5250 was a success
2025-01-06 06:55:04,603 - GivTCP - write       -  [INFO    ] - Setting battery discharge limit 5250 was a success

If this entity is commented out of the apps.yaml and the default 6000 value used there are no multiple writes.   I first discovered this problem a few weeks ago so has been present on all recent versions of Predbat.

- Inverter and battery setup
- Standard HAOS installer or Docker
- Anything else?

All in One, HAOS, have used both GivTCP v2 & v3.

Attached GivTCP log file of writes from midnight to midday.
[x533ea71a_givtcp_2025-01-06T11-37-11.289Z.log](https://github.com/user-attachments/files/18321843/x533ea71a_givtcp_2025-01-06T11-37-11.289Z.log)

springfall2008, 2025-01-07:
This strikes me as a GivTCP bug, but can you share the Predbat log for the same period?

springfall2008, 2025-01-07:
I had a look on my system and I also get the same thing sometimes

2025-01-06 23:31:10,503 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 1800 was a success
2025-01-06 23:34:07,976 - GivTCP - write       -  [INFO    ] - Setting battery charge rate 1800 was a success

It appears if you read the Predbat log that GivTCP has failed to do the first write and returned back the old value only for it to be a written a second time. In my case after the second write it seems to change okay.

7Van737, 2025-01-07:
Here are predbat logs that mostly overlap  with the GivTCP log above.    

Not sure what I'm looking for but can't see any 'errors' or 'fail' to write.

As the entity is in the apps.yaml and not something changed very often does it need to be written so frequently?   This frequency does not occur when the entity is commented out of the apps.yaml

THANKS for you help.

[predbat.4.log](https://github.com/user-attachments/files/18331778/predbat.4.log)
[predbat.5.log](https://github.com/user-attachments/files/18331813/predbat.5.log)