springfall2008/batpred #1856: Don't force export when car is waiting for charge using IOG and Zappi

stevedundee2, 2025-01-07:
**Is your feature request related to a problem? Please describe.**

I do not want predbat to force export if the power would be diverted to the car by my Zappi.

When my car is plugged in and waiting for charge, any export to the grid is detected by the Zappi as solar excess so it diverts the export into the car. This pointlessly discharges the battery.

This is similar to #1230 but that user wanted no export at all (solar or battery) to be diverted into the car. I would like to be able to divert solar (it tickles my green bone) but not end up diverting battery exports.

**Describe the solution you'd like**

The ideal rule would be  "If the car is plugged in at home and its target SOC is greater than its current SOC, then do not export from the battery". 

My car has the required sensors but I don't know how common that is. So perhaps a good enough rule would be "if there are future IOG charging slots, do not export ". I'm hoping that once the Zappi tells IOG the car is full then IOG will cancel any remaining slots, but if it doesn't then it's not a big deal - the slots will be cleared come 8am anyway.

Even simpler, "if the car is plugged in do not export". This will miss some export opportunities that would not have been diverted, but that's not the end of the world to me and is perhaps more universal.

**Describe alternatives you've considered**

I could use automations to monitor the car plugged in at home and SOC data and instruct predbat to not export or to resume allowing export accordingly, but I don't see a control to do that. Unless "force demand" or one of the "force export/charge freeze" would do it?

**Additional context**

stevedundee2, 2025-01-07:
I suppose not "force demand" because that would mean the house uses the battery when it needn't.

Tonight's plan includes many slots where the battery is in Demand status while the car will be charging. The car will use the battery here too. Not the same diversion issue but still undesirable to me.

Perhaps I should just write the automations mentioned above and toggle predbat between "monitor" and "control charge & discharge" modes.

[predbat_plan.html.txt](https://github.com/user-attachments/files/18338727/predbat_plan.html.txt)

cyberkryten, 2025-01-08:
The problem you have is that the Zappi has no knowledge of why the export is happening.  If you have it setup to divert to the car, then it will always do that - it cannot know if that power is coming from Solar or your Battery - export is export as far as the Zappi is concerned.  There are ways of dealing with it by changing how the system is wired and where the Harvi sensor is placed, but that's a lot of work.

Exports are made because they make sense financially, so it's not clear what benefit you would gain from not doing so when the car is connected?

There is a Predbat switch to allow the car to be charged from battery: switch.predbat_car_charging_from_battery - is that on or off?

If you want it to only charge based on the IOG then you can make a change to the Zappi config as was noted on the other issue.  If you set the export detection amount to greater than your maximum export (I used my inverter size of 8kW) then it will never trigger car charging because of export and only charge using IOG periods.

stevedundee2, 2025-01-08:
Hi

Yes I know all that. But while exporting solar makes financial sense, as I noted it suits me to use it to charge the car. It's not all about money.

As I subsequently noted, I had a similar issue with last night's plan - predbat was going to set the battery to Demand at the times the car was going to charge and so it seemed like the battery would be used to charge the car. In fact last night the battery didn't discharge to the car. I have switch.predbat_car_charging_from_battery set off. Perhaps this is why the predbat status was Hold for car and later Hold charging? So perhaps it's all good on this front though I'm not sure why there would be 2 statuses given that the car was charging steadily throughout.

I have tried some automation. It's pretty easy so perhaps as my requirement is pretty niche that will do.

cyberkryten, 2025-01-08:
Having spent months writing a whole myriad of automations when the system was installed, one I found Predbat, I decided that I'd taken the view that I'd let Predbat 'sort it out' in terms of cost/benefit on Import/Export and car charging.  

Is IOG setup to control your car or your Zappi?  Do you have the car SOC % available in Home Assistant?

If IOG is controlling the car, or you have Predbat setup doing the car charging, then setting the switch to stop the car charging from your battery should do what you are asking.  

The intent is that if Predbat sees the car charging it will set the battery discharge rate to zero - thereby allowing the car to charge from Grid (under IOG control) or Solar (if there is excess).  However, it does require that Car/IOG/Zappi linkage is all working correctly so that Predbat knows what is going on.

stevedundee2, 2025-01-09:
Hi

I'm similarly inclined. It gets too confusing otherwise. But there are some things that predbat doesn't seem to do.

My car's an ID.3 and has sensors for current and target SOC. The API has sometimes been rather ropey so IOG controls the Zappi.

That switch seems like it will avoid the battery being used to charge the car when IOG is charging it and predbat is in Demand status. It certainly seemed to the other night. I can keep an eye on this.

I don't think it solves the problem of the Zappi seeing a forced export as solar excess. I've previously seen the car charging from a forced export outwith an IOG slot. One evening it stopped exporting almost immediately but later that evening it forced export again and this time just carried on.

I think the solution to that is to set the predbat mode to "Control charge" when the car is plugged in (and the other conditions I mentioned above) and set it back to "Control charge and discharge" when the target SOC is reached or the car is unplugged. 

The first is simple because I already have an automation with those conditions (it unlocks the Zappi and tells Octopus how much charge I need). 

I am hoping that a template trigger "{{ states('sensor.car_battery_level') >= states('sensor.car_battery_target_charge_level') }}" will work for the second. I can monitor it - not the end of the world to stay in "Control charge" for longer than intended while I figure it out.

Do you know whether I am understanding "Control charge" correctly? I think it will make sure the house battery is charged without me needing to set up a charging slot on givenergy.

Thanks for your help.

cyberkryten, 2025-01-10:
I think your issue is that you cannot solve the problem of the Zappi not knowing if power came from Solar or Battery so have to deal with it another way.  The Zappi monitors AC and in a Hybrid system, AC could be either so you either accept the Zappi will 'grab' charge like it is doing, or increase the limit to above what your inverter can generate to stop it doing so.

If you want the Zappi to grab charge then you'll need to manage Export so that batteries are not discharge for Export when the car SOC is less than the car Max SOC, so swapping to 'Control Charge' would allow this.

But, if Predbat can do it for you, then all the better - I'm not entirely sure how the existing switch operates but I think it aims to do what you're after - not discharge the batteries when the car will be charging so it is probably worth running with the switch and enabling debug - then if you get an instance where you think it has behaved incorrectly, post the debug log and it can be looked at.

stevedundee2, 2025-01-10:
"Control charge" works - I tried it manually last evening and the battery started to charge when you'd expect it to. So that fixes the forced export problem. I just need to test the automations in anger next time the car needs charging, which is probably Tuesday night. And again next time there's enough sunshine for solar diversion ... April perhaps if we're lucky!

I think that perhaps the switch works by looking at the Octopus charge plan. I don't think Predbat knows whether the car is actually charging. Whatever the mechanism, this aspect did seem to work last time, with the caveat that two different predbat statuses were used over the period. I'll see what it does next time.

Good tip about debug logging. I'll try that on Tuesday night too. 

Thanks again for your help. I'm going to close this issue to keep things tidy for the predbat team. I'll update it next week though with what happens when the car next charges.

closed 2025-01-10 by stevedundee2