springfall2008/batpred #1858: Hold for car not holding battery

nickgee31, 2025-01-08:
**Describe the bug**
During IOG hours, hold for car is sometimes used where Predbat plans for the battery to be paused so that the SOC doesn't change. At the moment, Predbat is calling the charge_stop_service and therefore the battery is being discharged.

**Expected behaviour**
Can the charge_freeze_service be called when the status is Hold for car instead? This should mean that the battery is held at the SOC rather than dropping unexpectedly and resulting in Predbat having to charge when it realises 5-10 minutes later.

**Predbat version**

v8.10.0

**Environment details**

- FoxESS Inverter

**Log file**
[predbat.1.log](https://github.com/user-attachments/files/18344435/predbat.1.log) - 05:01:00 is a timestamp of when Hold for car is set as the status and charge_stop_service is called.
[apps.yaml.txt](https://github.com/user-attachments/files/18344452/apps.yaml.txt)

springfall2008, 2025-01-11:
I think this is a bug, will try to fix it when I get time

springfall2008, 2025-01-11:
Fix on 'main' and will be released soon

nickgee31, 2025-01-11:
Amazing thank you! Will reopen if I notice anything 🙂

closed 2025-01-11 by nickgee31