springfall2008/batpred #1859: Not charging before a high export rate

WillCalderwood, 2025-01-08:
**Describe the bug**
Not charging before a "GivBack" event.

**Expected behaviour**
When there's a high export rate in the future, I'd expect the battery to charge and discharge during the high export rate.

**Predbat version**
8.9.3

**Environment details**

- GivEnergy Inverter and battery
- Standard HAOS

**Screenshots**
[image]

**Log file**
Attached

**Predbat debug yaml file**
Attached

[predbat.log](https://github.com/user-attachments/files/18348701/predbat.log)
[predbat_debug_14_30_00.txt](https://github.com/user-attachments/files/18348726/predbat_debug_14_30_00.txt)

WillCalderwood, 2025-01-08:
I tried making the export rate really high to encourage it more, now it's doing this. Still definitely not optimal.
[image]

WillCalderwood, 2025-01-08:
Updated to 8.10.0, massive rate manipulation. Why would it charge at 1000 so it can discharge at 500 later?!
[image]

johnwb87, 2025-01-08:
Because you've selected a forced charge and not cleared it.

Try a forced discharge during the Saving Session window and see what happens.

johnwb87, 2025-01-08:
Also if 50p is the award level in your region your export override should be 50+15 as you get the award in addition to normal export.

gcoan, 2025-01-09:
Your rate overrides don't look right @WillCalderwood you only need a rates_export_override either with rate 87 (72p DFS + 15p normal export) or a rate_adjustment of 72 (the DFS amount).

You don't need a rates_import_override normally, as @johnwb87 says that can make it more of a refined calculation if you normally import in that slot, but its hard to get predbat to fully estimate the plan.

springfall2008, 2025-01-11:
If all the import rates are the same high rate then Predbat won't use them, especially if combine_charge is True, so overriding the import rate to a higher value for the actual saving slot fixes the issue.

closed 2025-01-11 by springfall2008