springfall2008/batpred #1862: Release 8.10.1 documentation question

jrviz, 2025-01-09:
Should the service API repeat option keyword  be "repeat" as shown in the documentation or "always"?

cyberkryten, 2025-01-10:
There was already a flag used in the code service_repeat which was used to see if the same service has been called before.

The change added service_always and the code is indeed looking for the key 'always'

inverter.py line 1982:

```
                for key in service_template:
                    if key == "service":
                        service_name = service_template[key]
                    elif key == "always":
                        service_always = service_template[key]
                    else:
                        value = service_template[key]
                        value = self.base.resolve_arg(service_template, value, indirect=False, index=self.id, default="", extra_args=full_data)
                        if value:
                            service_data[key] = value
```

jrviz, 2025-01-10:
Yes, that's what I read in the code so my issue was relating to the documentation which says "repeat" as a keyword. Or am I missing something?

springfall2008, 2025-01-11:
You are right I made a mistake and called it 'always' in the service

springfall2008, 2025-01-11:
Fix on 'main' and will be released soon

closed 2025-01-11 by gcoan