springfall2008/batpred #1863: Provide ability to have per-inverter charge scaling factors

gcoan, 2025-01-09:
**Is your feature request related to a problem? Please describe.**
In this colder weather I see the 5.2 battery attached to one inverter almost always charges at 1.7kWh whereas the 9.5 battery attached to the other battery charges at 2.4kWh

[image]

I have input_number.predbat_battery_rate_max_scaling set to 0.92 representing scaling the inverters supposed 2.6kWh rate to 2.4kWh, but as this is a global setting across all inverters it results in an incorrect charge rate for the 5.2 and an incorrect Predbat plan that can only be achieved if the battery cells are above 20 degrees (when the 5.2 charges at 2.4)

In the chart above, blue arrows are where the 5.2 charges at 1.7, red is where it charges at the same 2.4 as the 9.5 battery.

This is linked but not the same as #1844 (understanding temperature and impact on charging rate) and has a linkage to #830 (taking account of battery capacities for multiple inverter charge rates)

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
Probably need to move input_number.predbat_battery_rate_max_scaling into apps.yaml so you can specify a list per inverter, and add this to the list of fields that you can change with predbat_manual_api

Am not looking for Predbat to apply different charge rates based on temperature (that's #1844) and most of the time the 5.2 is below 20 degrees so charges at 1.7.  I'd be happy to let Predbat assume that its always 1.7 for the winter and I change it in summer to 2.4.

**Describe alternatives you've considered**
Could specify that both inverters charge at the lower rate and live with the batteries charging better than predicted.  Will result in a less efficient plan the other way round

springfall2008, 2025-01-11:
One issue is all the modelling is done with the two inverters as one big battery, so you couldn't have different rates for each one it would be a combined rate.

However you could have a weather compensation curve which you would average across the two inverters.

I do wonder if the modelling should be per inverter, but the downside is it could take twice as long to simulate.

gcoan, 2025-01-11:
it's definitely in the enhancement category, and I realise this isn't a straight forward request.

There are other issues as I have raised before about handling of my multi-inverter dissimilar battery setup, the prediction of battery charge and discharge for instance isn't totally correct because its assumed that 2x inverter charge rate is available whereas once one inverter/battery reaches 100%, that stops charging so only 1x charge rate is available.

Ultimately independent simulations might be the answer and for those of us with multiple inverters it becomes part of the trade-off for having a more accurate plan

gcoan, 2026-08-20:
no code changes have been made to allow multiple scaling factors, but I'm not sure how much of an issues this really is for me now, so closing

closed 2026-08-20 by gcoan