springfall2008/batpred #1864: Use secrets.yaml for API keys

gcoan, 2025-01-09:
**Is your feature request related to a problem? Please describe.**
At present Predbat provides the ability to configure various keys within apps.yaml, e.g.:
- HA API Key
- Solcast API Key
- GE Cloud API Key
- Possibly others for other non-GivEnergy inverters, IDK

This is insecure because if the user shares their apps.yaml file or the predbat debug file - e.g. in a github issue - then these API keys are available to anyone who reads the github issue

**Describe the solution you'd like**
Predbat should use /config/secrets.yaml which is the HA standard way of configuring passwords and other sensitive information within HA and minimising them being shared (as secrets.yaml need normally never be shared)

**Describe alternatives you've considered**
Potentially could move these API Keys to HA controls so they are not inside apps.yaml, but doesn't seem a good solution

Provide an obfuscation/ removing of keys in the debug.yaml file when Predbat creates it - this solves the problem for the debug file but not apps.yaml

Have a separate predbat_secrets.yaml file - probably least-worst alternative.  Would be better if we could use the standard secrets file though

springfall2008, 2025-01-11:
Good idea.

BTW: The predbat_debug.yaml does remove keys

gcoan, 2025-01-11:
> Good idea.
> 
> 
> 
> BTW: The predbat_debug.yaml does remove keys

thanks Trefor, it appears to keep my GE cloud key intact in the debug file 😬

springfall2008, 2025-01-12:
> > Good idea.
> > BTW: The predbat_debug.yaml does remove keys
> 
> thanks Trefor, it appears to keep my GE cloud key intact in the debug file 😬

Okay that's a bug, let me check