springfall2008/batpred #1865: Load shows 0 in the plan for first half of the day and weekends

kams19, 2025-01-10:
**Describe the bug**
The forecast load shows 0 for first half day and mostly all day on weekends.

**Expected behaviour**
The forecast load should show the average of what it was in the last week in that 30min slot.

**Predbat version**

8.10.1

**Environment details**

- Inverter and battery setup : SolarEdge and GivEnergy AIO gen 1
- Standard HAOS installer or Docker : Standard HAOS
[predbat.log](https://github.com/user-attachments/files/18376137/predbat.log)

- Anything else?

**Screenshots**
please see attached

**Log file**
please see attached

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.
[predbat_plan.html.pdf](https://github.com/user-attachments/files/18376214/predbat_plan.html.pdf)
[apps.yaml.txt](https://github.com/user-attachments/files/18376215/apps.yaml.txt)
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/18376216/predbat_debug.yaml.txt)

gcoan, 2025-01-10:
It says in the logfile:

2025-01-10 05:10:01.910344: Historical data totals for days [7] are [10.49] - min 10.49
2025-01-10 05:10:01.910986: Historical days now [7] weight [1]

So Predbat is only taking the historical load from a single day, 7 days ago.

Have a check of your givtcp load sensor, if that has gaps in the history 7 days ago then this will cause Predbat to predict the same.

Its generally recommended to have multiple days load history being used for Predbat that way an average will be taken rather than a single days data

I have days_previous set to 2 through to 8 for example

kams19, 2025-01-10:
> It says in the logfile:
>
> 2025-01-10 05:10:01.910344: Historical data totals for days [7] are [10.49] - min 10.49 2025-01-10 05:10:01.910986: Historical days now [7] weight [1]
>
> So Predbat is only [...]

Thanks @gcoan.
I have replicated the days_previous like yours now, the load prediction for today has gone down from 9kWh to 4kWh for today. Something doesnt seem correct.

Could you advise how to check for gaps in givtcp load sensor please?

gcoan, 2025-01-10:
simplest thing is to just look at the history graph of that sensor in home assistant, look at it for say today and each days_previous

gcoan, 2025-01-10:
if there are gaps in the load history then Predbat will report these in the logfile and will attempt to fill them in, but obviously it can't make up the data

kams19, 2025-01-10:
> simplest thing is to just look at the history graph of that sensor in home assistant, look at it for say today and each days_previous

am I looking at the right sensor?
[image]

kams19, 2025-01-10:
The plant has become even worse now - i had to manually freeze export as the plan mentions 0 Load. 
[image]

Is there a way I can start fresh with Predbat learning again about the Load Forecast? Or do you have anything else in mind to fix this issue as I dont want to fiddle with the configuration every couple of hours.

gcoan, 2025-01-10:
In your apps.yaml, you have (or had):

```
load_today:
    - sensor.givtcp_{geserial}_load_energy_today_kwh
```

that's the sensor you need to look at. Its energy used in the house today in kWh, not a power graph (the bottom chart you showed).

> Is there a way I can start fresh with Predbat learning again about the Load Forecast? Or do you have anything else in mind to fix this issue as I dont want to fiddle with the configuration every couple of hours

Predbat has no memory, every time it runs it loads the data from the sensors it is configured for, so if you change the configuration then it will re-run with the new config.

It suggests to me from your plan that you either have the load_today sensor mis-configured, gaps in the givtcp history or something very weird with the days_previous.
Can you share your latest apps.yaml and logfile?

If you set predbat to 'Control charge' it will only control battery charging and won't do any forced exports

kams19, 2025-01-10:
> load_energy_today_kw

[image]
this is the graph for that sensor - again no gaps.

I have attached the latest apps.yaml and logfile herewith. Do you see anything unusual in this?
[predbat_100120252221.log](https://github.com/user-attachments/files/18382946/predbat_100120252221.log)
[apps.yaml-latest.txt](https://github.com/user-attachments/files/18382947/apps.yaml-latest.txt)

gcoan, 2025-01-11:
I don't really understand what's happening.

The chart shows data throughout on the load sensor, its not so much gaps, more the line going horizontal for a while would indicate no load, but there isn't

But looking at the logfile Predbat says that overnight the cumulative load is flat from 2am to about 9am:
[image]
- which doesnt match the sensor data

but when the historical load data is loaded it appears to come in OK, with only one small data gap reported:
[image]

but one thing I do notice is that the historical data is being loaded saying its about 3-4kW/day in the most recent days, which doesn't match the sensor graph you shared

have you got input_number.predbat_load_scaling set? this might be scaling your load downwards and then the low overnight load is getting zero'd out

kams19, 2025-01-11:
> I don't really understand what's happening.
>
> The chart shows data throughout on the load sensor, its not so much gaps, more the line going horizontal for a while would indicate no load, but there [...]

This is baffling me and hence I thought I’d put it it here for don’ some help. I’m thinking of setting it out all over again, as I have put a lot of time troubleshooting this. This is ongoing since quite few days now.

The load scaling is set to 1.2, so in fact that should magnify the value.
[image]

let me know if you too think that I should start afresh?

gcoan, 2025-01-11:
I've noticed 

2025-01-10 22:15:29.733094: Today's actual load so far 4.87 kWh with 10.99 kWh Car/iBoost excluded and 2.05 kWh import ignored.

try commenting this line out in apps.yaml

  # Also adjust car_charging_energy_scale if it's not in kwH to fix the units
  car_charging_energy: 're:(sensor.givenergy_ev_power)'

gcoan, 2025-01-11:
> That seems like the issue, ev_power sounds like power and not energy

Ah yes, I hadn't spotted that, power instead of energy would not be good and is almost certainly the cause of @kams19 problem

As a suggestion @springfall2008 could the subtraction of EV load be made more explicit in the logfile, I was tracing the house load sensor thinking the issue was with that, not realising until just now what there was an EV load sensor that was being subtracted, giving a very low house load kWh.  I didn't see anything in the log about this EV load, only spotting it in apps.yaml made me realise this could be a cause

kams19, 2025-01-11:
> I've noticed
>
> 2025-01-10 22:15:29.733094: Today's actual load so far 4.87 kWh with 10.99 kWh Car/iBoost excluded and 2.05 kWh import ignored.
>
> try commenting this line out in apps.yaml
>
> # Also [...]

seems like that was it:
[image]

Thanks - will keep an eye on how it turns out from here

gcoan, 2025-01-11:
just to explain further,  Predbat can optionally subtract your car EV charging data from your total house load so it gives you a correct base load figure

however the sensor you had used for EV load was incorrect, you had selected EV Power, i.e. the Power the EV charger is consuming, not the total EV energy used during the day

if you want to subtract EV charging then you need to select the correct energy sensor.

Whether you need to exclude EV load also depends on how your system is wired up, its possible to connect the inverter CT clamp so that it doesn't 'see' the EV charging so you don't need to subtract it

kams19, 2025-01-12:
> just to explain further, Predbat can optionally subtract your car EV charging data from your total house load so it gives you a correct base load figure
>
> however the sensor you had used for EV load [...]

Thanks, and it seems the ct clamp is placed as you say. If you see the attached pic, the grid power shown is only for house load, the car is being charged at 1.2kw presently but grid per only shows home load. So should I just keep the config for ev power hashed out?
[image]

gcoan, 2025-01-12:
> If you see the attached pic, the grid power shown is only for house load, the car is being charged at 1.2kw presently but grid per only shows home load. So should I just keep the config for ev power hashed out?

yes

If the inverter only 'sees' the house load when the EV is charging (and doesn't include the EV load) then the givtcp energy_load_today_kwh is all you need.  Predbat will plan your battery charging and discharging based upon this house load.

Separately EV charging will be planned by Predbat.

The ev_power line you commented out is to tell Predbat what the historical EV charging energy was so that Predbat can subtract this from your house load sensor.  This is for people that the inverter DOES 'see' the EV charging as house load and the subtraction is to ensure you get the right house load figure.
In your case no subtraction is necessary, and when you were subtracting, was resulting in periods of zero house load.

And to compound it, it looks like you could have used the wrong sensor as well, one that was giving power not energy.

Anyway, what you have now is correct for your setup

kams19, 2025-01-12:
> yes
>
> If the inverter only 'sees' the house load when the EV is charging (and doesn't include the EV load) then the givtcp energy_load_today_kwh is all you need. Predbat will plan your battery [...]

Thanks for the detailed explanation. Much appreciated. This can now be closed as resolved, not sure if I can do that myself?

closed 2025-01-12 by gcoan