springfall2008/batpred #1870: fox.yaml pv_power is in kW but apps.yaml expects it to be in W

WyndStryke, 2025-01-12:
**Describe the bug**
Set up my Fox inverter via Nathan's modbus the other day.  I noticed while looking at the generated charts that 'pv power' was registering zero, so investigated, and it seems the sensor is being picked up, but that it is being supplied as kW, whereas predbat expects the sensor to be in W.  The fox.yaml template file just assigns sensor.pv_power as-is.  I created a template sensor to do the conversion.

**Expected behaviour**
Either fox.yaml should document that a template sensor is required to convert from kW to W. or alternatively, predbat should detect the units of the sensor and adjust accordingly.  I feel the latter would be preferable, but the former is probably easier in the short term.  There are probably multiple sensors which could have unit detection, but pv_power is the one I've encountered.

**Predbat version**

8.10.2

**Environment details**

- Fox ESS KH7 via Nathan's Modbus, EC4300-H4 battery
- HAOS via predbat add-on wrapper
- I have attached my apps.yaml, this is not intended to supplant the official fox.yaml since it is tailored to my system / wants, there are various things which aren't configured yet in it.

**Screenshots**
[image]
[image]
[apps.zip](https://github.com/user-attachments/files/18389472/apps.zip)

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

gcoan, 2025-01-12:
I think your problem could well just be the Apex chart that needs scaling from W to kW

I don't use the fox apps.yaml but my solar sensor supplies kW to Predbat, and Predbat generally expects things in kW https://springfall2008.github.io/batpred/apps-yaml/#data-from-home-assistant

WyndStryke, 2025-01-12:
Looks similar to: 

- https://github.com/springfall2008/batpred/issues/1717
- https://github.com/springfall2008/batpred/issues/1640

WyndStryke, 2025-01-12:
> I don't use the fox apps.yaml but my solar sensor supplies kW to Predbat, and Predbat generally expects things in kW 

That page shows:

`
pv_power - Current PV power in watts
`

gcoan, 2025-01-12:
> > I don't use the fox apps.yaml but my solar sensor supplies kW to Predbat, and Predbat generally expects things in kW 
> 
> 
> 
> That page shows:
> 
> pv_power - Current PV power in watts
> 
> The daily total is expected to be kWh, but the current power is expected to be W, if I'm reading it correctly.
> 

You are correct.  Predbat expects pv_power to be provided in Watts and pv_today in kWh.

The original issue describes a problem with the Apex chart not showing PV power correctly.  This is easily fixed by adding a transform into the chart configuration.

But I'm not sure what the impact of predbat receiving pv_power in kW would be.  The forward plan is derived from the Solcast forecast not actually what is being generated.  I suspect if pv_power is supplied as kW not W then the impact will be limited to some of the 'cost savings using solar' not being quite right.  Don't know for sure.

But at the moment a template would be needed to convert the units to what Predbat expects

springfall2008, 2025-01-12:
Predbat should convert between watts the kw automatically

```
                        if required_unit in ["kWh"] and unit in ["W"]:
                            state = state / 1000.0
                            integrate = True
                        elif required_unit in ["kWh"] and unit in ["kW"]:
                            integrate = True
                        elif required_unit in ["kW", "kWh", "kg", "kg/kWh"] and unit in ["W", "Wh", "g", "g/kWh"]:
                            state = state / 1000.0
                        elif required_unit in ["W", "Wh", "g", "g/kWh"] and unit in ["kW", "kWh", "kg", "kg/kWh"]:
                            state = state * 1000.0
                        else:
                            # Ignore data in wrong units if we can't converter
                            continue
```

WyndStryke, 2025-01-12:
This is how sensor.pv_power is defined (presumably by Nathan's modbus integration)

`
sensor.pv_power
state_class: measurement 
unit_of_measurement: kW 
device_class: power 
icon: mdi:solar-power-variant-outline 
friendly_name: PV Power
`
This is what my template sensor looks like:

`
sensor.pv_power_w
state_class: measurement
unit_of_measurement: W
device_class: power
friendly_name: pv_power_w
`
Will the 'config' page show the converted value or the original value, and similarly, will that Ajax graph show the original or the converted?   It was showing 0.0 for the sensor.pv_power, and a watts figure for sensor.pv_power_w, however  I will have to check both again when there is generation happening.

gcoan, 2025-01-12:
The config page and the apex chart will both show the sensor as provided to predbat. The code fragment is predbat converting the item to use internally in its code. 

If the apex chart is showing the sensor wrong then either add a transform in the apex chart to multiply the kw by 1000, or use your template sensor in the chart instead

springfall2008, 2025-01-13:
I think I know the issue now, Predbat converts only for history, hence the current values will still be in the original units. I maybe should fix that one day.

WyndStryke, 2025-01-13:
As long as it only affects the display, and doesn't affect the plan (that's the main thing I was concerned about) then I'm happy.

So maybe as a low priority enhancement, add 'units' to to the config display, if it is displaying the original, but really knowing that the plan is fine is enough.

TonyHoyle, 2025-06-17:
The only ones I'd question would be the force_discharge_power and force_charge_power as they're writeable and the PR doesn't mention them.

WyndStryke, 2025-06-17:
It seems to **read** it and pick up the units as expected, but being in read-only mode, I can't tell if it would **write** it in the correct units.

springfall2008, 2025-06-17:
> It seems to **read** it and pick up the units as expected, but being in read-only mode, I can't tell if it would **write** it in the correct units.

I'm afraid not (yet), its a good point that probably needs addressing

springfall2008, 2025-06-22:
The conversion during writes is now done and released

WyndStryke, 2025-06-22:
@springfall2008 
Hi Trefor, 

Many thanks for that!   Just tested with 8.21.7, on my Fox KH7.   Looks like it is trying to write Watts into it, rather than kW?  It looks like it converted successfully on the read.

9594 | 2025-06-22 11:07:14.426476: Warn: record_status Warn: Inverter 0 write to discharge_rate failed
9593 | 2025-06-22 11:07:14.388039: Warn: Inverter 0 Trying to write 7000 to discharge_rate didn't complete got 1196.0

I turned off my own scheduler,  turned on set_export_low_power and set_charge_low_power to encourage it to write to the two fields, and turned off read-only mode.  The original value of the field was 1.196kW, field definition below.  Field created by the FoxESS Modbus integration.

I've attached the apps.yaml and the log.

cc: @TonyHoyle 

```
[number.force_discharge_power]

(http://homeassistant.local:8123/developer-tools/state#)
Force Discharge Power
1.196	min: 0
max: 7
step: 0.001
mode: box
unit_of_measurement: kW
device_class: power
friendly_name: Force Discharge Power

```

```
9595 | 2025-06-22 11:07:14.426745: Inverter 0 adjust target soc for export to 0.0% based on requested all inverter soc 0%
-- | --
9594 | 2025-06-22 11:07:14.426476: Warn: record_status Warn: Inverter 0 write to discharge_rate failed
9593 | 2025-06-22 11:07:14.388039: Warn: Inverter 0 Trying to write 7000 to discharge_rate didn't complete got 1196.0
9592 | 2025-06-22 11:07:12.383591: Warn: Service call number/set_value data {'value': 7000, 'entity_id': 'number.force_discharge_power'} failed with response None
9591 | 2025-06-22 11:07:10.347769: Warn: Service call number/set_value data {'value': 7000, 'entity_id': 'number.force_discharge_power'} failed with response None
9590 | 2025-06-22 11:07:08.311889: Warn: Service call number/set_value data {'value': 7000, 'entity_id': 'number.force_discharge_power'} failed with response None
9589 | 2025-06-22 11:07:06.277019: Warn: Service call number/set_value data {'value': 7000, 'entity_id': 'number.force_discharge_power'} failed with response None
9588 | 2025-06-22 11:07:04.241111: Warn: Service call number/set_value data {'value': 7000, 'entity_id': 'number.force_discharge_power'} failed with response None
9587 | 2025-06-22 11:07:02.205266: Warn: Service call number/set_value data {'value': 7000, 'entity_id': 'number.force_discharge_power'} failed with response None
9586 | 2025-06-22 11:07:02.174111: Inverter 0 current discharge rate is 1196.0W and new target is 7000W

```

[apps_yaml.txt](https://github.com/user-attachments/files/20852150/apps_yaml.txt)

[predbat.log](https://github.com/user-attachments/files/20852147/predbat.log)