springfall2008/batpred #1873: Unnecessary charging after 100% SOC

Yuffi17, 2025-01-14:
**Describe the bug**
I don't know if this is the same issue I had before or if it is a new one. The evening/night plan shows excessive charging. 
By "excessive" I mean predbat plans to charge to 100% and then continue to charge when this is neither necessary to meet load nor economically sensible.

**Expected behaviour**
I have 19kWh available and a daily use of about 14kWh so expect a 2/3 cycle of the batteries. I don't mind charging to 100% to have some spare, but to start charging at10:30pm and continue charging (to 100%) until 11:30am seems crazy. 

I expected predbat to select the cheapest slots to charge my battery, and then stop charging so I can use energy from the battery to meet the house load.

**Predbat version**
8.10.2

**Environment details**
- Inverter and battery setup: GE Hybrid 5kW inverter with 2xGivBatt 9.5kWh batteries

**Screenshots**
[image]

I know this shows the longer term plan which is subject to change, but predbat sems to be struggling at the moment and I don't understand why.  This particular plan has persisted despite predbat running several times (it's not a one off plan that disappears on the next run).

**Predbat debug yaml file**
[predbat_debug_15_00_00.yaml.txt](https://github.com/user-attachments/files/18412197/predbat_debug_15_00_00.yaml.txt)

gcoan, 2025-01-14:
Do you have low charge mode turned on?  I have noticed some weird behaviour in #1866 with low power charge mode.

Its not entirely due to low power mode as turning this switch off Predbat still holds the SoC at 100% through the morning peak like you are getting

springfall2008, 2025-01-18:
Oddly this appears to be caused by 'metric_self_sufficiency' as if you set it to 0 the problem goes away. I'll take a look as to why as it appears to be a bug

Yuffi17, 2025-01-18:
Today's plan is also a bit dodgy (charging when not needed) so I have just tried setting metric_self_sufficiency to zero 0 to see the difference, but there wasn't any real change. 

I've taken a debug log of this plan so you can see. I only screen-grabbed the start of the plan - there's enough here to see it doesn't need to charge and the debug file has the full story anyway.

[image]

[predbat_debug_11_15_00.yaml.txt](https://github.com/user-attachments/files/18464229/predbat_debug_11_15_00.yaml.txt)

springfall2008, 2025-01-18:
Good spot, I found a potential fix which I have pushed to 'main' for testing.

Yuffi17, 2025-01-18:
Thanks! 
I've loaded Main (8.11.1) and run it a couple of times to let the plan settle. Unfortunately I'm still seeing charging after 100%. The plan does a really good job of picking the best slots to charge from up to 100%, it's just a bit over enthusiastic once the target is reached.

[image]

[predbat_debug_16_10_00.yaml.txt](https://github.com/user-attachments/files/18465216/predbat_debug_16_10_00.yaml.txt)

springfall2008, 2025-01-18:
I find if you set metric_self_sufficiency to 0 and best_soc_keep to 2 then the plan looks reasonable. I'm struggling to figure out why self sufficiency makes things worse.

Yuffi17, 2025-01-18:
I think it may be some interplay between Predbat and the varying rates of Agile. I know someone who has two AIO and was running Predbat on agile with the same problem - they swapped to cosy and the Predbat plan is now behaving as expected. I'll try to find out more info.

The original introduction of self_sufficiency was to provide a way to bias the plan towards the GE eco mode (meeting house load from battery where possible). I realise this sort of goes against the original "maximise return" aim of Predbat because they try to reduce low gain charge/discharge, and retain some battery capacity rather than making full use of it. 

I don't understand the code well enough to help properly, but I do have some questions: perhaps they might help spark some thoughts?
- Does Predbat assemble the best plan from a set of different models, and is it getting confused at this stage? I have seen the plan change back and forth sometimes as it updates (not got screen shots or debugs because I can't tell when this will happen)
- Is the excessive charging a form of hold charge? Sometimes the plan shows green charge to 100% and sometimes blue hold charge at 100% but both have the same effect. I have also seen plans swap between the two states.
- The issue I am seeing happens after the battery system reaches 100% SOC - I can't recall seeing such behaviour at the end of a charge slot that results in less than 100%. So what triggers the plan to stop charging? Or possibly, what is preventing the plan from recognising the 100% has been reached?
- Why does the plan seem to be stable on a tariff with fixed rates and unstable on Agile? What is it about the varying rates that causes this charging over-run?

Sorry I can only offer questions and not answers. And many thanks for all the time you spend trying to get this working perfectly for those of us who don't understand the code!

springfall2008, 2025-01-19:
Okay I've spent quite a bit of time looking at this and I can offer an explanation.

You have your losses set as

5% - charge
5% - discharge
6% - inverter

This means to charge the battery from grid and then supply it to your home you have losses of

1.05 x 1.06 x 1.06 x 1.05 = 23.9%

If you sit at 100% battery then none of these losses apply.

That means a 27.12p import slot when at 100% costs about the same as a 21.9p charge slot. 

If for example I change the inverter loss back to 3% then a get more of the plan you might expect.

The default loss values are 3% charge, 3% discharge and 4% inverter loss. How did you pick your figures?

[image]

Yuffi17, 2025-01-19:
Ok - that makes sense.

I took my loss figures from easyPV and the GivEnergy data sheets. I can't find the system efficiency data on easyPV anymore to check the values. The GE hybrid inverter sheet says 94% charge/discharge efficiency and 97.5% PV efficiency. 

If this means the inverter loses about 2.5% (dc panels to mains) and the battery side loses 3.5% then in effect I have entered the same figure twice. I'm happy to try the defaults.

Question: Does metric_self_sufficiency affect the comparison of import slot price to charge slot price?

gcoan, 2025-01-19:
In tests I've seen on the GivEnergy community most people have concluded that total round trip losses are in the range 10-20%.  Some people take 15% as a mid point and others 20% as an easy approximation.

The losses are surprisingly higher than you think they would be, but does look like yours are a bit high.
For comparison mine are 0.055, 0.05 and 0.04 - its your inverter loss that is making the big difference as its applied twice in charging and discharging the battery

springfall2008, 2025-01-19:
I would not include any PV Losses in this calculation, instead either adjust your Solcast settings or use pv_scaling to account for that.

Yuffi17, 2025-01-19:
Thanks. I really appreciate the help and advice. I'll adjust the figures and see how things go.

dabeeeenster, 2026-01-03:
Ah ok my confusion was based on the action of charge for several 30m segments when the battery is already at 100%? Is that expected?

gcoan, 2026-01-04:
The original post did identify a bug which has been fixed and the OP confirmed it was better, but had some comments about why the battery was being held and not released,  the explanation then for the plan behaviour was down to the losses that had been configured meant it was uneconomical to use the battery.   Your case is not really the same, predbat is just holding the battery full until the rates drop.

I'm going to close as this issue was resolved originally