springfall2008/batpred #1876: Support for GivEnergy EMS

markbw999, 2025-01-17:
Hi Trefor,
I'm planning on doubling my battery capacity ahead of getting an air source heat pump. I don't want to lose the investment in my existing kit and can't just add another battery as the existing inverter is limited to 3kw. The solution my installer has proposed is a second inverter and battery linked by an EMS. I don't know much about the EMS yet but I believe it talks to the inverters to prevent cross-charging, etc. I therefore assume in such a setup predbat would have to talk to the EMS via GivTCP, not the inverters. Can you support that? I don't really want to do the install and find I lose predbat.
PS - I posted this in the predbat chat but I think it has got lost in the noise.

gcoan, 2025-01-17:
I don't know if GivTCP supports the EMS yet.  Its a new product and I suspect that it may not present itself to GivTCP in exactly the same way as a conventional inverter does.  Mark has been quite responsive with changes for the 3 phase inverters and multiple AIO's via the Gateway, so am sure that the EMS is possible, its just you may be the guinea pig.

Best to post a question on the givtcp github to ask about current support and testing status and whether anyone else is using the EMS.  I've not heard of anyone either here or on the GivEnergy forum.

However predbat can talk to the EMS via the GivEnergy cloud using the GE Cloud integration - see https://springfall2008.github.io/batpred/inverter-setup/

There are some reductions in data as the GE cloud only has 5 minute portal snapshots, but as long as the EMS is operable via the GivEnergy cloud and its API's, this should work.
I have tried using the GE Cloud integration for my hybrid inverter and it works fine. Again you may be the first to try this as the EMS is a new product.

I'm not even sure the EMS is formally launched yet.  I don't see it on either its technologies.shop or midsummer wholesale's websites

springfall2008, 2025-01-18:
Yes, gcoan is right - EMS is supported via GECloud integration but I've not seen a local one with GivTCP yet.

markbw999, 2025-01-18:
Thanks both. So to confirm, predbat will continue to work with my proposed setup, albeit with a slower response to changing data, and I'll have to reconfigure it to use the cloud integration, correct?

gcoan, 2025-01-18:
yes

you can in fact install the ge cloud integration, change your apps.yaml to use the ge cloud sensors rather than the givtcp ones and it will all work with your current AC3.

Might be worth you doing that to give you confidence ahead of the EMS.

markbw999, 2025-01-18:
Excellent, thanks!

markbw999, 2025-01-31:
Sorry guys, another fly in the ointment with my proposed expansion. The DNO insists on limiting my export capacity as follows...

"Adopt a G100 Issue 2 Export Limitation Scheme – Install 13.5KW of generation as the total installed capacity but within this, set the Agreed Export Capacity to 10KW or below. 
Under G100 you can connect up to 13.5KW of generation however the export to grid must be limited to 10KW or below via an export limitation device.
You can do this by adopting an Export Limitation Scheme  (ELS) which will require a limitation device, possibly a separate device to the AC inverters or embedded within. The owner is responsible for ensuring that the suitable protection installed adheres to the restrictions as stated in the ENA G100 documents for export. The chosen method, for reducing the export capacity, is also subject to SSEN approval before it may connect.   New Schematics and a completed ENA G100 application will be required.  (£360 Witness test fee)
Total Power Station capacity: 13.5KW
Total Agreed Export capacity: 10KW"

My installer says he can limit the export in the inverters without affecting the power delivered to the house. However, I still want to maximise my return by exporting any remaining battery capacity by the end of the day / start of the cheap rate period.

Does predbat support a maximum export rate separate from the total inverter power? I'm presuming it would need to plan for a longer export period if my rate is throttled.

johnwb87, 2025-01-31:
So the new inverter is 10kW? Given you said you existing is 3kW.

gcoan, 2025-01-31:
> Does predbat support a maximum export rate separate from the total inverter power? I'm presuming it would need to plan for a longer export period if my rate is throttled.

Yes and no

Predbat assumes that any export limiting is done on the inverter.  In apps.yaml you can configure what your maximum export rate is, and Predbat will model this appropriately in the plan, but Predbat doesn't control the maximum export rate.

So if your installer can configure the export limit in the inverters or the EMS then Predbat can model it.

markbw999, 2025-01-31:
> So the new inverter is 10kW? Given you said you existing is 3kW.

I'm adding another 3kw inverter to an existing 3kw AC coupled and a 7.5kw solar inverter so total 13.5kw.

markbw999, 2025-01-31:
> Yes and no
>
> Predbat assumes that any export limiting is done on the inverter. In apps.yaml you can configure what your maximum export rate is, and Predbat will model this appropriately in the plan, [...]

So if I do

inverter_limit:
    - 10000

in apps.yaml all should be good?

gcoan, 2025-01-31:
> So if I do
> 
> inverter_limit:
> - 10000
> 
> in apps.yaml all should be good?

No you should set inverter_limit to the maximum all your inverters can generate (solar and AC) and set export_limit to the export limit your DNO has imposed and your installer has configured in the inverters

https://springfall2008.github.io/batpred/apps-yaml/#inverter-control-configurations

johnwb87, 2025-01-31:
> > So the new inverter is 10kW? Given you said you existing is 3kW.
> 
> I'm adding another 3kw inverter to an existing 3kw AC coupled and a 7.5kw solar inverter so total 13.5kw.

Everything is AC coupled then. That’s unfortunate from a G99/G100 point of view.

closed 2025-02-01 by markbw999