springfall2008/batpred #1881: Charge hold doesn't work on Growatt SPH via Solar Assistant

tomkentpayne, 2025-01-19:
**Describe the bug**
In charge hold mode, the inverter carries on charging the battery past Predbat's target SOC

**Expected behaviour**
The Growatt SPH inverter would be put into "Battery first" mode, with "number.growatt_sph_battery_first_stop_charge" set to the max target SOC (or potentially instead have battery first mode and "number.growatt_sph_battery_first_charge_rate" set to 0W, but that may stop solar charging too, I'm unsure).

**Predbat version**

v8.11.1

**Environment details**

- Inverter and battery setup - Growatt SPH inverter, connected via Solar Assistant, using the standard MQTT entities that get created within Home Assistant. Growatt Solar Assistant template used in apps.yaml, with "spa" changed to "sph".
- Standard HAOS installer or Docker - HAOS running on a proxmox VM, Predbat installed as an add-on.

I believe this may be able to be fixed by changing [this line](https://github.com/springfall2008/batpred/blob/0aed1c1e0a1b34b6982c11921b397ed78e463405/apps/predbat/inverter.py#L2050) in `adjust_charge_immediate()` inverter.py - as currently it seems that if the target SOC is reached, but there is no `charge_freeze_service` defined, `charge_start_service` is called. Should this not be calling `charge_stop_service` instead?
Apologies if I've misunderstood how it's supposed to work.

tomkentpayne, 2025-01-20:
Update - I think I've worked around the issue with a pair of automations:

One that fixes it by mirroring the charge limit:

~~~yaml
alias: Mirror charge limit SOC (predbat)
description: ""
triggers:
  - entity_id:
      - sensor.predbat_sa_0_charge_limit
    trigger: state
conditions:
  - condition: not
    conditions:
      - condition: state
        entity_id: select.predbat_mode
        state: Monitor
  - condition: not
    conditions:
      - condition: state
        entity_id: switch.predbat_set_read_only
        state: "on"
actions:
  - action: number.set_value
    metadata: {}
    data:
      value: "{{ states('sensor.predbat_sa_0_charge_limit') }}"
    target:
      entity_id: number.growatt_sph_battery_first_stop_charge
mode: single
~~~

And another one to mirror the discharge rate (that isn't needed to solve this problem but I think may be needed in some scenarios?):

~~~yaml
alias: Mirror discharge rate (predbat)
description: ""
triggers:
  - entity_id:
      - sensor.predbat_sa_0_discharge_rate
    trigger: state
conditions:
  - condition: not
    conditions:
      - condition: state
        entity_id: select.predbat_mode
        state: Monitor
  - condition: not
    conditions:
      - condition: state
        entity_id: switch.predbat_set_read_only
        state: "on"
actions:
  - action: number.set_value
    metadata: {}
    data:
      value: "{{ states('sensor.predbat_sa_0_discharge_rate') / 3000 * 100 }}"
    target:
      entity_id: number.growatt_sph_free_mode_discharge_rate
  - action: number.set_value
    metadata: {}
    data:
      value: "{{ states('sensor.predbat_sa_0_discharge_rate') / 3000 * 100 }}"
    target:
      entity_id: number.growatt_sph_grid_first_discharge_rate
mode: single
~~~

tomkentpayne, 2025-01-20:
Initial problem (battery power still at 3000W charging during "Hold charging" predbat status):

[image]

After my workaround automations (battery power drops to 0W during "Hold charging" status):

[image]

tomkentpayne, 2025-01-20:
I had also added these fields to apps.yaml in addition to the ones in the solar_assistant_growatt.yaml template, but I don't think they fixed the issue (unless maybe they would've, if it hadn't been for predbat's charge/discharge rates being in W but the Solar Assistant versions wanting a percentage?):

~~~yaml
  charge_limit:
    - number.growatt_sph_battery_first_stop_charge
  charge_rate:
    - number.growatt_sph_battery_first_charge_rate
  discharge_rate:
    - number.growatt_sph_grid_first_discharge_rate
  load_power:
    - sensor.growatt_sph_load_power
~~~

tomkentpayne, 2025-01-20:
Looking at [the documentation on the inverter settings](https://springfall2008.github.io/batpred/inverter-setup/#has_target_soc) and [the Solar Assistant settings](https://github.com/springfall2008/batpred/blob/1e24d6ce9654dd7bbc8c0045d38069419ce60ed0/apps/predbat/config.py#L1367) I've spotted that it has `"has_target_soc": False` (meaning if `charge_limit` is defined in apps.yaml it'd be ignored). Would `charge_limit` be satisfied by `number.growatt_sph_battery_first_stop_charge`?
If so then the fix for this would be for [the "SA" inverter type](https://github.com/springfall2008/batpred/blob/1e24d6ce9654dd7bbc8c0045d38069419ce60ed0/apps/predbat/config.py#L1367) to have `"has_target_soc": False`, and the Growatt Solar Assistant template to have:
~~~yaml
  charge_limit:
    - number.growatt_sph_battery_first_stop_charge
~~~

xhemp, 2025-07-04:
Hey @tomkentpayne , I am using a Growatt SPA and mimic your configuration adjusting to my entity names. It looks to be working well. I have one error that crops up sometimes and just wondering if you see the same sometimes on your side?

`2025-07-04 16:10:01.641960: Completed run status Freeze exporting with Errors reported (check log)`

closed 2026-08-28 by chalfontchubby