springfall2008/batpred #1883: Uninstalling does not delete entities

madmic1314, 2025-01-20:
**Describe the bug**
Installed via add-on store, ran, configured, tested, decided it wasn't for my needs.
Uninstalled add-on, rebooted and all entities remain. Unable to delete them as they're not manageable by the GUI

How do I remove the entities please?

**Expected behaviour**
Uninstalling should delete all entities.

**Predbat version**
Latest - installed via add-on store - 1.22

**Environment details**
N/A

**Screenshots**
[image]

**Log file**
N/A

**Predbat debug yaml file**
N/A

madmic1314, 2025-01-20:
Quick update - https://community.home-assistant.io/t/how-to-remove-entities-without-a-unique-id-where-are-they/428405/48 I ran a Purge Entities and this cleaned them up.

gcoan, 2025-01-20:
I'll stick something in the documentation about this