springfall2008/batpred #1886: help please :-D

Ryuisnod, 2025-01-20:
hello,

I feel I've got everything set up correctly but my predbat plan doesn't include any charging or exporting, I have been reading through the Docs for about 2 days checking every setting, any help will be appreciated as im lost. The plan looks reasonable if it only charges my battery!

[image]

[image]

[image]

# ------------------------------------------------------------------
# This is an example configuration, please modify it
# ------------------------------------------------------------------
---
pred_bat:
  module: predbat
  class: PredBat

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
   - '£'
   - 'p'

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # If you are using Predbat outside of HA then set the HA URL and Key (long lived access token here)
  #ha_url: 'http://homeassistant.local:8123'
  #ha_key: 'xxx'

  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  geserial: 'hiden'
  

  # Sets the maximum period of zero load before the gap is filled, default 30 minutes
  # To disable set it to 1440
  load_filter_threshold: 30

  #
  # Sensors, more than one can be specified and they will be summed up automatically
  #
  # For two inverters the load today would normally be the master load sensor only (to cover the entire house)
  # If you have three phase and one inverter per phase then you would need three load sensors
  #
  # For pv_today if you have multiple solar inverter inputs then you should include one entry for each inverter
  #
  load_today:
    - sensor.givtcp_{geserial}_load_energy_today_kwh
  import_today:
    - sensor.givtcp_{geserial}_import_energy_today_kwh
  export_today:
    - sensor.givtcp_{geserial}_export_energy_today_kwh
  pv_today:
    - sensor.givtcp_{geserial}_pv_energy_today_kwh

  # Load forecast can be used to add to the historical load data (heat-pump)
  # To link to Predheat
  # Data must be in the format of 'last_updated' timestamp and 'energy' for incrementing kWh
  #load_forecast:
  #  - predheat.heat_energy$external
  #
  # If you enable ge_cloud_data then the load/import and export data will be fetches from the GE cloud instead of from GivTCP sensors
[... 539 lines cut ...]
    #  100: 0.84

    # Sets the efficiency of a heat pump in COP at different input temperatures
    heat_pump_efficiency:
      -20: 2.10
      -18: 2.15
      -16: 2.2
      -14: 2.25
      -12: 2.3
      -10: 2.4
      -8: 2.5
      -6: 2.6
      -4: 2.7
      -2: 2.8
      0: 2.9
      2: 3.1
      4: 3.3
      6: 3.6
      8: 3.8
      10: 3.9
      12: 4.1
      14: 4.3
      16: 4.3
      18: 4.3
      20: 4.3

gcoan, 2025-01-20:
Very strange issue, never seen that before, the car charges but not the battery.

The way you have pasted your apps.yaml into the issue I can't see whether there is a problem with the file layout, but probably not.

Looking at the plan I see you have set best_soc_keep to 1.5, _min to 1.5 and _max to 20.9 - I would start with checking those first.
https://springfall2008.github.io/batpred/customisation/#battery-margins-and-metrics-options

If you set best_soc_min then predbat will always ensure you have that amount in your battery, so if your house load exceeds the prediction and the SoC drops below 1.5 then predbat will charge the battery *regardless of what the import rate is*.   Unless you need to keep something back for an EPS and you have lots of power cuts, set this to zero.

Can set best_soc_keep if you want, this is a soft constraint, but personally I have mine set to 0 also.

best_soc_max I have never seen it set by anyone, how did you come up with this value?  This may be the cause of your weird plan as it looks like it could be set to more than your inverter can deliver.  Again mine is set to 0.

springfall2008, 2025-01-20:
best_soc_max will break stuff, that sets the maximum charge level which means you are basically preventing any meaningful charge - like GCoan says set it to 0.

Ryuisnod, 2025-01-20:
sorry about the format, you should see the first edit, had too remove all the "#" as it was turning all the comments into `titles` but I've now noticed there is a code button so I've sorted it.

sadly i had all these things at 0 already and the issue was the same, i was mucking about trying to make things happen, ill put them all back to 0.

one thing to note is that i do have the 3-phase inverter and batteries. i didn't have "sensor.givtcp{geserial}battery_power" as an entity so I created it with a helper, which seems to be working from what I can see as the stats look fine, ill paste code below

 ```
- platform: template
   sensors:
    givtcp_xxxxxxx_battery_power:
      friendly_name: "Battery Power"
      unit_of_measurement: "W"
      device_class: power
      value_template: >
          {% set charge = states('sensor.givtcp_xxxxxxxx_battery_charge_power') | float(0) %}
          {% set discharge = states('sensor.givtcp_xxxxxxx_battery_discharge_power') | float(0) %}
          {{ discharge - charge }}

```

gcoan, 2025-01-20:
I'd recommend you change the template to being a time based trigger template.  The problem with the one you have is that every single change in battery charge or discharge power, even by a single watt will result in a new state record being generated for your battery power entity.

You'll fill your database up with unnecessary guff as well as slow HA down a bit.

Here's a similar one I have for PV power so I can see total PV power vs Solcast forecast.  I trigger this every 3 minutes but I have other sensors for things like net energy today on slower frequencies like 5 or 15 minutes:

```
# Total Solar power generation sensor, updated every 3 minutes
- trigger:
    - platform: time_pattern
      minutes: "/3"
  sensor:
    - name: "Total Solar Power"
      unique_id: "total_solar_power"
      unit_of_measurement: kW
      device_class: power
      state_class: measurement
      state: >
        {{ (states('sensor.g_xxxx_pv_power')|float(0) 
          + states('sensor.h_xxxx_pv_power')|float(0)
          + states('sensor.fit_solar_power')|float(0) )
          | round(1)
        }}
```

Ryuisnod, 2025-01-20:
Thanks for that I'll change the sensor, do you have any ideas to why I don't have charge or export on my plan?

gcoan, 2025-01-20:
Another copy of the plan would be useful, and if you can attach the apps.yaml file (just attach is straight into github, don't need to copy/paste it), but in the logfile I see this:

```
2025-01-20 19:30:04.779445: Today's predicted so far 31.22 kWh with 20.63 kWh car/iBoost excluded and 6.49 kWh import ignored and 0 forecast extra.
2025-01-20 19:30:04.779488: Today's actual load so far 34.01 kWh with 16.3 kWh Car/iBoost excluded and 4.6 kWh import ignored.
```

So basically its saying that most of your load today is from charging the car/iboost, leaving you with very little actual house load consumed.

Is this correct?  It might be why predbat sees no value in charging the battery

Ryuisnod, 2025-01-20:
[image]

[image]

[image]

[image][

[apps.txt](https://github.com/user-attachments/files/18482982/apps.txt)

](url)

gcoan, 2025-01-21:
You can comment all the GivTCP2 lines out (or remove them)

Looking at the yaml I can see at least two examples where you have commented out the label but not the sub items. This may cause problems, eg:

```
  # car_charging_now_response:
    - 'yes'
    - 'on'
    - 'true'

  # One per car, when true only one car can charge at once, when False multiple cars can charge at once
  #car_charging_exclusive:
   - True
```

gcoan, 2025-01-21:
Looking further I see other formatting errors in apps.yaml where sub-nodes are not indented two spaces, e.g.:

```
  pause_mode:
   - select.givtcp_{geserial}_battery_pause_mode
   - select.givtcp_{geserial2}_battery_pause_mode

  # Not all firmwares support pause start/end time, delete these if not supported
  # to avoid spurious writes/warnings
  pause_start_time:
   - select.givtcp_{geserial}_battery_pause_start_time_slot
   - select.givtcp2_{geserial2}_battery_pause_start_time_slot
  pause_end_time:
   - select.givtcp_{geserial}_battery_pause_end_time_slot
   - select.givtcp2_{geserial2}_battery_pause_end_time_slot

  #Â Battery temperature sensor per inverter, used outside REST mode to get current temperature
  #
  battery_temperature:
   - sensor.givtcp_battery_stack_1_bms_temperature
   - sensor.givtcp2_battery_stack_1_bms_temperature

  rates_export:
   -  rate: 15
```

Another thing I notice is you have days previous set to just 4, so Predbat will look at your history from 4 days ago and that's it.  Its not the cause of this but it does mean any unusual load on that day will be what predbat plans against.
More normal to use say day 2 to day 8,  or day 7 and 14 (but you will need to increase history retention in configuration.yaml)

WyndStryke, 2025-01-21:
> More normal to use say day 2 to day 8, or day 7 and 14 (but you will need to increase history retention in configuration.yaml)

Slightly of an aside here, but I seem to recall that history can also be stored in the form of 'long term statistics'?  Never looked into it, but if that's the case, then it would give a richer long-term history for this sort of thing.  So rather than just looking at 'last Tuesday' or whatever, it could be based on a weighted average of Tuesdays going back further in time.

I did very recently increase my history retention to 22 days (to allow 7/14/21 to be picked up by predbat), but I suspect that might cause me data volume & performance issues.  So perhaps a long-term-statistics solution would be a better approach in the long run.

gcoan, 2025-01-21:
> Slightly of an aside here, but I seem to recall that history can also be stored in the form of 'long term statistics'?

In HA there are three types of history:

1. sensor history, raw values, every state change
2. short term statistics, summarised sensor history giving max, min average every 5 minutes 
3. long term statistics, summarised to hourly

Predbat uses number 1.  The short term statistics would probably be granular enough but at the moment its not what predbat uses.

And yes, keeping lots of history can bloat your database.  I have been doing a lot of work on improving this and will update the predbat documentation with some advice soonish

Ryuisnod, 2025-01-22:
I've done all this and chanced to the dev givtcp and have a slight improvement

could this be part of the issue? 

Warn: record_status Inverter 0 unable to read REST data from http://homeassistant.local:6345/readData - REST will be disabled

there is inverter data at that address

[image]

gcoan, 2025-01-22:
As far as I know there's only been one other person trying to get Predbat working with a 3 phase inverter, @leefarm11 and he raised and has had progressed a number of issues with both givtcp and predbat

Predbat issues include #1426 #1607 #1621 #1625 and #1630

I know that there have been dev changes made to givtcp to improve the inverter support.  I think its still likely to be trial and error working out what's wrong and fixing it.  On the face of it your predbat appears to not be having any issues in reading the data but then you'd think it would devise a sensible plan?

Ryuisnod, 2025-01-22:
Yes I have been reading his posts on both predbat and givtcp and tagging
him in was probably my next step so thanks for that.

@LeeFarm11 do you think you would be able to help me? Have you got it
working correctly?  Or at least the plan working correctly?  I feel of I
can get the plan correct I can automate my way through the rest as I have
automations working for charge and and export already

LeeFarm11, 2025-01-22:
I am not very happy with where I am. I'm not happy with GivEnergy, and not happy that a year down the road the 3 phase Inverter is still not 'supported' by GivTCP. I would like to offer @britkat1980 more support, but I know he is busy. I believe he did spend another day at GivEnergy HQ recently, but I don't know how much progress he made with understanding and interfacing with the 3 phase Inverter. I would like to offer him to come to mine and play with my Inverter as much as he likes in an effort to move forward. 

Until then, my predbat plan continues to work and my automation script continues to handle the charging and discharging, but I suspect it will be much less successful if we ever get any significant sunshine.

I have concerns about the solar forecast data, but not had time to investigate properly.

I have not spent much time on it recently, too much other stuff going on. But -
for me I have REST **disabled** in apps.yaml

```
  # When set use the REST API rather than HA entity for control, should be more reliable/faster to control
  # Set one per inverter
  # If using Docker then change homeassistant.local to the Docker IP address
  #givtcp_rest:
  #  - 'http://192.168.1.204:6345'
```

Latest version of my automation is here -

```
alias: PredbatStatusChangeAutomation2
description: Modify Inverter Settings depending on predbat.status
triggers:
  - trigger: state
    entity_id:
      - predbat.status
    attribute: last_updated
conditions: []
actions:
  - if:
      - condition: or
        conditions:
          - condition: state
            entity_id: predbat.status
            state: Charging
          - condition: state
            entity_id: predbat.status
            state: Freeze charging
          - condition: state
            entity_id: predbat.status
            state: No charge
    then:
      - action: switch.turn_on
        metadata: {}
        data: {}
        target:
          entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
      - action: switch.turn_off
        metadata: {}
        data: {}
        target:
          entity_id: switch.givtcp_td2343g049_force_discharge_enable
      - action: switch.turn_on
        metadata: {}
        data: {}
        target:
          entity_id: switch.givtcp_td2343g049_force_charge_enable
    else:
      - if:
          - condition: or
            conditions:
              - condition: state
                entity_id: predbat.status
                state: Discharging
              - condition: state
                entity_id: predbat.status
                state: Freeze discharging
              - condition: state
                entity_id: predbat.status
                state: Exporting
        then:
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
          - action: switch.turn_on
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_discharge_enable
          - action: switch.turn_off
            metadata: {}
            data: {}
            target:
              entity_id: switch.givtcp_td2343g049_force_charge_enable
        else:
          - if:
              - condition: or
                conditions:
                  - condition: state
                    entity_id: predbat.status
                    state: Read-Only
                  - condition: state
                    entity_id: predbat.status
                    state: Hold charging
            then: []
            else:
              - action: switch.turn_off
                metadata: {}
                data: {}
                target:
                  entity_id: switch.givtcp_td2343g049_force_ac_charge_enable
              - action: switch.turn_off
                metadata: {}
                data: {}
                target:
                  entity_id: switch.givtcp_td2343g049_force_charge_enable
              - action: switch.turn_off
                metadata: {}
                data: {}
                target:
                  entity_id: switch.givtcp_td2343g049_force_discharge_enable
mode: single
trace:
  stored_traces: 50

```

gcoan, 2025-01-22:
> for me I have REST disabled in apps.yaml

I was going to suggest that might be an improvement. 

Bear in mind that Predbat still uses REST to obtain the inverter information from givtcp, but you have the choice as to whether to use REST to control the inverter or use Home Assistant controls.  Commenting the REST line out means Predbat uses the givtcp HA controls which is one thing less to have to be working right

Glad to hear you have made some progress @LeeFarm11 and it is working albeit with automation interventions to get around the givtcp control deficiencies.

Ryuisnod, 2025-01-22:
Could you share your apps.yaml for comparison please? A d anything else you
might find useful?

Ryuisnod, 2025-01-27:
> I am not very happy with where I am. I'm not happy with GivEnergy, and not happy that a year down the road the 3 phase Inverter is still not 'supported' by GivTCP. I would like to offer [...]

Could your share your apps.yamland tour config settings? All the numbers in my plan look about right to me but even when battery empty and electricity at 7p it will still not plan to charge. All sensors look right too so I must have something wrong

britkat1980, 2025-01-27:
@lee can you try the latest dev version, which has my updates from time at GivEnergy HQ last month. It should fix some of the control issues you have experienced. I’d be interested in your feedback.

LeeFarm11, 2025-01-28:
Hi @Ryuisnod 

[apps(2).yaml.zip](https://github.com/user-attachments/files/18570264/apps.2.yaml.zip)

GivTCP Configuration. I see I have set Inverter 1 to only show Battery data. I guess that is wrong. Will try to test what difference it makes later. EVC is disabled for now to save complications. Tariffs are wrong, but predabat is set with correct rates.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Ryuisnod, 2025-01-28:
Thanks for this is have a good look through in comparison to mine

Ryuisnod, 2025-01-28:
@LeeFarm11 Ok I seem to have it going now thanks for that. 

@britkat1980 I'm using the givtcp dev fork. Predbat is able to charge and export natively without automations. 

I didn't have predbat working before so I can't compare but I had made my own automations for charge and export and it required flicking a lot of switches but that seems to not be the case anymore.

Only things that haven't worked natively with predbat is battery power as there isn't an entity for that and the formatting of the entity names being different. 

If there is any other information you want or things I can test from my 3-phase setup just let me know. Thanks for you time and effort on this

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

Ryuisnod, 2026-08-20:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

You can just close it. Thanks

closed 2026-08-20 by gcoan