springfall2008/batpred #1891: Saving Session Plan

Meatballs1, 2025-01-21:
With the saving session import rate being 49p, and 39p export, I would expect PB to hold charge (or force charge) so it can fully cover and export for the full 1 hour slot. It instead covered the load but was expecting to run out (soc hits 20% at 16:30) before the saving session slot (initial plan).

When I was going to grab the debug it updated, now wanting to force charge, but only export for 30 min timeslot. If it is going to force charge and export for one timeslot would expect it to do enough to cover the full hour?

**Predbat version**

8.11.2

**Environment details**

Sunsynk 5kW with 20.5kWh battery min SOC 20%
Docker

**Screenshots**
Initial plan:

[image]

Plan slightly later:

[image]

**Log file**

[predbat.log](https://github.com/user-attachments/files/18492479/predbat.log)

**Predbat debug yaml file**

This is the debug for the later plan.
[predbat_debug_21_01_25.yaml.txt](https://github.com/user-attachments/files/18492455/predbat_debug_21_01_25.yaml.txt)

Meatballs1, 2025-01-21:
Slightly later plan I received after enabling debug:

[image]

[2nd_predbat_debug_21_01_25.yaml.txt](https://github.com/user-attachments/files/18492691/2nd_predbat_debug_21_01_25.yaml.txt)

Meatballs1, 2025-01-21:
Disabling Tweak it went for full export (I also reduced minimum export improvement to 2p)

[image]

Still would think it would best best off holding up to that period so it can fully export for the full slot?

springfall2008, 2025-01-21:
I've made a new release, the plan should have had a hold but it was removed later due to a bug

Meatballs1, 2025-01-21:
I followed the herd and jumped onto agile for tomorrow. It doesn't seem to want to hold even still on main, with the 92p available. I only have history of 1 so it thinks the 70p slots will be low usage due to todays saving session - so it cant be due to the load in those slots.

[predbat_debug-22_01_25.yaml.txt](https://github.com/user-attachments/files/18497500/predbat_debug-22_01_25.yaml.txt)

[image]

gcoan, 2025-01-21:
> I only have history of 1 so it thinks the 70p slots will be low usage due to todays saving session - so it cant be due to the load in those slots.

Is there a particular reason why you have a history of 1 day?  Its recommended that you don't do this and instead average several days history.

If you need to adjust your load history then you could put an import_rates_override in apps.yaml with the date and times and a load adjustment.  It doesn't need a rate adjustment, you can just add a % scaling factor to the load.

Interesting to join agile outgoing for tomorrow's peak rates.  I just won't have enough spare SoC to do similar, will need it all to run the house and heat pump and in fact the heat pump will have to be off or turned down for a lot of the day as its so expensive.  Hey ho

Meatballs1, 2025-01-21:
I'm on Go, not switching to Agile import so better off not to cover the load at ~25p and export for 70-90p in my head.

[image]

Load history is 1 as I've not dialed in predheat accurately yet or attempted predai and HP is biggest consumer and weather is more likely to be the same as the day before than it is 7 days ago. Tbh it hasn't mattered that much until now as I would charge overnight, and be discharged fully by HP by lunchtime, and on a flat tariff it hasn't really needed to hold charge for anything. With saving sessions and agile outgoing things are a bit more interesting.

gcoan, 2025-01-22:
fair logic.  I have mine averaging days 2-8 but as you say with a heat pump its all a bit immaterial as the heat pump sucks the batteries dry anyway.

I'm on Agile import so Predbat freeze charges through the day and keeps the batteries for the peak period and that works fine, except on days like tomorrow that's a tough ask.

springfall2008, 2025-01-22:
@Meatballs1 you have 'set_charge_freeze' disabled so the hold feature isn't turned on, not sure why you have it set that way?

springfall2008, 2025-01-22:
Oh wait its due to being SK inverter and it not being supported

raldred, 2025-01-22:
Just in-time to for todays session :) thanks @springfall2008

closed 2025-01-25 by springfall2008

springfall2008, 2025-03-15:
Re-visiting this I figured out why the saving session wasn't charging as much as it should and made a new fix