springfall2008/batpred #1892: Documentation error - Octopus Saving Sessions sensor.

johnwb87, 2025-01-21:
**Describe the bug**
I couldn't understand why predbat wasn't auto joining today's Saving Session. I checked apps.yaml templates and documentation to confirm I have the right sensor set. Those sources say binary.sensor related to the Octopus Account. I was using the templated regular expression to auto detect.

As I could see that the related event.entity and its attributes were showing the event details I changed the sensor to that in apps.yamland auto-join worked.

**Expected behaviour**
Documentation and templates to align with correct sensor.

**Predbat version**

8.11.2

johnwb87, 2025-01-21:
Hmm. Something else going on then. That entity isn’t regularly updating. I’ll put it back after the session and see what happens next time.

It was immediate in the logs though as soon as predbat restarted I can see it joining the session with the other rates and Octopus lines.

Opt in confirmed in my Octopus account.  

This is my first session with predbat and the api fully working so I’ve not had a running system for this aspect in the past.

gcoan, 2025-01-21:
Can you check the name of your octopus saving session event?

According to the documentation https://bottlecapdave.github.io/HomeAssistant-OctopusEnergy/entities/octoplus/ the event should be named event.octopus_energy{{ACCOUNT_ID}}octoplus_saving_session_events

Mine is and that's why Predbat is able to find the event details based on the binary sensor name in apps.yaml.

One person on the GivEnergy community has a similar problem and his event is named event.octoplus_saving_session_events_{{ACCOUNT_ID}} which would explain why Predbat can't find it and auto-join the session

tim-b2, 2025-01-22:
My event is also showing up as event.octoplus_saving_session_events_{{ACCOUNT_ID}} and predbat didn't find the session yesterday. 
The octopus integration binary sensor is in the same format as the predbat yaml (3binary_sensor.octopus_energy([0-9a-z_]+|)_saving_session(s|))

The octopus integration did auto sign up for the session correctly but never made it into predbat plan. 
Currently on Octopus integration v13.4 and predbat v8.10

gcoan, 2025-01-22:
@tim-b2   If you rename the event to match what it should be in the Octopus integration documentation, I expect it will work OK.

Could well be another DFS session today, Agile rates are sky-high

johnwb87, 2025-01-22:
So that seems to be my problem then my event is event.octoplus_saving_session_events_a_xxxxxxxx).

Will change that and go from there.

Thanks for the help. I will change the issue name to help other and the documentation tag can be added.

johnwb87, 2025-01-22:
Will raise an issue over on the Octopus integration as well. Perhaps it is something in a specific release that people may have installed.

gcoan, 2025-01-22:
I've added an extra highlight to the documentation of what the event name should be, and that if it is mis-named then Predbat won't be able to auto-join the saving session

johnwb87, 2025-01-22:
For interest NESO have issued a DFS requirement for today. Another hour or so to see if Octopus bid successfully.

johnwb87, 2025-01-22:
We are on for DFS today 16:30-17:30. I predict the price will be 56p/kWh.

I pulled the NESO published info for the past sessions and there is a consistent ratio between the published price and what Octopus gives us.

Lets see if my changed entity name fixes yesterday's issue.

tim-b2, 2025-01-22:
> [@tim-b2](https://github.com/tim-b2) If you rename the event to match what it should be in the Octopus integration documentation, I expect it will work OK.
> 
> Could well be another DFS session today, Agile rates are sky-high

That worked perfectly, thanks.

johnwb87, 2025-01-22:
Do we understand why predbat uses the binary.sensor but then uses that to find the event entity? Is there a functional reason or is it more about how the entities from the Octopus integration have evolved?

gcoan, 2025-01-22:
possibly caused a bit by evolution in the Octopus integration because the use of events came in some time in the last year.  Predbat does this for other things, the import and export rate sensors from apps.yaml it also uses those names to then find the matching event names where the actual rates are

Predbat is looking for the correct event names as per the octopus integration documentation, its just for a couple of people their event names is not correct.  Why I don't know, the vast majority of people (including new installs) don't have any such issue.  Maybe the integration changed the naming convention at some point 🤷‍♂️

closed 2025-01-31 by gcoan