springfall2008/batpred #1895: Switching between Agile and fixed on Octopus

TheSargey, 2025-01-22:
Morning folks,

Just a quick question for the more astute of you. With the Agile pricing currently all over the shop, I imagine a lot of people have switched to a more consistent tariff rather than sticking with Agile.

Is there a process I should keep in mind when changing from a dynamic pricing model like Agile to say something like Intelligent Octopus Go? Looking at apps.yaml it would seem that, due to the octopus_intelligent_slot, it should automatically pick up when the underlying account changes or am I confused as to what this sensor is actually doing?

Any pointers before I make the switch (and potentially break something) would be very much appreciated.

All the best,

Sarge

gcoan, 2025-01-22:
When you change tariff the Octopus integration automatically feeds the new tariff rates through to Predbat and you don't need to do anything.   I saw this when I swapped from Flux to Agile.

Do take a look at the general guidance for specific tariffs https://springfall2008.github.io/batpred/configuration-guide/

If you are moving to IOG though and want to take advantage of the additional IOG charging slots you will need to setup the car charging appropriately, presumably been on predbat led charging before

There are changes to be made to apps.yaml for the extra sensors.  Make sure you follow the car charging setup guide https://springfall2008.github.io/batpred/car-charging/

TheSargey, 2025-01-22:
Hi Geoff,

Thanks as always for the speedy response.  I'm not looking to setup car
charging as I don't have an EV but I'd assume it would work the same if I'm
just going to use it to charge my batteries instead.

I'll take a look through the config guides, thanks again for your help.

Cheers,

Sarge

gcoan, 2025-01-22:
Yes Predbat will work just the same charging your batteries etc,  its just recommended you check the default settings for IOG as to whether they are different from what you had before.

The car charging on IOG is just a bit more complex to setup, that's all

closed 2025-02-10 by TheSargey