springfall2008/batpred #1897: Hold charge is charging at full whack

chris-y, 2025-01-22:
**Describe the bug**
Predbat has decided to hold charge at an amount lower than the current charge.  However it has instead operated the inverter to charge at 72A

**Expected behaviour**
Inverter should be charging at 0A, or Predbat should be waiting for the charge to drop to the hold charge amount and then charging at 0A.

**Predbat version**

8.11.4

**Environment details**

- HAOS install with add-in
- Solis inverter using scripts to control

**Screenshots**

[image]

[image]

**Log file**

[predbat.log](https://github.com/user-attachments/files/18505653/predbat.log)

**Predbat debug yaml file**

[predbat_debug_12_50_00.yaml.txt](https://github.com/user-attachments/files/18505678/predbat_debug_12_50_00.yaml.txt)

chris-y, 2025-02-12:
So it looks like there has been some improvement since I raised this.

[image]

[image]

So now when is is "charging" to a lower level it's actually holding at the higher level.

I think it should still be just on demand until it gets down to near the level it is supposed to be holding at?

112233mark, 2025-02-14:
On version v8.11.5 and same Hold Charge was charging at 5.5Kw when battery was at 89% and predicted to be 100% from solar in 1.5 hours.

chalfontchubby, 2026-09-17:
Could you confirm whether this is still an issue? A lot has changed and in order to debug abything now we would need a new debug.yaml please.

If it's fixed (or we don't hear back to the contrary) I'll close this ticket in around a week so we can focus on live issues. Thanks.

chris-y, 2026-09-17:
I haven't noticed this for quite a while, so suspect it has been fixed.

chalfontchubby, 2026-09-17:
I'll close then - open a new one if you need to. Thank

closed 2026-09-17 by chalfontchubby