springfall2008/batpred #1902: Added input variable to set battery_charge_rate maximum to reduce potential failure of household electricity supply

DrJohnM61, 2025-01-24:
**Is your feature request related to a problem? Please describe.**
I have installed a second AIO and they now pull 12kW from the grid to charge. The intelligent Octopus Go randomly will charge during the same period at 7.5kW. A combination of other devices (dish washer, washing machine, hot water heater etc) will exceed my grid fuse of 100 amps. If I manually set the Inverter Charge Power Percent, then the AIO gateway charge rate reduces accordingly, however, predbat will always reset it to 100% at each re-evaluation run. I have not found a way to reduce the AIO charging rate (kW) without Predbat constantly resetting to the maximum.

There is no need for the AIOs to charge at 12kW as the low rate is for 6 hrs and the battery capacity is 27kWh

**Describe the solution you'd like**
A selector to set the maximum battery_charge_rate -- or -- read the rate as set in the inverter and if greater than zero, use the read number as the new next set number

**Describe alternatives you've considered**
Write an automation to pause predbat and take control of the charging process of the AIOs during the 6 hour intelligent Octopus Go period, however, Octopus may also charge the car outside of that window so that may also randomly clash with a call to charge the AIOs

**Additional context**
This should not be a hard set number (parameter in the configuration file) to accommodate the need to rapidly charge in the event of a high probable grid failure (such as the upcoming Storm Éowyn due to hit the UK tomorrow).

johnwb87, 2025-01-24:
You can set inverter_max_rate_charge in apps.yaml which would be permanent.

Or you could try low power charge which is more dynamic and allows the full rate to be accessible.

It would be expected behaviour for predbat to try charge in any extra IOG slots as the price is relatively so much lower.

I hope those ideas help.

DrJohnM61, 2025-01-24:
Thank you John for your comments.

Where do I find the "inverter_max_rate_charge" variable in apps.yaml? I have checked a number of examples in the repository (in case my apps.yaml is out of date) and cannot find such an item. I have added:

`inverter_max_rate_charge: 8000`

but it made no difference.

I also changed the reading of the battery_charge_rate to a fixed number, but that does not change anything as this section is ignored if the rest api is used.

`#charge_rate:`
 `#  - number.givtcpgw_{geserial}_battery_charge_rate`
  `charge_rate: 8000`
 
Please note that this is the GivEnergy gateway that controls parallel All-In-Ones. Changing the charger is not an option.

gcoan, 2025-01-24:
@DrJohnM61 
its inverter_limit_charge you need to set in apps.yaml and others with multi AIO's have used it quite successfully 

described in the documentation https://springfall2008.github.io/batpred/apps-yaml/#inverter-control-configurations

do not change the charge_rate control, that is what Predbat uses to set the charge rate your gateway will use

DrJohnM61, 2025-01-24:
Thank you @gcoan and I am kicking myself that I did not find that in the documentation. That has set the max rate as described. I also see that there is a `select.predbat_manual_api` call that I can use to change this value dynamically.

Thank you aging for the solution.
``

closed 2025-01-24 by DrJohnM61

johnwb87, 2025-01-24:
Hi. Sorry. I didn’t double check the entity. But yes that is the one I was thinking of.

Quite powerful to set it using an automation but predbat just can’t plan with that.

If you are on Intelligent Octopus Go I would try a few days with charge low power mode and see if that gets you closer to the behaviour you hope for.

gcoan, 2025-01-24:
No problem @DrJohnM61 there is a lot of documentation so not always easy to find things.  I know, I've been trying to restructure and rewrite the documentation over the last year.

Yes the predbat_manual_api can be used to dynamically change this, I'm going to be putting an example of how to use this, but here's a sneak preview

I find the easiest way to use the manual API is to write an automation script then you can invoke the script when you want to.  Here's an example changing the predicted house load:

```
alias: Send command to change Predbat predicted house load to Predbat manual API
sequence:
  - action: select.select_option
    data:
      option: >-
        rates_export_override?date={{          
        states('input_datetime.octopus_saving_session_date')| as_timestamp |
        timestamp_custom('%Y-%m-%d') }}&start={{          
        states('input_datetime.octopus_saving_session_start_time') }}&end={{
        states('input_datetime.octopus_saving_session_end_time') }}&load_scaling={{
        states('input_number.predbat_load_scaling_saving')|float }}
    target:
      entity_id: select.predbat_manual_api
mode: single
description: ""
```

DrJohnM61, 2025-01-24:
@gcoan I note that you are going to write some examples for the documentation and as such, I just wanted to comment  that these API calls appear to be persistent. I had written the following simple automation (using a light switch to rigger) to test:

```
alias: Set Battery Charge Rate Limit
description: ""
triggers:
  - type: turned_on
    device_id: aa5cf5591ce34a4ed0c32fd7697f58d6
    entity_id: 9fb0686a1d5c53f4392d0cfbe48b3cd2
    domain: light
    trigger: device
conditions: []
actions:
  - action: select.select_option
    metadata: {}
    data:
      option: "inverter_limit_charge=10000"
    target:
      entity_id: select.predbat_manual_api
mode: single
```

I then changed the value to 8000 and re-triggered, but no value change was seen. Changing the value to 7000 in the yaml and restarting predbat also had not effect. I see that the documentation says that there is a function to turn these off by adding [...] so adding `option: "[inverter_limit_charge=10000]"` turned off the 10000 override but did not change the setting to the 7000 defined in the apps.yaml. What it did was set the value to 8000, the value of the second test. Running the automation with `option: "[inverter_limit_charge=8000]"` then cleared that and reverted to the yaml setting of 7000.

There is the option to clear all overrides `option: "off"`, but you may be setting two different variables for discrete reasons.

It may be important to stress the unsetting of the override in your example as it contains sensors to define the rates etc and may be difficult to specifically cancel. This worked for me (as a test automation):

```actions:
  - action: select.select_option
    metadata: {}
    data:
      option: "off"
    target:
      entity_id: select.predbat_manual_api
  - action: select.select_option
    metadata: {}
    data:
      option: inverter_limit_charge=8000
    target:
      entity_id: select.predbat_manual_api```

gcoan, 2025-01-24:
From what I understand I think you will be OK with the way Predbat works

If you set the manual_api_control to set an inverter _limit_charge then you are correct, if you want to cancel that override so that predbat defaults to the apps.yaml value then you need to set manual_api_control to [off]

What but I expect will happen is that Predbat won't instruct the inverter with the new charge limit until it next wants to do a charge,  which could either be at max rate or low rate depending on whether you have low rate charge set on or not.

At that point Predbat will check what the current inverter charge limit is, and then if its not what Predbat needs it to be, it will change it on the inverter (gateway in your case).  This way Predbat only sends control commands to the inverter when it needs to, minimising inverter registry writes

springfall2008, 2025-01-24:
The charge rate is reset every Predbat cycle so you won't need to wait