springfall2008/batpred #1908: Predbat not rolling prices correctly at 11pm & 11:30pm

gcoan, 2025-01-26:
**Describe the bug**
11:00pm and 11:30pm prices are being taken incorrectly from 10:30pm Agile slot.

**Expected behaviour**
It is 9:54am Sunday morning.  Agile prices available up to 10:30pm tonight.  Norddata prices not yet available so 11:00pm and 11:30pm prices tonight should be rolled over from Saturday 11:00pm and 11:30pm.

**Predbat version**
8.11.5

**Environment details**
HAOS 2025.1.4

**Screenshots**
Agile prices showing Saturday 11:00 and 11:30pm prices that should be rolled over:
[image]

Predbat plan showin incorrect prices for Sunday evening:
[image]

**Log file**

gcoan, 2025-01-26:
10:12am, now that the norddata prices have arrived, predbat is correctly showing those predicted prices in the 11:00 and 11:30 slots, so that bit of the logic is fine

[image]

[predbat.log](https://github.com/user-attachments/files/18549612/predbat.log)

springfall2008, 2025-01-26:
Ideally a debug YAML would be useful for this one, maybe you can reproduce by turning off nordpool ?

gcoan, 2025-01-26:
No problem, I did wonder if you would be able to investigate it just from the logfile.  Probably just an index incorrectly iterating

Turned norddata off in apps.yaml

Plan again shows the issue
[image]

Debug file
[1045.txt](https://github.com/user-attachments/files/18549692/1045.txt)

Logfile with debug
[predbat.log](https://github.com/user-attachments/files/18549696/predbat.log)

springfall2008, 2025-01-26:
Actually managed to reproduce it myself

springfall2008, 2025-01-26:
PR is here: https://github.com/springfall2008/batpred/pull/1911

The question becomes is it really more accurate to use 24 hours ago, given the rates have dropped so much?

gcoan, 2025-01-26:
> PR is here: https://github.com/springfall2008/batpred/pull/1911
> 
> The question becomes is it really more accurate to use 24 hours ago, given the rates have dropped so much?

In this particular case then yes, the repeated 11pm 2.65p rate is probably a better prediction, and that was how I spotted it because I didn't think there had been such cheap rates overnight last night so had to reconfigure the Agile rate card to check (I don't normally look at the historic rates).

But its not how Predbat is supposed to work so hence why I raised it.  Minor in the grand scheme of things but thought you would want to fix it.

springfall2008, 2025-01-26:
Fix now released

gcoan, 2025-01-26:
Thanks Trefor, I'll close this

closed 2025-01-26 by gcoan