springfall2008/batpred #1915: Predbat grid charing one inverter when in Freeze Charging mode

gcoan, 2025-01-26:
**Describe the bug**
Predbat status is freeze charging with limit 17% and has been for the entire 30 minute slot.
One inverter with SoC 23% has enable charge schedule turned on, the other inverter with SoC 4% has enable charge schedule turned off.
Possibly could understand it if the inverter with the lower SoC was being charged up to the SoC limit, but it was the inverter with the higher SoC that was already above the SoC limit.
I did turn enable charge schedule off on that inverter in case it had been left on accidentally, but Predbat turned it back on again so it was deliberate on Predbat’s part.

**Expected behaviour**
With no solar I would have expected freeze charging to just hold the SoC as-is, not for the inverters to have charge schedule enabled.

Based on the current import rate for the slot and the rate for subsequent slots I don’t believe there is any benefit to Predbat importing in this slot.

**Predbat version**
8.12.0

**Environment details**
HAOS 2025.1.4
2x Gen 1 hybrid inverters each with their own battery

**Screenshots**
Predbat plan showing Freeze charge planned for this slot:
[image]\

Predbat status set to freeze charge for the entire slot:
[image]

Inverter controls and power flows, showing charge schedule enabled for inverter G and battery charging, but not for H:
[image]

Enable charge schedule for G showing it was turned on by Predbat at the start of the slot, I turned it off, then Predbat turned it back on again:
[image]

Enable charge schedule for H showing no Predbat actions since lunchtime.  I turned it on thinking that both inverters should have been charging, then turned it off again:
[image]

**Log file**
[predbat.1.log](https://github.com/user-attachments/files/18551537/predbat.1.log)
[predbat.2.log](https://github.com/user-attachments/files/18551536/predbat.2.log)
[predbat.3.log](https://github.com/user-attachments/files/18551535/predbat.3.log)

**Predbat debug yaml file**
[1950.txt](https://github.com/user-attachments/files/18551541/1950.txt)

springfall2008, 2025-01-27:
Ignore the PR tagged here I used the wrong ticket ID.

Will take a look soon

gcoan, 2026-08-20:
old issue, closing

closed 2026-08-20 by gcoan