springfall2008/batpred #1921: Error: Exception raised '>' not supported between instances of 'NoneType' and 'float' after update

tractor137, 2025-01-27:
**Describe the bug**
Error: Exception raised '>' not supported between instances of 'NoneType' and 'float' after updating to v8.12.0. This meant planned export didn't happen. All other functions operated as expected. 

**Expected behaviour**
not this!

**Predbat version**

v8.12.0
Tried reverting v8.11.6 but still had the same problem so am on v8.11.4 which seems to be fine now. 

**Environment details**

- Givenergy gen1 5kw and 9.5kwh battery
- Standard HAOS installer

**Screenshots**
If applicable, add screenshots to help explain your problem. The most useful ones can be your battery chart, the Predbat HTML plan and your current settings in HA.

**Log file**

2025-01-27 08:52:41.226633: Error: Exception raised '>' not supported between instances of 'NoneType' and 'float'
2025-01-27 08:52:41.228968: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 940, in update_time_loop
    self.update_pred(scheduled=False)
  File "/config/predbat.py", line 585, in update_pred
    status, status_extra = self.execute_plan()
  File "/config/execute.py", line 307, in execute_plan
    inverter.adjust_force_export(True, discharge_start_time, discharge_end_time)
  File "/config/inverter.py", line 1828, in adjust_force_export
    if current > self.reserve_percent:
TypeError: '>' not supported between instances of 'NoneType' and 'float'

2025-01-27 08:52:41.254649: Info: record_status Error: Exception raised '>' not supported between instances of 'NoneType' and 'float'
2025-01-27 08:52:41.254803: Error: '>' not supported between instances of 'NoneType' and 'float'

**Predbat debug yaml file**
Sorry have to go to work but can add later if the above log isn't enough!

gcoan, 2025-01-27:
It possibly could be an issue with givtcp retrieving bad data which is causing predbat to fail.

Can you look in your givtcp logfile, do you see any givtcp issues?  Maybe try restarting GivTCP

I am running 8.12.0 and before that 8.11.6 with a Gen 1 hybrid with no errors so if it persists failing for you then definitely will need a logfile attaching

springfall2008, 2025-01-27:
Looks like it relates to the discharge control, I'll add something to catch if the value isn't a number

tractor137, 2025-01-27:
The givtcp log shows info level only logs of successful charge slot settings. Last night (following the update that caused the problems) doesn’t show any attempts to request timed export. To me it looks like the request to export is not being sent to givtcp (although I wouldn’t profess to be anything more than slightly computer literate!)

pacemaker82, 2025-01-27:
Also getting this error, it is stopping me from exporting whenever planned.

Here's my log:

2025-01-27 22:10:15.867696: Inverter 0 Disabled charge window while waiting for schedule (now 01-27 22:10:00 target set_window_minutes 30 charge start time 01-27 23:00:00)
2025-01-27 22:10:15.867957: Adjust idle time, charge 00:00:00-00:00:00 discharge 00:00:00-00:00:00
2025-01-27 22:10:15.868582: Adjust demand (idle) time computed is 22:10:00-23:59:00
2025-01-27 22:10:15.868865: Inverter 0 set new demand (idle) start time to 22:10:00 was 23:01:00
2025-01-27 22:10:25.945026: Inverter 0 write_and_poll_value: Wrote 22:10:00 to idle_start_time, successfully now 22:10:00
2025-01-27 22:10:25.945494: Next export window will be: 2025-01-27 21:05:00+00:00 - 2025-01-27 23:01:00+00:00 at reserve 4
2025-01-27 22:10:25.945653: Exporting now - current SoC 5.33 and target 0.38
2025-01-27 22:10:25.945936: Inverter 0 Adjust force export to True, change times from 00:25:00 - 03:01:00 to 21:05:00 - 23:01:00
2025-01-27 22:10:25.946020: Adjust idle time, charge 00:00:00-00:00:00 discharge 21:05:00-23:01:00
2025-01-27 22:10:25.946521: Adjust demand (idle) time computed is 23:01:00-23:59:00
2025-01-27 22:10:25.946729: Inverter 0 set new demand (idle) start time to 23:01:00 was 22:10:00
2025-01-27 22:10:36.029685: Inverter 0 write_and_poll_value: Wrote 23:01:00 to idle_start_time, successfully now 23:01:00
2025-01-27 22:10:36.030076: Inverter 0 set new export start time to 21:05:00
2025-01-27 22:10:36.030218: Inverter 0 Set new export end time to 23:01:00 was 03:01:00
2025-01-27 22:10:36.030378: Error: Exception raised '>' not supported between instances of 'NoneType' and 'float'
2025-01-27 22:10:36.031321: Error: Traceback (most recent call last):
  File "/config/predbat.py", line 995, in run_time_loop
    self.update_pred(scheduled=True)
  File "/config/predbat.py", line 586, in update_pred
    status, status_extra = self.execute_plan()
  File "/config/execute.py", line 307, in execute_plan
    inverter.adjust_force_export(True, discharge_start_time, discharge_end_time)
  File "/config/inverter.py", line 1828, in adjust_force_export
    if current > self.reserve_percent:
TypeError: '>' not supported between instances of 'NoneType' and 'float'

2025-01-27 22:10:36.085402: Info: record_status Error: Exception raised '>' not supported between instances of 'NoneType' and 'float'
2025-01-27 22:10:36.085738: Error: '>' not supported between instances of 'NoneType' and 'float'

pacemaker82, 2025-01-27:
Update: Just upgraded to latest 8.13.0 and looks to have resolved. Sorry didn't realise - am on a flight to california so the wifi is slow and wasn't able to trigger update until now.

tractor137, 2025-01-28:
V8.13.0 has fixed this for me too

closed 2025-01-28 by tractor137