springfall2008/batpred #1925: Predbat preventing batteries from charging to 100% overnight?

DanielGallo, 2025-01-27:
**Describe the bug**
I noticed a strange issue this morning - my batteries are set to always charge up to 100% overnight from 11:30pm to 5:30am (on Intelligent Octopus), but for some reason this morning at 2:55am, Predbat's status changed from "Charging" to "Hold charging", and then every 10 minutes it kept alternating between "Hold charging" and "Demand" until 5:30am, resulting in the batteries being a lot lower in charge from 5:30am.
I did update to Predbat 8.11.6 yesterday (from an earlier 8.11.x version, I believe) - could it be related?

**Expected behaviour**
Normally there is no "demand" or discharge between 11:30pm and 5:30am, when the import is cheaper. The batteries always charge to 100% until 5:30am, after which they start discharging.

**Predbat version**
8.11.6

**Environment details**
Two GivEnergy Gen-3 5kW Hybrid inverters, on a single phase. One inverter has 2 x 8.2kWh batteries, the other has 2 x 9.5kWh batteries. This setup has been working fine for about 11 months (on various versions of Predbat), and I'm using the balance inverters feature.

**Screenshots**
[image]
[image]

**Log file**
[predbat.1.log](https://github.com/user-attachments/files/18557944/predbat.1.log)

**Predbat debug yaml file**
[predbat_debug_12_20_00.yaml.txt](https://github.com/user-attachments/files/18557943/predbat_debug_12_20_00.yaml.txt)

DanielGallo, 2025-01-28:
@springfall2008 Hi! I downgraded back down to 8.11.1 last night, and oddly the same thing happened overnight - at 2:50am, Predbat went from "Charging" to "Hold charging", with inverter one only reaching 66% and inverter two reaching 87% SOC. Then shortly after 3am it kept alternating between "Demand" and "Hold charging". Do you happen to have any thoughts please?

[image]

gcoan, 2025-01-28:
> When I looked last night, the battery plan showed "Charging" to 100% until 5:30am, and the same is showing for this evening. So it's odd that something isn't allowing the batteries to charge to 100%.

This could be perfectly normal behaviour, depends on your tariff and projected solar and load for tomorrow.

Predbat plans the battery activity at a 5 minute level of granularity, and it executes the plan every 5 minutes, i.e. every time it runs.

Every 10 minutes it re-runs the plan calculation, i.e. every other time it runs.

In the HTML plan though Predbat has to show a summary of this 5-minute plan.  In a 30 minute slot there could be a mixture of Charging and Hold Charging occurring at the 5 minute slot level, but in the plan it can only show one activity.
Predbat only charges the battery as much as it needs to to get you through the next day with minimised expenditure, so a mixture of charge and hold charge might be the right plan for you

What tariff are you on and do you have a copy of the projected plan from yesterday for today?  

Maybe tonight capture a debug log and copy of the plan sometime in the evening

DanielGallo, 2025-01-28:
@gcoan Thanks for the reply. 

This is the first time I've observed this kind of behavior though (two nights in a row the batteries aren't charging to 100%), despite using Predbat for about 11 months. With solar generation at a minimum this time of year, I'm pretty much guaranteed to fully deplete the batteries by the evening, so it's strange that it's limiting the SOC when the plan showed a target charge of 100%.

I'm on Intelligent Octopus, but the car hasn't been plugged in for a few days, so I'm just getting the standard 11:30pm to 5:30am off-peak rate at the moment.

Will definitely try and capture debug logs tonight (should I leave debug mode enabled all night long, or is that too much of an overhead?) and a copy of the HTML plan.

Here's what the plan currently looks like for today and this evening (last night looked the same, i.e. off-peak from 11:30pm to 5:30am, and charge to 100%).

[image]

gcoan, 2025-01-28:
> With solar generation at a minimum this time of year, I'm pretty much guaranteed to fully deplete the batteries by the evening, so it's strange that it's limiting the SOC when the plan showed a [...]

Can't see it from the plan yet, but if Predbat's prediction is that your house load will exceed the battery capacity then yes it should be charging to 100%, especially if you are on IOG where your night rate is so much better than your day rate. 

The charging plan looks to be OK, presumably you don't have low rate charge mode turned on or Predbat would plan the charging to finish later on than it does.  And its slightly unusual to see the charging delayed and it only to be overnight charging, a lot of people on IOG Predbat will charge, then discharge a bit and then charge for the remainder of the period - taking advantage of the import/export rate difference.  I suspect that the metric self sufficiency and battery value is preventing Predbat from doing that.

Have you got the temperature charge curve configured in apps.yaml?

A debug file and logfile from some time tonight will be fine, ideally about 4 hours or less before the start of the charging period.  You can leave debug turned on all the time but it does add volumes to the logfile and creates a debug file every 5 minutes so personally I only turn debug mode on when I need it and I have an automation that turns debug mode off after 10 minutes to stop it needlessly filling the logs up.

DanielGallo, 2025-01-28:
Thanks Geoffrey.

> presumably you don't have low rate charge mode turned on or Predbat would plan the charging to finish later on than it does

"Set Charge Low Power Mode" is disabled (I don't think it's ever been enabled).

> And its slightly unusual to see the charging delayed and it only to be overnight charging, a lot of people on IOG Predbat will charge, then discharge a bit and then charge for the remainder of the [...]

Back in November, I increased the Metric Self Sufficiency from the default of "0" (following a suggestion from another user in the Predbat Facebook group)  - I was noticing that during the day when there was actually a small amount of solar, the excess was being exported to the grid, when I wanted it to be stored in the batteries for later use. I guess this might be a side effect of that setting (preferring to retain energy rather than export it)? I only plan to have this setting enabled during the winter months. I'm not sure whether you're in the Predbat Facebook group, but I discussed the issue here: https://www.facebook.com/groups/1477599886299106/posts/1609532023105891.

> Have you got the temperature charge curve configured in apps.yaml?

Not that I can see, no.

> A debug file and logfile from some time tonight will be fine, ideally about 4 hours or less before the start of the charging period.

Thanks! Will try that tonight.

gcoan, 2025-01-28:
No I'm not on facebook, I figure I have enough trouble keeping up with my own life without seeing pictures of other people's lives 🤣

Although the Predbat group would be useful I'm sure.  Am sticking to github and the givenergy community discussions.

Cheers !

DanielGallo, 2025-01-29:
I took a closer look at one of the Predbat logs last night, and saw that although GivTCP appeared to be returning correct SOC values (I could see the history of the SOC values returned by GivTCP in Home Assistant), Predbat's logs indicated it was seeing a much higher SOC (even though Predbat is using GivTCP's REST API to get the same data). 

Predbat saw the batteries of both inverters reached 100% charge around 2:50am, when they were in fact much lower at that time, causing the weird behavior described above.

I did actually restart one of my inverters several days ago because it stopped "seeing" one of the 9.5 batteries - I noticed only one battery was discharging (one was at 40-something percent and the other was almost full - I use my web dashboard to always monitor the SOC of all batteries: https://github.com/DanielGallo/GivEnergy-Smart-Home-Display.

I've no idea why the inverter stopped seeing the other battery - the GivEnergy portal only showed 1 battery pack, when there are 2! The affected battery wouldn't power down via a long press of the button, so I guess something was frozen. I had to turn off the DC switches on the batteries and was then able to power down the frozen battery, then powered down the other battery and inverter, started it all up again, and it was fine after that.

So perhaps the restart of the inverter/batteries caused things to get into a weird state with Predbat and/or GivTCP. In any case, I restarted the Predbat service (and GivTCP just in case) last night and the batteries fully charged up overnight.

thewookiewon, 2025-02-24:
Hello,

I've noticed similar behaviour my batteries always charged to 100% as night rate is 6.7p but tomorrow is only charging to 47% and then let solar top it up.

My export is 16.5p so the logic seems wrong somehow, I've restarted both GivTCP and Predbat but makes no difference.

gcoan, 2025-02-24:
> I've noticed similar behaviour my batteries always charged to 100% as night rate is 6.7p but tomorrow is only charging to 47% and then let solar top it up.

can you include a debug file please to eliminate that its wrong setting causing this

chalfontchubby, 2026-09-17:
Could you confirm whether this is still an issue? A lot has changed and in order to debug abything now we would need a new debug.yaml please.

If it's fixed (or we don't hear back to the contrary) I'll close this ticket in around a week so we can focus on live issues. Thanks.

closed 2026-09-26 by chalfontchubby