springfall2008/batpred #1926: Allow DayOfWeek filter in rates_import

peter-dolkens, 2025-01-28:
My retailer (and I don't believe I'm the only one) charges different rates on the weekend, vs weekdays. I can't see any way with the current `rates_import` configuration options to reflect this accurately in my setup.

**Describe the solution you'd like**
Add an optional `dayOfWeek` property that can accept days of the week in your desired format.

E.G.

```yaml
  rates_import:
  # Weekdays (Monday to Friday)
  - start: "00:00:00"
    end: "07:00:00"
    rate: 0.2526
  - start: "07:00:00"
    end: "09:00:00"
    rate: 0.3966
  - start: "09:00:00"
    end: "17:00:00"
    rate: 0.3854
  - start: "17:00:00"
    end: "20:00:00"
    rate: 0.3966
  - start: "20:00:00"
    end: "22:00:00"
    rate: 0.3854
  - start: "22:00:00"
    end: "00:00:00"
    rate: 0.2526
  # Weekends (Saturday to Sunday)
  - start: "00:00:00"
    end: "00:00:00"
    rate: 0.2526
```

**Describe alternatives you've considered**
I believe there's ways to achieve this with additional overrides, but this seems fragile and overly complex

gcoan, 2025-01-28:
You could use the predbat manual api to set rates_import_override for the weekend.  Would need an automation to set it each week for the specific weekend dates

This would work, but I agree with your FR, having the ability to specify days of the week in rates_import in apps.yaml would make this easier to configure

peter-dolkens, 2025-01-29:
Yup - that's what I was referring to in my `alternatives you've considered` - for now I'm still in monitoring mode to evaluate if I'm going to use the product further, so just dealing with weekday rates at the moment.

WyndStryke, 2025-01-31:
How are you dealing with home assistant's energy dashboard?  I had to do an automation for that, so it'd be able to pick up the prices anyway.  

(although I do find Predbat's solution for simple rates easier to configure than home assistant's one.  I wish HA could do it more like predbat does it).

gcoan, 2025-01-31:
> How are you dealing with energy rates on home assistant's energy dashboard? I already had to do an automation for that, so extending it to set predbat's rate prices as well should be just a small extra step.

The way I would do it would be to have 'entity contains current rate' and an automation to set the entity value at the appropriate days/times.

Alternatively you could use a utility meter with different tariffs, e.g. weekday/weekend, which means you can see the different rates and import kWh in say the monthly view, but I do find utility meters can 'spike' when you restart HA which messes up the Energy dashboard and requires manual correction to the HA statistics.

Personally I'd do the first with an entity containing the rate

springfall2008, 2025-02-01:
I've just completed adding this and merged the code to 'main' for testing.

The option is 'day_of_week' and it takes a number between 1 and 7 which is Monday-Sunday

e.g.

```
- rate: 5.0
   day_of_week: "1,2,3,4,5"
- rate: 10.0
   day_of_week: "6,7"
```

WyndStryke, 2025-02-01:
Nice one

I wish home assistant would do something like this for it's own energy dashboard.  So much easier to set up in predbat compared to home assistant itself.

peter-dolkens, 2025-02-02:
I think the main thing is home assistant has an entity to track, which wouldn't really work for predbat as it needs to predict the future, which is why I didn't suggest that as an alternative.

The home assistant method with a single template sensor works quite well, and thanks to @springfall2008, now so does predbat.

Thanks 🙏

closed 2025-02-02 by peter-dolkens