springfall2008/batpred #1928: Predbat overwriting FoxESS Modbus sensor

david00yates, 2025-01-28:
**Describe the bug**
When in Control Charge mode, predbat overwrites the battery charge sensor with a state value of 2600 and blanks the state attributes
This is with the FoxEss Modbus integration

**Expected behaviour**
From what I can work out predbat should be reading this sensor value, not writing to it

**Predbat version**

v8.13.0

**Environment details**

- FoxEss H1-5.0-E-G2 inverter, 2x EP5 batteries
- HAOS running predbat as an add-on
- FoxEss Modbus integration

[image]

[image]

**Log file**

[predbat.log](https://github.com/user-attachments/files/18576612/predbat.log)

**Predbat debug yaml file**

[predbat_debug_16_45_00.yaml.txt](https://github.com/user-attachments/files/18576703/predbat_debug_16_45_00.yaml.txt)

david00yates, 2025-01-30:
It would be good to understand what the purpose of charge_rate and discharge_rate are. It's not clear from the documentation but it does look like these should be to control the charge rate, in which case I have configured my instance incorrectly.

I'll change the config and update on the results

gcoan, 2025-01-30:
the charge rate and discharge rate in apps.yaml should point to entities that:

- tell Predbat what the current inverter charge and discharge rate is
- enable Predbat to change the inverter charge and discharge rate 

I can add something more to the documentation if its not clear?

WyndStryke, 2025-01-30:
This is what I have from the integration (Nathan's FoxESS modbus, since there may be others)

- sensor.bms_charge_rate   (maximum possible charge rate from the BMS - this may further be limited by the inverter, purely a sensor and should not be written to)
- sensor.bms_discharge_rate   (maximum possible discharge rate from the BMS - this may further be limited by the inverter, purely a sensor and should not be written to)
- number.force_charge_power (kW, how to control the charge rate when force-charging, this is the one that predbat should be writing to)
- number.force_discharge_power (kW, how to control the discharge rate when force-discharging, this is the one that predbat should be writing to)
- number.max_charge_current (kW, theoretical maximum from the battery stack, will be further limited by the BMS and inverter)
- number.max_discharge_current (kW, theoretical maximum from the battery stack, will be further limited by the BMS and inverter)
- sensor.battery_charge (kW, instantaneous measurement of what the battery is charging right now, purely a sensor, should not be written to)
- sensor.battery_discharge (kW, instantaneous measurement of what the battery is discharging right now, purely a sensor, should not be written to)
- sensor.inverter_battery_power (kW, instantaneous measurement of what the inverter is drawing or sending to the battery, can be positive or negative, purely a sensor, should not be written to)
- sensor.inverter_power (kW, instantaneous measurement of what the inverter is converting, can be positive or negative, purely a sensor, should not be written to)

So for example, I have a 7kW inverter, my bms charge rate is 35A (currently limited by temperature), and my bms discharge rate is 50A, the max charge current is 50A, the max discharge current is also 50A, but in reality my upper limit is 7kW which is 28A because of the inverter.

david00yates, 2025-01-30:
For clarity it is Nathan's Modbus integration I am using.

springfall2008, 2025-02-02:
Do we need support for the charge/discharge rate to be in kW too?

WyndStryke, 2025-02-02:
Easy to convert between kW and W in a helper, it might already be doing that?  (You mentioned that the kW pv_power sensor which I complained about a couple of weeks ago **https://github.com/springfall2008/batpred/issues/1870** is converted to W in history, so I'm guessing the others are likely to do the same).

I'm using predbat in read-only mode, and it seems to be fine as far as that is concerned.   I can pick up predbat's outputs and use that in my own automations.

I have some doubts about the fox.yaml template, in particular it doesn't let predbat put the inverter into Fox's "feed-in first" mode to allow PV to be directly exported rather than being stored in the battery, and I don't think it sets up all the available sensors. I was going to experiment with that but haven't had time.  It works at a MVP level at least.

bfayers, 2025-02-02:
The number entities for max charge/discharge current are write-only. You would need to be looking at sensor.invbatcurrent (negative when charging) or sensor.bat_current (negative when discharging) to get the values for instantaneous charge/discharge

As for @gcoan 's comment about these:

> the charge rate and discharge rate in apps.yaml should point to entities that:
>
>   - tell Predbat what the current inverter charge and discharge rate is
>    - enable Predbat to change the inverter charge and discharge rate

I'm really not sure how you could configure those on a fox setup? Personally at the moment I have these settings:
```yaml
  charge_start_service:
    service: select.select_option
    entity_id: select.work_mode
    option: "Force Charge"
  # charge_stop_service:
  #   service: select.select_option
  #   entity_id: select.work_mode
  #   option: "Self Use"
  charge_stop_service:
    service: script.inverter_self_use_40a_charge_discharge #Script to set work mode to Self Use and number.max_charge_current / number.max_discharge_current to 40A.
  charge_freeze_service:
    service: number.set_value
    entity_id: number.max_charge_current
    value: 0
  discharge_freeze_service:
    service: number.set_value
    entity_id: number.max_discharge_current
    value: 0
  discharge_start_service:
    service: select.select_option
    entity_id: select.work_mode
    option: "Force Discharge"

  charge_rate:
    - input_number.force_charge_power_watts
  discharge_rate:
    - input_number.force_discharge_power_watts
  battery_power:
    - sensor.invbatpower
  pv_power:
    - sensor.pv_power
  load_power:
    - sensor.load_power
  soc_kw:
   - sensor.bms_kwh_remaining
  soc_percent:
     - sensor.battery_soc
```

Those input_numbers are helper entities and then I have an automation to sync the watts value from those into the force charge/discharge kW number entities provided by the integration (see here: https://gist.github.com/bfayers/82f5a45155281d187f2758c66872c162)

WyndStryke, 2025-02-02:
One thing I've noticed (today) about using the Fox force discharge power settings, is that if you are being delicate with them (for example, using the low charge rate functionality), and your household load is high, it'll draw from the grid to cover the difference, instead of increasing the discharge rate dynamically.   So might be necessary to add an automation to watch the house load and override the discharge power rate.

gcoan, 2025-02-02:
> One thing I've noticed (today) about using the Fox force discharge power settings, is that if you are being delicate with them (for example, using the low charge rate functionality), and your household load is high, it'll draw from the grid to cover the difference

that is the same behaviour on the givenergy inverters.  If you lock the discharge rate to a low value and house load exceeds that rate then the extra has to come from grid import

WyndStryke, 2025-02-10:
Have a look at this thread : https://github.com/springfall2008/batpred/issues/1964

And @TonyHoyle 's workarounds for kW / W mismatches:   https://github.com/TonyHoyle/foxess_predbat

I think in the long run it would be good for predbat to be able to work with either W or kW sensors / input_numbers.

Also I think the fox.yaml template could do with some TLC, but that's really on us Fox users to figure out, since Trefor doesn't have a Fox inverter.

WyndStryke, 2025-02-10:
Incidentally I have found out that these two can also control the charge and discharge rate.  Previously I thought they were read-only, and that they were the theoretical maximums, but apparently not.  

```
number.max_charge_current 
number.max_discharge_current 

```

I think that number.max_charge_current limits all charging, at the battery pack level (both PV and force-charge), whereas number.force_charge_power will only affect the charge rate when in force charge mode specifically.  Also I _think_ that they are independent (i.e., the practical limit is the lower of the two).  the force_charge_power and force_discharge_power limits the inverter's AC/DC conversion, whereas the max_change_current/max_discharge_current is at the BMS level.

`There are two configurable parameters, "Force Charge Power" and "Force Discharge Power". These control the maximum charge/discharge power when using Force Charge and Force Discharge respectively. Note that this is the power through the inverter's AC interface, not the battery power nor the power drawn from / supplied to the grid.`

WyndStryke, 2025-07-23:
> 2\. Draw power from the usb connection on the inverter (H3 model).

Probably a good idea to check whether you can use it for power - at least on my KH inverter, the USB connection is only for firmware updates and can't power a device.  Perhaps ask on the Fox customer forum, I'm sure Will or Dave will know.  https://foxesscommunity.com/viewforum.php?f=12

closed 2026-08-28 by chalfontchubby