springfall2008/batpred #1940: PredBat Excess Battery Export Causing Zappi Charger to Charge Car Outside of Plan

Jorthax, 2025-01-30:
Not sure the appropriate place to put this, but since I have entrusted my Solar + Batteries to PredBat for so long I figured I'd start here.

I'm new to IOG with an electric car since Monday, I have a Zappi charger linked with IOG, first couple of charges seemed to work great. However at 21:30 tonight I had excess battery which PredBat correctly decided to export @ 15p/unit. 

This however, started the Zappi (in Eco+ mode as it must be for IOG) to start charging the car as this was seen as excess power.

Should PredBat be setting my Zappi to Stop when it exports? (I'm not sure if this is 'allowed' on IOG?) Or do I need to set something up on the Zappi? 

The car is a 2024 Kona EV - so the IOG is via the Zappi, not the car.

Thanks in advance for everything.

gcoan, 2025-01-30:
You probably need to set an export margin in the Zappi so that it doesn't start charging when it sees excess export

Jorthax, 2025-01-30:
I've just been out in the cold and set a 5000W export margin. I don't think that can ever be triggered. Will test the behaviour at 23:30 when the car should start charging. 

Thanks - sorry to clutter this space!

Not sure my CT clamp is right for my hybrid setup, will do more research tomorrow.

gcoan, 2025-01-30:
no apology needed.  If this cures the problem I will add something to the documentation to advise others

Jorthax, 2025-01-30:
OK - 5000W correctly let the battery discharge from 27% to 4% or pushing out about 5kwh total to the grid without the Zappi kicking back in.

It's now switch to car charging correctly (Zappi in ECO+ as controlled by Octopus directly).

Plan has one export slot planned overnight, where the car will not charge for 30 mins (again as planned by IOG).

It's currently on a status I've not seen before called "Hold for Car" which I cannot see fully explained but I've only just tried to look. Plan looks good for batteries to hit 100% at 5:30am so all good there. Funnily enough the car will be done then also.

Even as typing this it updated on the 15 minute mark and stopped holding for car, back to full draw!

[image]

gcoan, 2025-01-31:
Hold for car is I think Predbat waiting for the car to start charging under IOG

I probably should add that to the documentation

cyberkryten, 2025-01-31:
I had the same issue and changing the export margin fixed mine too as it stops the Zappi charging when it sees export.

If adding to the docs then it makes sense to set the Zappi Export Margin (or other charger) to 8000W so it can never divert excess solar to the car (on a single phase system), or maybe even suggest inverter maximum size + 1000W.  

5000W would allow systems with larger inverters to divert to car and so may cause confusion.

For those of us with export limits (mine is 3.68kW) it complicates things as in theory the car could be used as an iBoost sink when there is excess solar and the batteries are full - the idea would be that by setting 3680W you can do that but there are a couple of issues.

Car chargers have minimum times and limits (spec is minimum 1.4kW), so cannot turn on/off that quickly or with small amounts of excess.

Also export values are really transient and will go over the limit (eg when a 3kW kettle turns off), which if detected could start a charge that then persists for the minimum duration into the car when it is better to export.

Jorthax, 2025-01-31:
I am in agreement that I think - if you want to prevent ever using the battery/excess-solar for car charging, that even 8000W is better than 5000W.

With fixed Oct.Export and IOG, I cannot imagine any time I would want solar to be anything other than exported, when my batteries are full (or likely to be full all day). Currently PredBat is in Freeze Exporting - as I'm at 85% battery at almost midday and predicted to vent about 5-6kwh at 9:30, like last night.

All behaviours were as expected overnight, and today the plan seems to be looking good.

gcoan, 2025-01-31:
I've updated the documentation to add this tip about setting export margin

closed 2025-11-05 by gcoan