springfall2008/batpred #1941: Charge Freeze/Manual_charge_freeze doesn't revert after time window

savethemanual, 2025-01-31:
**Describe the bug**
I notice that when Predbat has a charge freeze in the plan, it can initiate it, setting the battery_max_discharge_current to 0A, and changing the SOC, but it doesn't seem to revert the battery_max_discharge_current back to the previous level afterwards. This means that the battery doesn't discharge and I draw from the grid instead. I haven't been able to capture this in the logs for a standard time window in the plan, but it can be replicated by doing a manual_charge_freeze, where I can see the Sunsynk registers being written (Shown below), and some being reverted afterwards, except for the battery_max_discharge_current register.

**Expected behaviour**
I believe Predbat should revert the battery_max_discharge_current value after the time slot in the plan finishes, thus allowing battery discharge again.

**Predbat version**

v8.13.1

**Environment details**

- Sunsynk 8kW, Single phase LV Hybrid, Software: M 5.3.8.7 / S  1.5.1.5 / C E.4.3.D, Fogstar 15.5kWh
- Standard HAOS installer, 
- Core 2025.1.4
- Supervisor 2024.12.3
- Operating System 14.2
- Frontend 20250109.2

**Log file**
Kellerza add-on logs for register writes to the inverter:
10:46:02 INFO    Queue update load_limit=Zero Export
10:46:02 INFO    Queue update prog6_capacity=100
10:46:03 INFO    Writing sensor prog6_capacity=100 Registers:{273=0x0064} 
10:46:03 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
10:46:04 INFO    Queue update load_limit=Zero Export
10:46:04 INFO    Queue update prog6_capacity=5
10:46:05 INFO    Writing sensor prog6_capacity=5 Registers:{273=0x0005} 
10:46:05 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
10:56:34 INFO    Queue update battery_max_discharge_current=0
10:56:35 INFO    Writing sensor battery_max_discharge_current=0 Registers:{211=0x0000} 
10:56:39 INFO    Queue update prog6_capacity=76
10:56:40 INFO    Writing sensor prog6_capacity=76 Registers:{273=0x004c} 
10:56:49 INFO    Queue update load_limit=Zero Export
10:56:49 INFO    Queue update grid_charge_enabled=ON
10:56:50 INFO    Writing sensor grid_charge_enabled=ON Registers:{232=0x0001} [Register 1-->1]
10:56:50 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
11:00:23 INFO    Queue update prog6_capacity=5
11:00:24 INFO    Writing sensor prog6_capacity=5 Registers:{273=0x0005} 
11:00:28 INFO    Queue update load_limit=Zero Export
11:00:28 INFO    Queue update grid_charge_enabled=OFF
11:00:29 INFO    Writing sensor grid_charge_enabled=OFF Registers:{232=0x0000} [Register 1-->0]
11:00:29 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
11:23:54 INFO    Queue update load_limit=Zero Export
11:23:54 INFO    Queue update prog6_capacity=100
11:23:55 INFO    Writing sensor prog6_capacity=100 Registers:{273=0x0064} 
11:23:58 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
11:24:04 INFO    Queue update battery_max_discharge_current=166
11:24:05 INFO    Writing sensor battery_max_discharge_current=166 Registers:{211=0x00a6} 
11:24:09 INFO    Queue update prog6_capacity=5
11:24:10 INFO    Writing sensor prog6_capacity=5 Registers:{273=0x0005} 
11:25:09 INFO    Queue update battery_max_discharge_current=0
11:25:09 INFO    Writing sensor battery_max_discharge_current=0 Registers:{211=0x0000} 
11:25:14 INFO    Queue update prog6_capacity=76
11:25:14 INFO    Writing sensor prog6_capacity=76 Registers:{273=0x004c} 
11:25:25 INFO    Queue update load_limit=Zero Export
11:25:25 INFO    Queue update load_limit=Zero Export
11:25:25 INFO    Queue update grid_charge_enabled=ON
11:25:26 INFO    Writing sensor grid_charge_enabled=ON Registers:{232=0x0001} [Register 0-->1]
11:25:26 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
11:25:27 INFO    Queue update load_limit=Zero Export
11:25:28 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 
11:27:54 INFO    Queue update prog6_capacity=5
11:27:54 INFO    Queue update prog6_capacity=5
11:27:55 INFO    Writing sensor prog6_capacity=5 Registers:{273=0x0005} 
11:27:57 INFO    Queue update load_limit=Zero Export
11:27:57 INFO    Queue update grid_charge_enabled=OFF
11:27:58 INFO    Writing sensor grid_charge_enabled=OFF Registers:{232=0x0000} [Register 1-->0]
11:27:58 INFO    Writing sensor load_limit=Zero Export Registers:{244=0x0002} 

**Predbat debug yaml file**

[predbat_debug_11_30_00.yaml.txt](https://github.com/user-attachments/files/18617650/predbat_debug_11_30_00.yaml.txt)
[predbat.20250131.1129.log](https://github.com/user-attachments/files/18617651/predbat.20250131.1129.log)
[predbat_debug_10_50_00.yaml.txt](https://github.com/user-attachments/files/18617652/predbat_debug_10_50_00.yaml.txt)
[predbat_debug_10_55_00.yaml.txt](https://github.com/user-attachments/files/18617653/predbat_debug_10_55_00.yaml.txt)
[predbat_debug_11_25_00.yaml.txt](https://github.com/user-attachments/files/18617654/predbat_debug_11_25_00.yaml.txt)

springfall2008, 2025-02-01:
There's two things odd here

1. Freeze charge is disabled as inverter does not support it.... but then the override makes it happen anyhow
2. Somehow its calling freeze charge with start and stop

2025-01-31 11:25:19.051393: Inverter 0 Calling service charge_stop_service domain discharge service_name select/select_option with data {'entity_id': 'select.sunsynk_load_limit', 'option': 'Zero Export'}
2025-01-31 11:25:19.059909: Inverter 0 Calling service charge_start_service domain charge service_name select/select_option with data {'entity_id': 'select.sunsynk_load_limit', 'option': 'Zero Export'}
2025-01-31 11:25:19.067697: Inverter 0 count register writes 2
2025-01-31 11:25:19.103917: Total inverter register writes now 584
2025-01-31 11:25:23.541206: Wrote debug yaml to /addon_configs/6adb4f0d_predbat/debug/predbat_debug_11_20_00.yaml
2025-01-31 11:25:23.541316: Completed run status Freeze charging
2025-01-31 11:25:23.823110: Info: record_status Freeze charging target 76%

springfall2008, 2025-02-01:
I'm confused how come you charge start and charge stop service are the same service?

springfall2008, 2025-02-01:
Potential fix on 'main' for testing

savethemanual, 2025-02-03:
Thanks @springfall2008 . I did wonder about the charge start/stop service, but that's what's in the template for Sunsynk inverters.

Would it just be easier to toggle switch.sunsynk_grid_charge_enabled on/off for this service instead? You don't really need to set it on the timeslots if we're just using slot6 for controlling the SOC.