springfall2008/batpred #1948: Tarrif switching using - https://github.com/eelmafia/octopus-minmax

Johnr24, 2025-02-02:
I would love to see predbat use the ability to tarrif switch as seen here - https://github.com/eelmafia/octopus-minmax

**Describe the solution you'd like**
predbat would run the useage at the end of each day at around 11pm and switch the user to the cheapest tarrif,

gcoan, 2025-02-02:
Personally I am not in favour of such bot solutions, it goes against the spirit of being able to change tariffs and will doubtless lead to Octopus clamping down on such behaviours to the detriment of everyone.

With Agile you have to accept that there will be good days and bad days and part of accepting the bad expensive days is the benefit of cheaper and even negative price days.  Over the longer term you get hopefully cheaper rates than you would on another tariff.  
If the sums don't work then of course change tariff, but not change on a daily basis.

al090187, 2025-02-04:
I sort of agree with gcoan. There's always a few that ruin it for the masses. But, someone has figured it out so a lot will be trying it now. Might as well take advantage while we still can.

But, you need to work out the best plan before the day starts. It's no good finding out at the end of the day that Go was the cheapest, but you've done all your charging at 9am on the best Agile slot there was.

Agile prices released at 4pm. Predict both plans after that. Work out the best plan. Do the switch just after midnight. Then Predbat knows whether to discharge everything and charge at 00:30 on Go slot, or hold the charge and wait for a cheaper Agile slot.

yellowduck, 2025-02-04:
I also agree with the other comments that this feature does not belong in predbat. I have a general unease with several aspects of this feature. I would not have a problem if predbat did the calculations, and users then chose to act upon that decision themselves, but this is not a battery management feature which should be the prime objective of the software.

Personally, I struggle with the ease in which new features are added. It makes it very difficult to decide whether to update, because there is always a bug introduced! It is very difficult to run a parallel system to test before deployment because of the nature of the system, which means I am very often a long way behind any updates.

With that in mind, is there any way we can discuss potential features prior to implementation and decide upon a test plan? With unittests we could then be more confident of changes being valid before deployment. The current system relies upon users in an active system to report bugs. Scary stuff when it is relied upon so heavily. 
I would be able to help with unittests, but would need to be able to run in an isolated system. Maybe a short tutorial on that?

springfall2008, 2025-02-04:
I'm working on a tariff comparison feature but might well stop short of the actual switch itself.

> With that in mind, is there any way we can discuss potential features prior to implementation and decide upon a test plan? With unittests we could then be more confident of changes being valid [...]

I don't think thats totally correct I have a large amount of unit level tests now, but the coverage is incomplete so its always possible to create new bugs.

owinwood, 2025-02-07:
Good to hear a tariff comparison feature is in the works. I currently use the Octopus Compare app to compare against a fixed tariff as it at least gives me some indication of what I'm spending versus the norm. But being able to make an informed switch (not automated) to another smart tariff sounds great.

chibbert, 2025-02-27:
I understand concerns around automated switching and gaming the system.

Even for seasonal switching I think it would be nice to be able to tell predbat that you're going to be changing tariff the next day, so that it makes a plan with that in mind. Have I missed an existing feature that lets you do that. e.g.

-You decide on Monday evening that it's time to make the switch from Go to Agile, and you want it to apply from tomorrow
-Tell predbat that the agile should apply from Tuesday early morning, so it makes a plan based on Agile prices
-You request the tariff change on Tuesday, and accept the T&Cs, so tariff applies from Tuesday early morning

And vica versa going from Agile -> Go

gcoan, 2025-02-27:
Its a useful idea, can see the value in it, but it is a bit of an edge case for a limited audience.

One challenge I can foresee is that there are 3 different ways that Predbat can get the energy rate information:

- Current tariff from the Octopus integration
- Octopus tariff URL's in apps.yaml
- Manual tariff configuration (start/end times & rates) in apps.yaml

It would be an awful lot of extra clutter and code complexity to support 'tomorrow' versions of all of these different options for both import and export rates. And these different energy rate options are needed as not everyone is on Octopus or indeed in the UK.

chalfontchubby, 2026-10-18:
_Posted by Claude on behalf of @chalfontchubby._

Closing. The tariff comparison feature mentioned above now exists (see the Compare tab and the [compare docs](https://springfall2008.github.io/batpred/compare/)). Automatically switching your supplier tariff is out of scope for Predbat, and tools such as octopus-minmax already do that part.

closed 2026-10-18 by chalfontchubby