springfall2008/batpred #1954: Unexpected alterations to SOC during low rate charging

dpe926, 2025-02-02:
Bug - Using predbat with one low rate charge period overnight (12:30 to 05:00). Low power mode and combine power slots is enabled.
The plan at midnight shows a sensible ‘charge to SOC’ and when charging starts the rate is proportional to deliver that SOC at the charge slot end. However in the charge slot predbat increases the ‘charge to SOC’ which has the effect of charging the battery at full rate for the last hour. This would be understandable if the solar forecast had changed but it had not. This also has the effect of increasing the number  of inverter writes and makes it difficult to predict what will be SOC in the morning.

Expected - that the plan just before the start of the charge period has only minor alterations if the solar forecast is constant

2025-02-02 00:00:34,443 - GivTCP - read   [INFO    ] - Midnight, so resetting Day/Night stats...
2025-02-02 00:01:22,020 - GivTCP - read - [INFO    ] - Saving current energy stats at start of night rate tariff (Dynamic)
2025-02-02 00:30:04,494 - GivTCP - write   [INFO    ] - Setting battery charge rate 400 was a success
2025-02-02 01:00:08,922 - GivTCP - write [INFO    ] - Setting Charge Target 25 was a success
2025-02-02 01:45:04,585 - GivTCP - write  [INFO    ] - Setting battery charge rate 3000 was a success
2025-02-02 01:45:15,065 - GivTCP - write  [INFO    ] - Setting Charge Schedule to disable was a success
2025-02-02 01:45:15,545 - GivTCP - write[INFO    ] - Setting battery reserve 26 was a success
2025-02-02 01:45:16,034 - GivTCP - write [INFO    ] - Setting battery discharge limit 0 was a success
2025-02-02 02:10:05,530 - GivTCP - write   [INFO    ] - Setting battery charge rate 400 was a success
2025-02-02 02:10:16,765 - GivTCP – write [INFO    ] - Setting Charge Schedule to enable was a success
2025-02-02 02:10:17,250 - GivTCP - write  [INFO    ] - Setting battery discharge limit 3000 was a success
2025-02-02 02:10:17,723 - GivTCP - write [INFO    ] - Setting battery reserve 4 was a success
2025-02-02 02:15:05,644 - GivTCP - write  [INFO    ] - Setting battery charge rate 3000 was a success
2025-02-02 02:15:17,123 - GivTCP - write  [INFO    ] - Setting Charge Schedule to disable was a success
2025-02-02 02:15:17,603 - GivTCP - write  [INFO    ] - Setting battery reserve 26 was a success
2025-02-02 02:15:18,094 - GivTCP - write  [INFO    ] - Setting battery discharge limit 0 was a success
2025-02-02 03:05:08,039 - GivTCP - write  [INFO    ] - Setting battery charge rate 400 was a success
2025-02-02 03:05:18,523 - GivTCP - write   [INFO    ] - Setting Charge Schedule to enable was a success
2025-02-02 03:05:18,999 - GivTCP - write [INFO    ] - Setting battery discharge limit 3000 was a success
2025-02-02 03:05:19,474 - GivTCP - write [INFO    ] - Setting Charge Target 28 was a success
2025-02-02 03:05:19,945 - GivTCP - write [INFO    ] - Setting battery reserve 4 was a success
2025-02-02 03:35:07,388 - GivTCP - write [INFO    ] - Setting battery charge rate 3000 was a success
2025-02-02 03:35:07,884 - GivTCP - write [INFO    ] - Setting Charge Target 19 was a success
2025-02-02 03:45:08,653 - GivTCP - write  [INFO    ] - Setting Charge Target 22 was a success
2025-02-02 04:05:07,497 - GivTCP - write  [INFO    ] - Setting Charge Target 28 was a success
2025-02-02 04:15:19,923 - GivTCP - write  [INFO    ] - Setting Charge Schedule to disable was a success
2025-02-02 04:15:20,393 - GivTCP - write   [INFO    ] - Setting battery reserve 50 was a success
2025-02-02 04:15:20,880 - GivTCP - write   [INFO    ] - Setting battery discharge limit 0 was a success
2025-02-02 04:35:08,121 - GivTCP - write   [INFO    ] - Setting battery reserve 4 was a success
2025-02-02 04:45:09,104 - GivTCP - write   [INFO    ] - Setting battery reserve 50 was a success
2025-02-02 04:50:14,355 - GivTCP - write  [INFO    ] - Setting Charge Schedule to enable was a success
2025-02-02 04:50:14,833 - GivTCP - write  [INFO    ] - Setting battery discharge limit 3000 was a success
2025-02-02 04:50:15,316 - GivTCP - write  [INFO    ] - Setting Charge Target 49 was a success
2025-02-02 04:50:15,789 - GivTCP - write   [INFO    ] - Setting battery reserve 4 was a success
2025-02-02 04:55:17,753 - GivTCP - write   [INFO    ] - Setting Charge Schedule to disable was a success
2025-02-02 05:00:16,281 - GivTCP - read  [INFO    ] - Saving current energy stats at start of day rate tariff (Dynamic)

 
[image]

[image]

gcoan, 2025-02-02:
could you add a debug file and a copy of the predbat logfile please

dpe926, 2025-02-02:
I think because I updated predbat today, neither of my downloaded files contains the data for the time of interest. Sorry for this

gcoan, 2025-02-03:
its going to be difficult to diagnose a problem without the predbat log or a debug file of your current configuration.  Would be worth seeing if the same happens tonight

dpe926, 2025-02-03:
These are the files from overnight. The same thing happened but was less pronounced

[image]
[givtcp.txt](https://github.com/user-attachments/files/18641534/givtcp.txt)
[predbat.log](https://github.com/user-attachments/files/18641536/predbat.log)
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/18641535/predbat_debug.yaml.txt)
[image]

dpe926, 2025-02-04:
By way of further information, I noted that on the charging graph last night (above) the charge started at midnight rather than the expected 00:30 and at the full rate. I looked tonight ant the same occurred. 
I downgraded predbat to 8.12.0 and the charging stopped. It restarted as expected at 00:30 with a reduced charging rate

gcoan, 2025-02-08:
> I downgraded predbat to 8.12.0 and the charging stopped. It restarted as expected at 00:30 with a reduced charging rate

there was a bug in 8.13 that treated the midnight slot as being zero pence

this is fixed in 8.14