springfall2008/batpred #1957: Battery temperature prediction

johnwb87, 2025-02-03:
**Describe the bug**
Predbat appears to not be handling GivTCP transient drops to zero well in its temperature prediction.

**Expected behaviour**
Predbat to handle these drops gracefully. We know that GivTCP is prone to this occasionally on some sensors.

**Predbat version**

8.13.3

**Environment details**

- GivTCP v3
- HAOS Add-On

**Screenshots**

[image]

**Log file**

[predbat (1).log](https://github.com/user-attachments/files/18644242/predbat.1.log)

**Predbat debug yaml file**

[predbat_debug_15_25_00.yaml.txt](https://github.com/user-attachments/files/18644398/predbat_debug_15_25_00.yaml.txt)

johnwb87, 2025-02-03:
It eventually reset itself. But only when the value of the battery temperature sensor changed.

chalfontchubby, 2026-08-20:
Thanks for the report. This issue has been open for over a year with no recent activity - we're volunteers working through a large backlog and need to filter what's still relevant. If this is still a problem for you, please let us know and we'll keep it open. Otherwise, we'll close it in a week (around 2026-08-28).

johnwb87, 2026-08-21:
Battery temperature not a personal concern with current set up.

closed 2026-08-21 by johnwb87