springfall2008/batpred #1958: Predbat ignoring car charging plan

andreweacott, 2025-02-03:
**Describe the bug**
Predbat correctly chooses the cheapest overnight slots to charge an EV and it initiates the charge in the first of the selected slots.  However, it then doesn't stop charging when the planned slot finishes.  Instead, it recalculates the charging plan and continues the charge in the next slot.  This continues until the car has reached target battery level.  In effect, it's ignoring the originally calculated charging plan.

**Expected behaviour**
Charging plan is followed, using only the cheapest slots.

**Predbat version**
8.13.1
(Note: attached predbat_debug file was taken after an upgrade to 8.13.3)

**Environment details**
- GivEnergy Inverter
- HAOS installation

**Screenshots**
See log file

**Log file**
```
# Before charge started, plan says charge should finish at slot 1380 and start again at slot 1530
# Charge was originally supposed to start at 22:30 (1350) but did not due to an unrelated error
2025-02-01 22:45:01.794353: Car 0 charging plan is: [{'start': 1365, 'end': 1380, 'kwh': 1.85, 'average': 22.37, 'cost': 41.38}, {'start': 1530, 'end': 1560, 'kwh': 3.7, 'average': 23.17, 'cost': 85.73}, {'start': 1590, 'end': 1620, 'kwh': 3.7, 'average': 22.87, 'cost': 84.62}, {'start': 1620, 'end': 1650, 'kwh': 3.7, 'average': 23.17, 'cost': 85.73}, {'start': 1650, 'end': 1680, 'kwh': 3.7, 'average': 22.64, 'cost': 83.77}, {'start': 1680, 'end': 1700.0, 'kwh': 1.924, 'average': 23.29, 'cost': 44.81}, {'start': 1710, 'end': 1740, 'kwh': 3.7, 'average': 22.71, 'cost': 84.03}, {'start': 1740, 'end': 1800, 'kwh': 7.4, 'average': 23.1, 'cost': 170.94}]
2025-02-01 22:50:01.675198: Car 0 charging plan is: [{'start': 1370, 'end': 1380, 'kwh': 1.233, 'average': 22.37, 'cost': 27.59}, {'start': 1530, 'end': 1560, 'kwh': 3.7, 'average': 23.17, 'cost': 85.73}, {'start': 1590, 'end': 1620, 'kwh': 3.7, 'average': 22.87, 'cost': 84.62}, {'start': 1620, 'end': 1650, 'kwh': 3.7, 'average': 23.17, 'cost': 85.73}, {'start': 1650, 'end': 1680, 'kwh': 3.7, 'average': 22.64, 'cost': 83.77}, {'start': 1680, 'end': 1705.0, 'kwh': 2.54, 'average': 23.29, 'cost': 59.16}, {'start': 1710, 'end': 1740, 'kwh': 3.7, 'average': 22.71, 'cost': 84.03}, {'start': 1740, 'end': 1800, 'kwh': 7.4, 'average': 23.1, 'cost': 170.94}]

# Car starts charging (delayed due to an unrelated error)
2025-02-01 22:55:01.431094: Car is charging now slot {'start': 1350, 'end': 1380, 'average': 22.365}
2025-02-01 22:55:01.431215: Car 0 charging plan is: [{'start': 1375, 'end': 1380, 'kwh': 0.617, 'average': 22.365, 'cost': 13.79}, {'start': 1530, 'end': 1560, 'kwh': 3.7, 'average': 23.17, 'cost': 85.73}, {'start': 1590, 'end': 1620, 'kwh': 3.7, 'average': 22.87, 'cost': 84.62}, {'start': 1620, 'end': 1650, 'kwh': 3.7, 'average': 23.17, 'cost': 85.73}, {'start': 1650, 'end': 1680, 'kwh': 3.7, 'average': 22.64, 'cost': 83.77}, {'start': 1680, 'end': 1700.0, 'kwh': 2.31, 'average': 23.29, 'cost': 53.79}, {'start': 1710, 'end': 1740, 'kwh': 3.7, 'average': 22.71, 'cost': 84.03}, {'start': 1740, 'end': 1800, 'kwh': 7.4, 'average': 23.1, 'cost': 170.94}]

# Reaches original planned end of slot (1380) and continues charging with new planned end at 1410  
2025-02-01 23:00:00.968251: Car is charging now slot {'start': 1380, 'end': 1410, 'average': 24.675}
2025-02-01 23:00:00.968314: Car 0 charging plan is: [{'start': 1380, 'end': 1410, 'kwh': 3.7, 'average': 24.675, 'cost': 91.3}, {'start': 1530, 'end': 1560, 'kwh': 3.7, 'average': 23.17, 'cost': 85.73}, {'start': 1590, 'end': 1620, 'kwh': 3.7, 'average': 22.87, 'cost': 84.62}, {'start': 1620, 'end': 1645.0, 'kwh': 2.926, 'average': 23.17, 'cost': 67.8}, {'start': 1650, 'end': 1680, 'kwh': 3.7, 'average': 22.64, 'cost': 83.77}, {'start': 1710, 'end': 1740, 'kwh': 3.7, 'average': 22.71, 'cost': 84.03}, {'start': 1740, 'end': 1800, 'kwh': 7.4, 'average': 23.1, 'cost': 170.94}]

2025-02-01 23:05:02.096387: Car is charging now slot {'start': 1380, 'end': 1410, 'average': 24.675}
2025-02-01 23:05:02.096503: Car 0 charging plan is: [{'start': 1385, 'end': 1410, 'kwh': 3.083, 'average': 24.675, 'cost': 76.08}, {'start': 1530, 'end': 1560, 'kwh': 3.7, 'average': 23.17, 'cost': 85.73}, {'start': 1590, 'end': 1620, 'kwh': 3.7, 'average': 22.87, 'cost': 84.62}, {'start': 1620, 'end': 1645.0, 'kwh': 2.695, 'average': 23.17, 'cost': 62.43}, {'start': 1650, 'end': 1680, 'kwh': 3.7, 'average': 22.64, 'cost': 83.77}, {'start': 1710, 'end': 1740, 'kwh': 3.7, 'average': 22.71, 'cost': 84.03}, {'start': 1740, 'end': 1800, 'kwh': 7.4, 'average': 23.1, 'cost': 170.94}]

2025-02-01 23:10:01.632074: Car is charging now slot {'start': 1380, 'end': 1410, 'average': 24.675}
2025-02-01 23:10:01.632190: Car 0 charging plan is: [{'start': 1390, 'end': 1410, 'kwh': 2.467, 'average': 24.675, 'cost': 60.87}, {'start': 1530, 'end': 1560, 'kwh': 3.7, 'average': 23.17, 'cost': 85.73}, {'start': 1590, 'end': 1620, 'kwh': 3.7, 'average': 22.87, 'cost': 84.62}, {'start': 1620, 'end': 1650.0, 'kwh': 3.312, 'average': 23.17, 'cost': 76.74}, {'start': 1650, 'end': 1680, 'kwh': 3.7, 'average': 22.64, 'cost': 83.77}, {'start': 1710, 'end': 1740, 'kwh': 3.7, 'average': 22.71, 'cost': 84.03}, {'start': 1740, 'end': 1800, 'kwh': 7.4, 'average': 23.1, 'cost': 170.94}]

2025-02-01 23:15:01.256017: Car is charging now slot {'start': 1380, 'end': 1410, 'average': 24.675}
2025-02-01 23:15:01.256135: Car 0 charging plan is: [{'start': 1395, 'end': 1410, 'kwh': 1.85, 'average': 24.675, 'cost': 45.65}, {'start': 1530, 'end': 1560, 'kwh': 3.7, 'average': 23.17, 'cost': 85.73}, {'start': 1590, 'end': 1620, 'kwh': 3.7, 'average': 22.87, 'cost': 84.62}, {'start': 1620, 'end': 1640.0, 'kwh': 2.233, 'average': 23.17, 'cost': 51.73}, {'start': 1650, 'end': 1680, 'kwh': 3.7, 'average': 22.64, 'cost': 83.77}, {'start': 1710, 'end': 1740, 'kwh': 3.7, 'average': 22.71, 'cost': 84.03}, {'start': 1740, 'end': 1800, 'kwh': 7.4, 'average': 23.1, 'cost': 170.94}]

2025-02-01 23:20:01.599613: Car is charging now slot {'start': 1380, 'end': 1410, 'average': 24.675}
2025-02-01 23:20:01.599733: Car 0 charging plan is: [{'start': 1400, 'end': 1410, 'kwh': 1.233, 'average': 24.675, 'cost': 30.43}, {'start': 1530, 'end': 1560, 'kwh': 3.7, 'average': 23.17, 'cost': 85.73}, {'start': 1590, 'end': 1620, 'kwh': 3.7, 'average': 22.87, 'cost': 84.62}, {'start': 1620, 'end': 1645.0, 'kwh': 2.849, 'average': 23.17, 'cost': 66.01}, {'start': 1650, 'end': 1680, 'kwh': 3.7, 'average': 22.64, 'cost': 83.77}, {'start': 1710, 'end': 1740, 'kwh': 3.7, 'average': 22.71, 'cost': 84.03}, {'start': 1740, 'end': 1800, 'kwh': 7.4, 'average': 23.1, 'cost': 170.94}]

2025-02-01 23:25:01.235718: Car is charging now slot {'start': 1380, 'end': 1410, 'average': 24.675}
2025-02-01 23:25:01.235850: Car 0 charging plan is: [{'start': 1405, 'end': 1410, 'kwh': 0.617, 'average': 24.675, 'cost': 15.22}, {'start': 1530, 'end': 1560, 'kwh': 3.7, 'average': 23.17, 'cost': 85.73}, {'start': 1590, 'end': 1620, 'kwh': 3.7, 'average': 22.87, 'cost': 84.62}, {'start': 1620, 'end': 1645.0, 'kwh': 2.618, 'average': 23.17, 'cost': 60.67}, {'start': 1650, 'end': 1680, 'kwh': 3.7, 'average': 22.64, 'cost': 83.77}, {'start': 1710, 'end': 1740, 'kwh': 3.7, 'average': 22.71, 'cost': 84.03}, {'start': 1740, 'end': 1800, 'kwh': 7.4, 'average': 23.1, 'cost': 170.94}]

# Reaches revised planned end of slot (1410) and continues charging with new planned end at 1440...
2025-02-01 23:30:01.776453: Car is charging now slot {'start': 1410, 'end': 1440, 'average': 24.3495}
2025-02-01 23:30:01.776563: Car 0 charging plan is: [{'start': 1410, 'end': 1440, 'kwh': 3.7, 'average': 24.3495, 'cost': 90.09}, {'start': 1530, 'end': 1560.0, 'kwh': 3.235, 'average': 23.17, 'cost': 74.95}, {'start': 1590, 'end': 1620, 'kwh': 3.7, 'average': 22.87, 'cost': 84.62}, {'start': 1650, 'end': 1680, 'kwh': 3.7, 'average': 22.64, 'cost': 83.77}, {'start': 1710, 'end': 1740, 'kwh': 3.7, 'average': 22.71, 'cost': 84.03}, {'start': 1740, 'end': 1800, 'kwh': 7.4, 'average': 23.1, 'cost': 170.94}]
```

**Predbat debug yaml file**
[predbat_debug_19_35_00.yaml.txt](https://github.com/user-attachments/files/18647355/predbat_debug_19_35_00.yaml.txt)
(Note: Have updated to Predbat 8.13.3 since the issue was last seen.  Issue occurred on v8.13.1)

springfall2008, 2025-02-03:
Do you have 'car_charging_now' set as this would cause such behaviour?

andreweacott, 2025-02-04:
> Do you have 'car_charging_now' set as this would cause such behaviour?

Aah, yes.  I've got `car_charging_now` set to a template sensor based on whether the car is currently receiving charge.  Is that incorrect ?

I assumed that option was used to tell Predbat that it's signal to start charging was successful.  
Should I not be using that config option if I want Predbat to control the charge session ?

gcoan, 2025-02-04:
If you set car charging now to indicate that the car is charging then Predbat will assume that the current slot is a low rate slot and will charge the car regardless of the plan https://springfall2008.github.io/batpred/apps-yaml/#planned-car-charging

So once the car starts charging the plan will be overwritten.

You shouldn't normally need to tell Predbat that the start charging signal was successful, Predbat will assume that it is working correctly

andreweacott, 2025-02-04:
> If you set car charging now to indicate that the car is charging then Predbat will assume that the current slot is a low rate slot and will charge the car regardless of the plan [springfall2008.github.io/batpred/apps-yaml#planned-car-charging](https://springfall2008.github.io/batpred/apps-yaml/#planned-car-charging)
> 
> So once the car starts charging the plan will be overwritten.
> 
> You shouldn't normally need to tell Predbat that the start charging signal was successful, Predbat will assume that it is working correctly

Cool, understood.  Thanks for the explanation :)

closed 2025-02-04 by andreweacott