springfall2008/batpred #1960: Zero cost between midnight and first price change

al090187, 2025-02-04:
**Describe the bug**
Since an update on 2nd Feb, Predbat thinks electricity is free between midnight and the first change in prices. I have the prices hard coded in apps.yaml, and one is from 5:30am to 1am. It overlaps the day boundary. It appeared correct in the plan though.

**Expected behaviour**
I expect it to use the price from the last period on the previous day, not auto fill with zero cost.

**Predbat version**

8.13.3

**Environment details**

**Screenshots**
Plan 

[image]

Price chart

[image]

**Log file**

[predbat (1).log](https://github.com/user-attachments/files/18652936/predbat.1.log)

**Predbat debug yaml file**

[predbat_debug2.yaml.txt](https://github.com/user-attachments/files/18652929/predbat_debug2.yaml.txt)

LeeFarm11, 2025-02-04:
I appear to have same issue. 'Plan' shows to charge battery from 02:00 when import rate starts cheap period, and this seems to remain the plan until just before midnight. However, the recalculations from midnight through to 02:00 are deciding that it is best to charge the battery immediately.

This has happened for 2 nights.

Here is snippet from predbat.1.log for last night showing 11:55 and 00:00 calculations.

predict best end_record at 23:55:19 shows SOC to drop until going up from 02:00
predict best end_record at 00:02:39 shows SOC rising immediately and fully charged by 02:00

```
2025-02-03 23:55:03.055326: --------------- PredBat - update at 2025-02-03 23:55:00+00:00 with clock skew 0 minutes, minutes now 1435
2025-02-03 23:55:03.056460: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
2025-02-03 23:55:03.088810: Using cached GITHub data for https://api.github.com/repos/springfall2008/batpred/releases age 80.0 minutes
2025-02-03 23:55:03.089134: Predbat /config/predbat.py version v8.13.3 currently running, latest version is v8.13.3 latest beta v8.13.3
2025-02-03 23:55:03.089264: Autoupdate: Currently up to date
2025-02-03 23:55:03.141710: Configuration: forecast_hours 48 num_cars 0 debug enable is False calculate_plan_every 10
2025-02-03 23:55:03.142640: Predbat mode is set to Control charge & discharge
2025-02-03 23:55:04.056206: Found 5761 load_today datapoints going back 4 days
2025-02-03 23:55:05.076992: Car charging hold False threshold 6.0
2025-02-03 23:55:05.077094: Current data so far today: load 52.8 kWh import 52.6 kWh export 6.1 kWh pv 7.7 kWh
2025-02-03 23:55:05.079998: Adding rate rates_import: {'start': '02:00:00', 'end': '07:30:00', 'rate': 13.44} => 02-03 02:00:00 to 02-03 07:30:00 @ 13.44 date None day_of_week [] increment False
2025-02-03 23:55:05.083685: Adding rate rates_import: {'start': '07:30:00', 'end': '02:00:00', 'rate': 30.16} => 02-03 07:30:00 to 02-04 02:00:00 @ 30.16 date None day_of_week [] increment False
2025-02-03 23:55:05.209914: Adding rate rates_export: {'rate': 15.0} => 02-03 00:00:00 to 02-04 00:00:00 @ 15.0 date None day_of_week [] increment False
2025-02-03 23:55:05.219518: Standing charge is set to 54.56999999999999 p
2025-02-03 23:55:05.279703: Adding rate rates_import_override: {'date': '2024-11-24', 'start': '07:00:00', 'end': '09:00:00', 'rate': 0} => 11-24 07:00:00 to 11-24 09:00:00 @ 0.0 date 2024-11-24 00:00:00 day_of_week [] increment False
2025-02-03 23:55:07.733442: Rate min forward looking: now 13.44 at end of forecast 13.44
2025-02-03 23:55:07.733947: Import rates min 13.44 max 30.16 average 26.33
2025-02-03 23:55:07.805267: Export rates min 15.0 max 15.0 average 15.0
2025-02-03 23:55:07.805371: Rate thresholds (for charge/export) are import 29.66p (0.0) export 14.9p (0.0)
2025-02-03 23:55:07.828560: High export rate found rates in range 15.0 to 15.0
2025-02-03 23:55:07.840252: Low Import rate found rates in range 13.44 to 13.44
2025-02-03 23:55:07.840387: Battery level now 4.0 -1hr 2.97 midnight 7.02 battery value change hour 14.87 day -43.59 rate_forward 14.43
2025-02-03 23:55:07.841347: Hour energy 0.0 import 0.0 export 0.0 car 0 load 0.34 cost 0.0 import 0.0 export 0.0 car 0.0 carbon 0.0 kG
2025-02-03 23:55:16.218054: Today's energy import 52.6 kWh export 6.1 kWh total 46.5 kWh cost 362.52 p import 399.45 p export -91.5 p carbon 0.0 kg
2025-02-03 23:55:16.218213: Using Solcast integration from inside HA for solar forecast
2025-02-03 23:55:16.218473: PV Data for pv_forecast_today total 12.64 kWh
2025-02-03 23:55:16.218605: PV Data for pv_forecast_tomorrow total 13.97 kWh
2025-02-03 23:55:16.218726: PV Data for pv_forecast_d3 total 41.47 kWh
2025-02-03 23:55:16.219079: PV Data for pv_forecast_d4 total 55.08 kWh
2025-02-03 23:55:16.245455: PV Forecast for today is 12.64 (5.82 10% 23.91 90%) kWh and left today is 0 (0 10% 0 90%) kWh
2025-02-03 23:55:16.423420: PV Forecast for day tomorrow is 13.97 (3.67 10% 39.14 90%) kWh
2025-02-03 23:55:16.495328: PV Forecast for day d2 is 41.47 (16.06 10% 49.11 90%) kWh
2025-02-03 23:55:16.582831: PV Forecast for day d3 is 55.08 (21.25 10% 55.08 90%) kWh
2025-02-03 23:55:16.726547: Historical data totals for days [1, 2, 3] are [52.85, 53.91, 59.51] - min 52.85
2025-02-03 23:55:16.726625: Model filter enabled - Discarding day 1 as it is the lowest of the 3 datapoints
2025-02-03 23:55:16.728100: Historical days now [2, 3] weight [0.75, 0.5]
2025-02-03 23:55:16.838179: Today's load divergence 91.6 % in-day adjustment 92.02 % damping 0.95x
2025-02-03 23:55:16.838269: Today's predicted so far 28.65 kWh with 0.0 kWh car/iBoost excluded and 26.46 kWh import ignored and 0 forecast extra.
2025-02-03 23:55:16.838322: Today's actual load so far 26.42 kWh with 0.0 kWh Car/iBoost excluded and 26.38 kWh import ignored.
2025-02-03 23:55:16.967701: Inverter 0: Type GE GivEnergy
2025-02-03 23:55:16.968034: Inverter 0 does not have timed pause support enabled
2025-02-03 23:55:16.968458: Invertor time 2025-02-03 23:54:33+00:00, Predbat computer time 2025-02-03 23:55:16.968369+00:00, difference -0.73 minutes
2025-02-03 23:55:16.968574: Reserve min: 4.0% Battery_min:4.0%
2025-02-03 23:55:16.968680: Inverter 0 with soc_max 20.84 kWh nominal_capacity 20.84 kWh battery rate raw 12000.0 w charge rate 12.0 kW discharge rate 12.0 kW battery_rate_min 200.0 w ac limit 11.5 kW export limit 11.5 kW reserve 4.0 % current_reserve 4.0 % temperature 20 c
[... cut ...]
2025-02-04 00:03:28.106389: Inverter 0 write_and_poll_value: Wrote 100 to charge_limit, successfully now 100
2025-02-04 00:03:28.106717: Inverter 0 Current reserve is 4.0 already at target
2025-02-04 00:03:28.106762: Inverter 0 count register writes 3
2025-02-04 00:03:28.838397: Total inverter register writes now 4217
2025-02-04 00:03:28.980293: Completed run status Charging
2025-02-04 00:03:29.058181: Info: record_status Charging target 19%-100%
2025-02-04 00:03:29.088276: Saved current settings to /addon_configs/6adb4f0d_predbat/predbat_config.json
```

al090187, 2025-02-04:
He's just said in the Facebook chat that he's found the bug, and releasing an update shortly 👍

springfall2008, 2025-02-04:
Fix on main for testing

closed 2026-01-03 by gcoan