springfall2008/batpred #1963: Understanding import ignored?

johnwb87, 2025-02-04:
Hello,

I'm noticing that some days predbat chooses to ignore the overnight load of the immersion heater and other days does not. This leads to less accurate predictions and larger than necessary in-day adjustments. The immersion is a 3kW load so should not be triggering the threshold based car hold. I also use an energy sensor for the car charging hold so that should be accurate.

I can see in the logs that in addition to car charging being removed from the load forecast there is also an import ignored value. So I'm interested in what cause load to be ignored and if that is the likely cause of what I'm seeing.

Thanks.

gcoan, 2025-02-04:
please can you share your logfile and apps.yaml so we can understand your configuration

ideally if you can show the logfile ignoring the immersion load

johnwb87, 2025-02-04:
See attached. An example from the below log file would be this.

2025-02-04 11:15:13.439953: Today's predicted so far 3.18 kWh with 9.21 kWh car/iBoost excluded and 3.64 kWh import ignored and 0 forecast extra.
2025-02-04 11:15:13.440058: Today's actual load so far 5.74 kWh with 27.74 kWh Car/iBoost excluded and 5.41 kWh import ignored.

[apps.yaml.txt](https://github.com/user-attachments/files/18659991/apps.yaml.txt)
[predbat (2).log](https://github.com/user-attachments/files/18659992/predbat.2.log)

gcoan, 2025-02-04:
> 2025-02-04 11:15:13.439953: Today's predicted so far 3.18 kWh with 9.21 kWh car/iBoost excluded and 3.64 kWh import ignored and 0 forecast extra.
> 2025-02-04 11:15:13.440058: Today's actual load so far 5.74 kWh with 27.74 kWh Car/iBoost excluded and 5.41 kWh import ignored.

I get very similar:

2025-02-04 14:55:02.653097: Today's load divergence 92.97 % in-day adjustment 93.32 % damping 0.95x
2025-02-04 14:55:02.653151: Today's predicted so far 14.4 kWh with 0.0 kWh car/iBoost excluded and 11.64 kWh import ignored and 0 forecast extra.
2025-02-04 14:55:02.653165: Today's actual load so far 11.28 kWh with 8.52 kWh Car/iBoost excluded and 7.56 kWh import ignored.

And I don't have a car or an iBoost configured.

I think it is excluding battery charge or something like that, but having said that, the excluded and imported figures don't match my import or charge today figures at 3pm, so I can't explain them, will need @springfall2008 to advise.

Maybe they need to be labelled more clearly in the log as to what is being excluded and what is ignored.

springfall2008, 2025-02-04:
There's a few things here:

- If you use car_charging_hold without a car sensor then all load above the threshold will be removed, this means if you have car + immersion then both will be removed. If you use a car energy sensor then you will get more accurate removal.

- For the in-day adjustments all load during battery charging periods is ignored, this doesn't impact the load prediction directly only the divergence.

You shouldn't be seeing Car/Iboost excluded if you don't have a car or iboost.

gcoan, 2025-02-04:
I don’t have a car or iboost configured, so something weird happening with mine

Here’s the full logfile and the debug file:

[predbat.log](https://github.com/user-attachments/files/18663232/predbat.log)

[2000.txt](https://github.com/user-attachments/files/18663235/2000.txt)

gcoan, 2025-02-04:
> If you use car_charging_hold without a car sensor then all load above the threshold will be removed, this means if you have car + immersion then both will be removed. If you use a car energy sensor then you will get more accurate removal

In the apps.yaml documentation it says

 **input_number.predbat_car_charging_threshold** (default 6 = 6kW)- Sets the kW power threshold above which home consumption is assumed to be car charging
and **input_number.predbat_car_charging_rate** will be subtracted from the historical load data.

Would appear that only the defined car charging rate is subtracted not all load?

springfall2008, 2025-02-05:
> In the apps.yaml documentation it says
>
> **input_number.predbat_car_charging_threshold** (default 6 = 6kW)- Sets the kW power threshold above which home consumption is assumed to be car charging and [...]

Correct, only up to the defined rate, e.g. 6kW in this case.

johnwb87, 2025-02-05:
> * For the in-day adjustments all load during battery charging periods is ignored, this doesn't impact the load prediction directly only the divergence.

I think it is this that I'm seeing then. As mentioned I use an energy sensor for car_charging_hold. Given that GivTCP reports load excluding battery charging what is the constraint that requires this approach? Is it because other inverters don't report load in the same way but import is always import?

Thanks.

closed 2026-08-21 by johnwb87