springfall2008/batpred #1964: Predbat not predicting any battery usage

TonyHoyle, 2025-02-04:
**Describe the bug**
Predbat consistently suggests I'll use at most 0.005w of battery which means it never predicts any battery usage.  Other values seem reasonable and historical battery usage is plotted correctly.

Used the fox.yaml from https://raw.githubusercontent.com/springfall2008/batpred/main/templates/fox.yaml with some minor adjustments for Octopus and setting the soc_max to 10.36 to match my battery.

**Expected behaviour**
Some prediction of usage, even it it's not accurate yet due to shortage of data.

**Predbat version**

8.13.3

**Environment details**

- Fox ESS H1 G2, 2xEP5 batteries
- HAOS installer

**Screenshots**

[image]

[image]

**Log file**
Can you capture a log file from the time of the issue, debug mode is not normally required.

**Predbat debug yaml file**
[predbat_debug.yaml.txt](https://github.com/user-attachments/files/18658359/predbat_debug.yaml.txt)

Enable debug mode and let Predbat run for one cycle. Look in the 'debug' directory under the Predbat add-on and take the .yaml file for the time period in question.
Rename the predbat_debug_XX_XX_XX.yaml to predbat_debug_XX_XX_XX.yaml.txt and upload it to Github. This will allow your plan to be reproduced and also stores all your settings for review.

gcoan, 2025-02-04:
have you got some of the fox sensors set to kW rather than Watts or vice versa?

If you look in the apps.yaml documentation it explains what each sensor needs to be set to

TonyHoyle, 2025-02-04:
I don't see how that could be changed.. the sensors are what they are.. it's up to predbat to be compatible with them.  This is the example config from the documentation, which should work.

WyndStryke, 2025-02-05:
When I look at the power chart, the battery line looks normal to me.  So I'm not sure what the difference is between your config and mine.

[image]

> I don't see how that could be changed..

It's easy enough to do by adding a template sensor to multiply or divide by 1000.  However I understand that some conversions already happen internally (but aren't necessarily visible on the config/charts), for example the sensor.pv_power which confused me for a while.  So a figure might be off by 1000x in the chart and the config, but be might correct in the plan (which is the important bit).  https://github.com/springfall2008/batpred/issues/1870

I do think the fox.yaml could do with a bit of TLC, for example to enable Fox's 'feed-in first' mode (which I _think_ corresponds to 'freeze charge').  But it is really up to us Fox users to propose & test the changes to fox.yaml, because Trevor does not have a Fox system.  There's also some discussion here: https://github.com/springfall2008/batpred/issues/1928

For comparison purposes, here is my apps.yaml: 

Note that I am running in 'readonly' mode, so don't know if the control side of things is working.  I have also never succeeded in getting the carbon intensity working, it reads in the forecasts but that's it.

```
pred_bat:
  module: predbat
  class: PredBat

# MM version for KH7+ECS4300-H4  20241222
# soc_max nominal battery capacity for ECS4300-H4 =16.59kWh   (original was 8.29, maybe for ECS4300-H2)
# battery_rate_max maximum charge/discharge rate, 50A*230V=11500W (note that recommended is 35A, this is handled later) (original was 4200)
# inverter_limit 7000 for KH7 (was 5000, maybe for H1 5000)
# export_limit 7000 (was 3600 commented out, optional in this case since same as inverter_limit)
# inverter_limit_charge/_discharge 8050 (for ECS4300-H4 recommended charge rate 35A * 230v = 8050)
# battery_charge_power_curve for ECS4800, presumably similar to ECS4300, based on 50A max, original was commented out
# rates_import / rates_export as E-on Next Drive + E-on Export
# metric_standing_charge 0.5784 (57.84p)
#
# ### nordpool wholesale prices
#
# carbon_intensity_uk - integration installed
# then input_number.predbat_carbon_metric and switch.predbat_carbon_enable to enable it
#
# Turn on low power mode - switch.predbat_set_charge_low_power
#
# predbat_inverter_loss - 0.02

  # Sets the prefix for all created entities in HA - only change if you want to run more than once instance
  prefix: predbat

  # Timezone to work in
  timezone: Europe/London

  # Currency, symbol for main currency second symbol for 1/100s e.g. $ c or £ p or e c
  currency_symbols:
    - '£'
    - 'p'

  # Days previous is the number of days back to find historical load data
  # Recommended is 7 to capture day of the week but 1 can also be used
  # if you have more history you could use 7 and 14 (in a list) but the standard data in HA only lasts 10 days
  # https://springfall2008.github.io/batpred/apps-yaml/#days_previous
  # Average load data over previous 3 weeks for any given week day.  Homeassistant/configuration.yaml sets history to 22 days.
  days_previous:
  - 7
  - 14
  - 21

  # Number of threads to use in plan calculation
  # Can be auto for automatic, 0 for off or values 1-N for a fixed number
  threads: auto

  # XXX: This is a configuration template, delete this line once you edit your configuration
#  template: True

  # Set to auto-match with a GivEnergy serial number, but you can override the serial or the sensor names
  # if it doesn't work or if you have more than one inverter you will need to list both
  #geserial: 're:sensor.givtcp_(.+)_soc_kwh'
  #geserial2: 're:sensor.givtcp2_(.+)_soc_kwh'
[... 453 lines cut ...]
  #futurerate_url: 'https://dataportal-api.nordpoolgroup.com/api/DayAheadPrices?date=DATE&market=N2EX_DayAhead&deliveryArea=UK&currency=GBP'
  #futurerate_adjust_import: True
  #futurerate_adjust_export: False
  #futurerate_peak_start: "16:00:00"
  #futurerate_peak_end: "19:00:00"
  #futurerate_peak_premium_import: 14
  #futurerate_peak_premium_export: 6.5

  # Watch list, a list of sensors to watch for changes and then update the plan if they change
  # This is useful for things like the Octopus Intelligent Slot sensor so that the plan update as soon as you plugin in
  # Only uncomment the items you actually have set up above in apps.yaml, of course you can add your own as well
  # Note those using +[] are lists that are appended to this list, whereas {} items are single items only
  #watch_list:
  #  - '{octopus_intelligent_slot}'
  # - '{octopus_ready_time}'
  # - '{octopus_charge_limit}'
  #  - '{octopus_saving_session}'
  #  - '+[car_charging_planned]'
  #  - '+[car_charging_soc]'
  #  - '{car_charging_now}'
  
  # Carbon Intensity data from National grid
carbon_intensity: 're:(sensor.carbon_intensity_uk)'

```

TonyHoyle, 2025-02-05:
One thing I notice is charge_rate and discharge_rate are in amps not watts as the documentation says they should be, so either the documentation is wrong or they should be multiplied by sensor.batvolt..

But you're using the amps version and it appears to be working for you, so no idea..

WyndStryke, 2025-02-05:
Well, firstly, I'm only using it for monitoring so if there are issues with control I wouldn't see them.

You could try hardwiring a template sensor to return an arbitrary figure in watts to see if that changes what you see.  Perhaps if predbat doesn't like the units, it discards the sensor and just uses a default instead.

zeclab, 2025-02-05:
Hi @TonyHoyle , 
I've just got my Fox setup last week and have been playing with Predbat. I see what you are saying about the BMS charge rate/discharge being in amps and can see the same as yourself on the chart.

[image]

I was looking the entities and found sensor.battery_charge & sensor.battery_discharge in Kw and changed them to W.

[image]

After updating the apps.yaml with these sensors and the chart is looking more along the lines of what I should be seeing?

[image]

TonyHoyle, 2025-02-05:
It isn't clear from the documentation whether charge_rate is supposed to be - it just says 'Battery charge rate entity in watts'.  
That could be interpreted the current charging current or the available charge rate.. based on its usage in the original yaml I assumed available charge rate & created a sensor for that.

That seems to clear up the graphs but of course may be wrong..

[image]

`    bms_current_charge_rate:
            friendly_name: "Current charge rate"
            unique_id: 'foxess_bms_current_charge_rate'
            unit_of_measurement: "W"
            device_class: power
            value_template: >
                {{ (( ( (states('sensor.batvolt') | float(0)) * states('sensor.bms_charge_rate') | float(0)))) | round(0) }}
                
    bms_current_discharge_rate:
            friendly_name: "Current discharge rate"
            unique_id: 'foxess_bms_current_discharge_rate'
            unit_of_measurement: "W"
            device_class: power
            value_template: >
                {{ (( ( (states('sensor.batvolt') | float(0)) * states('sensor.bms_discharge_rate') | float(0)))) | round(0) }}                
`

zeclab, 2025-02-05:
Cheers @TonyHoyle after adding those sensors, I'd say that my chart is looking more along the lines of what it should be.

[image]

gcoan, 2025-02-05:
> It isn't clear from the documentation what charge_rate is supposed to be - it just says 'Battery charge rate entity in watts'.
> That could be interpreted the current charging current or the available charge rate

The sensor that is configured in apps.yaml is expected to be an inverter sensor that (a) gives the current charge rate and (b) is something that Predbat can set to control the charge rate.

By using a template sensor for the entity, part b isn't going to work ....

Trefor has recently introduced the ability to control charge and discharge rate by % instead of watts value.  Is this something the Fox inverter supports?

I can certainly expand the documentation to make this clearer,  and if the Fox template needs changes, I can make those.

WyndStryke, 2025-02-05:
These are all the sensors available

**Sensors**
```
Ambient Temp 30.0 °C
Battery Charge 0 kW
Battery Charge Today 16.0 kWh
Battery Charge Total 251.1 kWh
Battery Discharge 2.14 kW
Battery Discharge Today 15.0 kWh
Battery Discharge Total 267.3 kWh
Battery SoC 28%
Battery SoH Unavailable
Battery Temp 21.2 °C
BMS Cell mV High 32 mV   (obviously not correct)
BMS Cell mV Low 31 mV   (obviously not correct)
BMS Cell Temp High 17.8 °C
BMS Cell Temp Low 13.2 °C
BMS Charge Rate 20.0 A
BMS Discharge Rate 35.0 A
BMS kWh Remaining 16.58 kWh
Connection Status Connected
CT2 Meter 0.0 kW
eon_next_drive_utility_meter_grid_daily cheap 17.300 kWh
eon_next_drive_utility_meter_grid_daily day 0.100 kWh
Feed-in 1.65 kW
Feed-in Today 15.900 kWh
Feed-in Total 106.100 kWh
Grid Consumption 0 kW
Grid Consumption Today 17.4 kWh
Grid Consumption Total 292.0 kWh
Grid CT 1.65 kW
Grid Frequency 50.0 Hz
Grid Voltage 248 V
Input Energy Today 16.6 kWh
Input Energy Total 228.5 kWh
Inverter Battery Current 9 A
Inverter Battery Power 2.14 kW
Inverter Battery Voltage 230 V
Inverter Current 8 A
Inverter Fault Code None
Inverter Power 2.06 kW
Inverter State On Grid
Inverter Temp 23.5 °C
Load Energy Today 5.8 kWh
Load Energy Total 245.4 kWh
Load Power 0.42 kW
Max Charge Current 50.0 A
Max Discharge Current 50.0 A
Max SoC 100%
Min SoC 10%
Min SoC (On Grid) 10%
PV Power 0.0 kW
PV1 Current 0 A
PV1 Power 0.0 kW
PV1 Power Total 0.000 kWh
PV1 Voltage 0 V
PV2 Current 0 A
PV2 Power 0.0 kW
PV2 Power Total 58.288 kWh
PV2 Voltage 0 V
PV3 Current 0 A
PV3 Power 0.0 kW
PV3 Power Total 65.257 kWh
PV3 Voltage 8 V
PV4 Current 0 A
PV4 Power 0.0 kW
PV4 Power Total 0.012 kWh
PV4 Voltage 0 V
Solar Generation Today 9.0 kWh
Solar Generation Total 126.2 kWh
Version: Manager 1.37
Version: Master 1.41
Version: Slave 1.01
Yield Today 21.1 kWh
Yield Total 288.7 kWh
```

TonyHoyle, 2025-02-05:
Will predbat cope with the charge in watts and convert internally or is it necessary to create a battery_charge_w (and the same for discharge) in the same way as we create a pv_power_w?

WyndStryke, 2025-02-05:
> as we create a pv_power_w?

My understanding is that we don't actually **need** to create a pv_power_w, it'll convert internally for the key stuff (but the config page and the charts won't see the converted figure).

gcoan, 2025-02-06:
> So if I understand things correctly (somewhat questionable!), we need an object to sit in the middle, if queried it should return sensor.battery_charge for the instantaneous battery charge rate (which could be changing from one moment to the next, if there is PV input, for example), and if written it should write to number.force_charge_power for the commanded force charge rate (in both case doing kW-W conversions if needed). Seems strange to me since it is conflating two different things, so I might be on the wrong track here.

No, this isn't quite right.

Instantaneous current rate that the battery is charging or discharging at is defined as 'battery power' in apps.yaml https://springfall2008.github.io/batpred/apps-yaml/#battery-information

The charge rate https://springfall2008.github.io/batpred/apps-yaml/#chargedischarge-rate (either in watts or percentage value) is the rate that the inverter is set to force charge at.  When you do a charge of the battery at full rate then predbat sets this to the maximum value for your inverter and then issues a 'start charge' command.  To stop charging predbat will set this to zero as well as issuing a stop charge command.

from what you have said you just need to configure charge rate to number.force_charge_power in apps.yaml (and similar for discharge)

WyndStryke, 2025-02-06:
> from what you have said you just need to configure charge rate to number.force_charge_power in apps.yaml (and similar for discharge)

Great stuff, that sounds way easier :-)

Thank you.

TonyHoyle, 2025-02-06:
It looks like that isn't enough.

Service call number/set_value data {'value': 5800, 'entity_id': 'number.force_charge_power'} failed with response None

predbat seems to be expecting this (write only) value to return something

WyndStryke, 2025-02-06:
If you look in the entities list for the FoxESS modbus device, can you see it's contents?

In the predbat/config page, I see:

```
charge_rate | - number.force_charge_power = 0.2 | - number.force_charge_power = 0.2-- | -- | --- number.force_charge_power = 0.2
discharge_rate | - number.force_discharge_power = 0.2 | - number.force_discharge_power = 0.2 - number.force_discharge_power = 0.2

```

and in develoment tools, I see:

```
number.force_charge_power
Force Charge Power | 0.2 | min: 0 max: 7 step: 0.001 mode: box unit_of_measurement: kW device_class: power friendly_name: Force Charge Power
-- | -- | --
Force Charge Power 0.2	
min: 0
max: 7
step: 0.001
mode: box
unit_of_measurement: kW
device_class: power
friendly_name: Force Charge Power

```

TonyHoyle, 2025-02-06:
A W to KW translator seems to work, as in it sets what seem to be sane values.
'I've never got the charge curve stuff to work but I don't think it matters as it'll presumably default to 100%.

[predbatt.yaml.txt](https://github.com/user-attachments/files/18693142/predbatt.yaml.txt)

WyndStryke, 2025-02-06:
> 'I've never got the charge curve stuff to work but I don't think it matters as it'll presumably default to 100%.

What I was wondering was if the automatic charge curve might have been the cause of your battery line being flat in your chart.

If it is trying to generate it from kW figures without converting, it might have decided that your charge curve was basically 0.001 all the way through the SoC range, hence a flat line.

Just a theory, don't know if it holds water or not, but might explain why mine was looking OK and yours wasn't.

WyndStryke, 2025-02-06:
Since I think we understand this aspect of the configuration now, how about the other things in the fox.yaml?  For example, I think it would be good to get Feed-in First etc working.  Not entirely sure I understand which Fox work mode maps onto which predbat work mode, but this is my first guess:

- Self-use   Use PV for load first, then charge battery, then export if battery full
- Feed-in First  Use PV for load first, then export, hold battery level steady as long as PV available (it will drop if PV unavailable).  I think Freeze charge and/or freeze discharge is the most similar to this?  
- Force charge / force discharge - I think the same as in predbat

```
# Services to control charging/discharging
# charge_start_service - Should be set to a service that is called when charging starts
  charge_start_service:
    service: select.select_option
    entity_id: select.work_mode
    option: "Force Charge"
#charge_freeze_service - If your inverter supports charge freeze set to a service that starts this mode
  charge_freeze_service:
    service: select.select_option
    entity_id: select.work_mode
    option: "Feed-in First"
#charge_stop_service - Should be set to a service that is called when charging/charge freeze stops
  charge_stop_service:
    service: select.select_option
    entity_id: select.work_mode
    option: "Self Use"
#discharge_start_service - Should be set to a service that is called when force export (discharge) starts
  discharge_start_service:
    service: select.select_option
    entity_id: select.work_mode
    option: "Force Discharge"
#discharge_stop_service - Should be set to a service that is called when export/export freeze stops
  discharge_stop_service:
    service: select.select_option
    entity_id: select.work_mode
    option: "Self Use"
#discharge_freeze_service - If your inverter supports export freeze set to a service that starts this mode
  discharge_freeze_service:
    service: select.select_option
    entity_id: select.work_mode
    option: "Feed-in First"

```

gcoan, 2025-02-06:
> If a charge curve isn't supplied, I think it'll try to figure it out from your historical charge data (IIRC, it doesn't simply default to 100%). 

The charge curve is optional

Predbat does default the charge curve to 100% all the way if you don't specify anything in apps.yaml, or if you set the charge curve to 'auto' it will try to work it out from historical data - but this auto-determination is clearly logged as such in the logfile.

https://springfall2008.github.io/batpred/apps-yaml/#battery-chargedischarge-curves

WyndStryke, 2025-02-06:
> or if you set the charge curve to 'auto' it will try to work it out from historical data

Ah that must be it, thank you

gcoan, 2025-02-06:
> Self-use Use PV for load first, then charge battery, then export if battery full, just the normal state I guess
> Feed-in First Use PV for load first, then export, hold battery level steady as long as PV available (it will drop if PV unavailable). I think Freeze charge and/or freeze discharge is the most similar to this?
> Force charge / force discharge - I think the same as in predbat

The mappings you have presented sound logical, except I see you have mapped Feed-in First to both Freeze charge and Freeze Discharge services.

[The Predbat terminology changed recently so what was previously called Discharge is now called Export.  The service names look to have not changed though]

In Predbat the behaviour of these is different:

- in freeze charge the battery level is held but if there is more solar than house load the battery will charge
- in freeze export the battery level is held, any solar will be used by the house or exported

https://springfall2008.github.io/batpred/what-does-predbat-do/#predbat-modes

It sounds from your description that Feed-in First more closely matches Freeze Export.  If there isn't an equivalent of Freeze Charge then you can set support_discharge_freeze to False

https://springfall2008.github.io/batpred/inverter-setup/#i-want-to-add-an-unsupported-inverter-to-predbat

TonyHoyle, 2025-02-07:
I had a go at creating charge_rate_kw and discharge_rate_kw but although it appeared to work I wasn't happy with the amount I had to change (I've never worked on a homeassistant plugin before, and I'm not going to submit anything that would break stuff).

Ideally it'd just understand which it was supposed to be using and translate automatically, but where the reading seems to go through a routine that does that, writing all goes direct and it's probably a big change to make that happen..

chalfontchubby, 2026-08-20:
The fox.yaml services proposed above (charge_freeze_service/discharge_freeze_service etc.) are now present in the template, so this looks resolved. Please open a new issue if there's still something specific to chase, e.g. the backup/peak-shaving work-mode questions.

closed 2026-08-20 by chalfontchubby

WyndStryke, 2026-08-20:
> The fox.yaml services proposed above (charge_freeze_service/discharge_freeze_service etc.) are now present in the template, so this looks resolved. Please open a new issue if there's still something specific to chase, e.g. the backup/peak-shaving work-mode questions.

👍 

All the basic stuff seems to be working now, particularly after the automatic kW to W (and vice-versa) conversions were added into predbat last year.   

There's one issue which I haven't yet written up (and probably won't this year), there does seem to be a difference in the way that Fox does force charge/force discharge to what predbat expects - Fox is controlling the inverter speed, not the battery charge/discharge speed.  So if predbat is asking for a slow discharge while there is PV (and PV power > requested discharge power), the system will actually slowly charge the battery instead of discharging.  This is only really noticeable if the slow-discharge-speed option is turned on.  I turned it off for other reasons, and most likely won't see this scenario again until next year.

chalfontchubby, 2026-08-20:
I merged something that might fix that today.