springfall2008/batpred #1966: Predheat heat to temp always 0c

symonchester, 2025-02-05:
As you can see from the screenshot Predheat always has predheat.heat_to_temp at 0c but looking ahead on the chart it shows the correct target temps.
The indoor target temp sensor reads correctly, it has 10 days of history and the number of days previous that Predheat is set to is 7.

Thanks
Symon

[image]

springfall2008, 2025-02-15:
Heat to temp will be non-zero when the heating is on and zero otherwise.

symonchester, 2025-02-15:
It's always at zero in the plan even when the heating is on. Another screenshot attached, current time is 21:50 and heating is currently at 20c with a setback of 18c at 23:00, as it's a heat pump it's always on but at different temps depending on time of day

[image]

symonchester, 2025-02-28:
Is it because I dont have this set, I have a heat pump?
    # Heating is turned on history
    # heating_active: binary_sensor.boiler_heatingactive

If its required then when data is it expecting?

symonchester, 2025-03-03:
You can close this ticket I've turned the feature off

closed 2025-03-03 by gcoan