springfall2008/batpred #1970: Terminate FreezeCharge for 30 minute timeslot if a larger power draw than historical data indicates is occuring

bigsmelly, 2025-02-06:
**Is your feature request related to a problem? Please describe.**

Predbat has the history, so it will decide that at 15:00 to 15:30 based on historical data, my use is low and the import rate is low enough that we will FreezeCharge (i.e. draw from the grid), so only adding a few pence. 

However if i then whack on an applicance unexpectedly, predbat is still in Freezecharge mode.  Then I get twitchy as we use from the grid and the price climbs.

**Describe the solution you'd like**

Predbat to notice if we have a long-running power draw (above what is expected from the historical data) that's not a transient like a kettle boiling. If the power draw has been for >5 minutes, predbat should recalcuate the plan with the guess that power draw will continue for the rest of the half hour slot.

so i'd expect, that if i turn on an applicance, after 5 minutes, predbat notices, and we might/might not move out of FreezeCharge depending on the future rates etc.

If the power draw has rolled over into the next 30 minute slot, then we should also not Freezecharge for that one.

So probably some logic when the draw drops back down to around the historical figure, we go back to normal. 

**Describe alternatives you've considered**

Predbat may already do this? Or I may not fully understand the logic. 

If this happens, I manually go to read-only mode if I think that it would be better to run off battery. 

**Additional context**

Hopefully someone can discuss if this is desirable or implementable.

chalfontchubby, 2026-09-18:
If the price climbs so that it changes the economics, then the freeze charge will stop and battery will take over. If it's freezing due to low import prices, that is because it's economically better, and switching to battery would cost more - would be counter productive?

I suggest we close this without a clearer reason to do something that's going to cost more? Please left me know if I'm missing something?

bigsmelly, 2026-09-23:
So this behaviour seems to not happen any more with newer predbat versions - perhaps some optimisation fixed it.
Happy to close

closed 2026-09-23 by bigsmelly