springfall2008/batpred #1976: Predbat addon

mjs7uk, 2025-02-07:
Trying to move from pred bay app deamon to the pred bat addon.

I get the following error when I run the addon (app deamon is stopped at this point)

Bootstrap Predbat
Startup
Traceback (most recent call last):
  File "/config/hass.py", line 5, in <module>
    import predbat
ModuleNotFoundError: No module named 'predbat'
Shutdown, sleeping 20 seconds before restarting
[20:09:11] INFO: Predbat init script running

Any help would be great

gcoan, 2025-02-07:
Suggest you delete the predbat add on, restart home assistant and install it again.

It can take quite a while to install and build the predbat add on as it's quite large, maybe it wasn't properly installed before?

springfall2008, 2025-02-09:
I would try deleting apps.yaml from the addon directory and then restarting the add-on so it downloads the files again. Then copy your apps.yaml across from appdaemon

closed 2026-08-28 by chalfontchubby